Ysurac/openmptcprouter #3407: Help reverting back to original firmware on RUTX50 device.

QRUXEL, 2024-06-15:
## Expected Behavior

A guide to help reverting back to Teletonica firmware

## Current Behavior

There is a guide for installing the firmware, but it does not say anywhere how to save a firmware backup of your boot config or firmware. I admit I'm not a firmware developer or well educated in Linux systems, but I really need to get the original firmware back ASAP and I don't know what to do and very afraid of bricking the device. When I try flashing the teltonika original firmware from ssh it says that the device "teltonika, rutx50" is not compatible with the image target "teltonika, rutx" 
Should I force the installation anyway with -F? 

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 

Teltonika, rutx50

Ysurac, 2024-06-15:
I never tried.
Maybe something like, but really not sure for the mtd part...:
```
   $ echo 1 > /proc/boot_info/rootfs/primaryboot
   $ cat /proc/boot_info/getbinary_bootconfig > /tmp/bootconfig.bin
   $ cat /proc/boot_info/getbinary_bootconfig1 > /tmp/bootconfig1.bin
   $ mtd write /tmp/bootconfig.bin /dev/mtd0
   $ mtd write /tmp/bootconfig1.bin /dev/mtd1
```

QRUXEL, 2024-06-15:
Hello Ysurac and thanks for helping, 

The first command results in error "can't create /proc/boot_info/rootfs/primaryboot: nonexistent directory"

closed 2024-09-19 by github-actions[bot]