Ysurac/openmptcprouter #3408: Offering OpenMPTC as Wifi Hotspot

nicolaeser, 2024-06-15:
## Expected Behavior
- .

## Current Behavior
Hey, how can I share the OpenMPTC as Wifi Hotspot?
I would love to share a Hotspot and when you join that Hotspot you get the bonded connection of OpenMPTC!

Regards

## Specifications
- OpenMPTCProuter version: v0.6
- OpenMPTCProuter VPS version: My Console shows: < OpenMPTCProuter VPS 0.1030 >
- OpenMPTCProuter VPS provider: NexoSystems
- OpenMPTCProuter platform: RPI3 (my one), RPI5 (my friends one)

nicolaeser, 2024-06-15:
[image]

It isn't connecting! Nothing happens except it is shown as loading.

stwr, 2024-06-15:
Have you tried adding the wifi interface to the lan firewall zone?

nicolaeser, 2024-06-15:
@stwr How would I do that? :)

nicolaeser, 2024-06-15:
So I can look if i did that

Ysurac, 2024-06-16:
You can maybe look at this doc: https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

nicolaeser, 2024-06-16:
@Ysurac Yeah seems like it isnt possible with only one ethernet port. What about Orange Pi? Is there an image for example for the Orange Pi 5+

Ysurac, 2024-06-18:
This example was made on a RPI, so only one port.

closed 2024-09-22 by github-actions[bot]