Ysurac/openmptcprouter #3413: Unable to Ping 10.255.255.2 from VPS - Routing Issue with OpenMPTCProuter

Martronic-SA, 2024-06-18:
## Expected Behavior

Use the IP address of the OpenMPTCProuter router via gt-tun0 on the VPS side to forward to a machine connected to the router on a specific port.

## Current Behavior

No communication, I cannot ping the address 10.255.255.2 from the VPS.

## Specifications

- OpenMPTCProuter version: 6.6.30
- OpenMPTCProuter VPS version: 6.1.0-21-amd64 0.1030
- OpenMPTCProuter VPS provider: OVH
- OpenMPTCProuter platform: 12th Gen Intel(R) Core(TM) i7-1265U

Ysurac, 2024-06-18:
You should be able to ping from VPS. What is the VPN used ? Seems to be glorytun TCP IP, the default is now OpenVPN TCP.

Martronic-SA, 2024-06-18:
I found that the 'orm' line was enabled in OpenVPN. As soon as I disabled it, it worked. Thanks again!

Martronic-SA, 2024-06-19:
**When comparing the ping times of my two installations, I observe significant differences in response times. The installation with the older version, which has been running for over 12 months, shows the following results:**

Période 1: nombre de ping = 1, le temps écoulé = 1639.00 ms (1.64 s), en moyenne = 1639.00 ms
Période 2: nombre de ping = 1, le temps écoulé = 1192.00 ms (1.19 s), en moyenne = 1192.00 ms
Période 3: nombre de ping = 2, le temps écoulé = 3406.00 ms (3.41 s), en moyenne = 1703.00 ms
Période 4: nombre de ping = 2, le temps écoulé = 3240.00 ms (3.24 s), en moyenne = 1620.00 ms
Période 5: nombre de ping = 1, le temps écoulé = 1021.00 ms (1.02 s), en moyenne = 1021.00 ms
Période 6: nombre de ping = 2, le temps écoulé = 3623.00 ms (3.62 s), en moyenne = 1811.50 ms
Période 7: nombre de ping = 1, le temps écoulé = 1097.00 ms (1.10 s), en moyenne = 1097.00 ms
Période 8: nombre de ping = 3, le temps écoulé = 6506.00 ms (6.51 s), en moyenne = 2168.67 ms
Période 9: nombre de ping = 2, le temps écoulé = 4487.00 ms (4.49 s), en moyenne = 2243.50 ms
Période 10: nombre de ping = 1, le temps écoulé = 1406.00 ms (1.41 s), en moyenne = 1406.00 ms
Période 11: nombre de ping = 2, le temps écoulé = 4800.00 ms (4.80 s), en moyenne = 2400.00 ms
Période 12: nombre de ping = 1, le temps écoulé = 1232.00 ms (1.23 s), en moyenne = 1232.00 ms
Période 13: nombre de ping = 2, le temps écoulé = 4358.00 ms (4.36 s), en moyenne = 2179.00 ms
Période 14: nombre de ping = 2, le temps écoulé = 3127.00 ms (3.13 s), en moyenne = 1563.50 ms
Période 15: nombre de ping = 4, le temps écoulé = 11097.00 ms (11.10 s), en moyenne = 2774.25 ms
Période 16: nombre de ping = 1, le temps écoulé = 1002.00 ms (1.00 s), en moyenne = 1002.00 ms
Période 17: nombre de ping = 1, le temps écoulé = 1352.00 ms (1.35 s), en moyenne = 1352.00 ms
Période 18: nombre de ping = 1, le temps écoulé = 1808.00 ms (1.81 s), en moyenne = 1808.00 ms
Période 19: nombre de ping = 3, le temps écoulé = 7032.00 ms (7.03 s), en moyenne = 2344.00 ms
Période 20: nombre de ping = 1, le temps écoulé = 1047.00 ms (1.05 s), en moyenne = 1047.00 ms
Période 21: nombre de ping = 4, le temps écoulé = 11626.00 ms (11.63 s), en moyenne = 2906.50 ms
Période 22: nombre de ping = 1, le temps écoulé = 1304.00 ms (1.30 s), en moyenne = 1304.00 ms
Période 23: nombre de ping = 2, le temps écoulé = 3156.00 ms (3.16 s), en moyenne = 1578.00 ms
Période 24: nombre de ping = 1, le temps écoulé = 1327.00 ms (1.33 s), en moyenne = 1327.00 ms
Période 25: nombre de ping = 1, le temps écoulé = 1052.00 ms (1.05 s), en moyenne = 1052.00 ms
Période 26: nombre de ping = 2, le temps écoulé = 3273.00 ms (3.27 s), en moyenne = 1636.50 ms
Période 27: nombre de ping = 1, le temps écoulé = 1622.00 ms (1.62 s), en moyenne = 1622.00 ms
Période 28: nombre de ping = 2, le temps écoulé = 2903.00 ms (2.90 s), en moyenne = 1451.50 ms
Période 29: nombre de ping = 1, le temps écoulé = 1966.00 ms (1.97 s), en moyenne = 1966.00 ms
Période 30: nombre de ping = 1, le temps écoulé = 1631.00 ms (1.63 s), en moyenne = 1631.00 ms
Période 31: nombre de ping = 1, le temps écoulé = 1505.00 ms (1.50 s), en moyenne = 1505.00 ms
Période 32: nombre de ping = 3, le temps écoulé = 6375.00 ms (6.38 s), en moyenne = 2125.00 ms
Période 33: nombre de ping = 2, le temps écoulé = 4297.00 ms (4.30 s), en moyenne = 2148.50 ms
Période 34: nombre de ping = 3, le temps écoulé = 7734.00 ms (7.73 s), en moyenne = 2578.00 ms
Période 35: nombre de ping = 1, le temps écoulé = 1168.00 ms (1.17 s), en moyenne = 1168.00 ms
Période 36: nombre de ping = 3, le temps écoulé = 6065.00 ms (6.07 s), en moyenne = 2021.67 ms
Période 37: nombre de ping = 2, le temps écoulé = 3157.00 ms (3.16 s), en moyenne = 1578.50 ms
Période 38: nombre de ping = 1, le temps écoulé = 1919.00 ms (1.92 s), en moyenne = 1919.00 ms
Période 39: nombre de ping = 7, le temps écoulé = 34330.00 ms (34.33 s), en moyenne = 4904.29 ms
Période 40: nombre de ping = 1, le temps écoulé = 1096.00 ms (1.10 s), en moyenne = 1096.00 ms
Période 41: nombre de ping = 1, le temps écoulé = 1557.00 ms (1.56 s), en moyenne = 1557.00 ms
Période 42: nombre de ping = 2, le temps écoulé = 3051.00 ms (3.05 s), en moyenne = 1525.50 ms
Période 43: nombre de ping = 1, le temps écoulé = 1491.00 ms (1.49 s), en moyenne = 1491.00 ms
Période 44: nombre de ping = 2, le temps écoulé = 3957.00 ms (3.96 s), en moyenne = 1978.50 ms
Période 45: nombre de ping = 5, le temps écoulé = 16054.00 ms (16.05 s), en moyenne = 3210.80 ms
Période 46: nombre de ping = 3, le temps écoulé = 6731.00 ms (6.73 s), en moyenne = 2243.67 ms
Période 47: nombre de ping = 2, le temps écoulé = 4331.00 ms (4.33 s), en moyenne = 2165.50 ms
Période 48: nombre de ping = 1, le temps écoulé = 1721.00 ms (1.72 s), en moyenne = 1721.00 ms
Période 49: nombre de ping = 2, le temps écoulé = 4859.00 ms (4.86 s), en moyenne = 2429.50 ms
Période 50: nombre de ping = 1, le temps écoulé = 1357.00 ms (1.36 s), en moyenne = 1357.00 ms
Période 51: nombre de ping = 1, le temps écoulé = 1361.00 ms (1.36 s), en moyenne = 1361.00 ms
Période 52: nombre de ping = 1, le temps écoulé = 1608.00 ms (1.61 s), en moyenne = 1608.00 ms
Période 53: nombre de ping = 1, le temps écoulé = 1111.00 ms (1.11 s), en moyenne = 1111.00 ms
Période 54: nombre de ping = 2, le temps écoulé = 2902.00 ms (2.90 s), en moyenne = 1451.00 ms

**In contrast, on the new installation, the ping times are significantly slower:**

[... cut ...]
Période 22: nombre de ping = 7, le temps écoulé = 32584.00 ms (32.58 s), en moyenne = 4654.86 ms
Période 23: nombre de ping = 2, le temps écoulé = 4184.00 ms (4.18 s), en moyenne = 2092.00 ms
Période 24: nombre de ping = 1, le temps écoulé = 1122.00 ms (1.12 s), en moyenne = 1122.00 ms
Période 25: nombre de ping = 7, le temps écoulé = 33044.00 ms (33.04 s), en moyenne = 4720.57 ms
Période 26: nombre de ping = 5, le temps écoulé = 19248.00 ms (19.25 s), en moyenne = 3849.60 ms

**Do you have an effective testing method to highlight the issue?**

Martronic-SA, 2024-06-21:
I was so focused on the issue of apparent connection losses that I didn't realize I was using an incorrect IP address. After reconfiguring and using the correct address, I no longer experience connection losses and they are now stable and performant.

closed 2024-09-24 by github-actions[bot]