Ysurac/openmptcprouter #3414: Shared bandwidth across all connections when downloading or streaming

CN-Creator, 2024-06-18:
## Current setup
I have successfully set up the router with, for testing, 2 internet connections. Furthermore I am creating a Wifi network with my Pi to "expose" the internet to the clients. Now I want to stream with the maximum bandwidth and have redundant connections.

## Expected Behavior
I expect to be streaming to some platform like Twitch across all connections. So if I have 3x 4k LTE, I have a combined upload of 12k. When one connections is lost I still want to be streaming across the others and not drop the stream.

## Current Behavior
During my testing I encountered the problem that download as well as streaming is only done over one connection at a time and this connections does not seem to change during the stream. The resulting problem is that I am only able to download/stream with the speed of this one connection and if this connections drops/disconnects the download or stream is canceled and not able to restore until a manual reconnect.

I dont now if my wish is to advanced for MPTCP. As I understand the protocol, it is deciding for one connection at the beginning of the download/stream or anything else and can not split or change it during it. Only when starting a new connection another path can be chosen.

## Specifications

  - OpenMPTCProuter version:  v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Thanks a lot in advance for your help and feedback!
Have a great day. :)

Ysurac, 2024-06-18:
What protocol you use for streaming ? based on TCP or UDP ?

cn0ss, 2024-06-18:
> What protocol you use for streaming ? based on TCP or UDP ?

I am using OBS to stream to Twitch and I would say its TCP RTMP streaming. Dont know if OBS supports UDP.

cn0ss, 2024-06-19:
I also often have the problem that when restarting the filesystem is readonly.

Ysurac, 2024-06-19:
If it's not an update, then you the proxy used for TCP by default is Shadowsocks-Rust and should work correctly.
The status page is full green ?

Hieuit0101, 2024-06-26:
> What protocol you use for streaming ? based on TCP or UDP ?

i also have the same problem, i use tcp stream, blest mode, bbr2 congestion, what do i need to check, pls help, thanks

closed 2024-09-30 by github-actions[bot]