Ysurac/openmptcprouter #3415: What is the IP address of my raspberry pi if I use 10.0.X.X

viorizz, 2024-06-18:
## Expected Behavior

Well the raspberry should be accessible using 192.168.100.1

## Current Behavior

Since I don't use this ip range but I use 10.0.X.X I can't access it. Or atleast I don't think I can, because when I try to access it by typing 10.0.100.1 it doesn't work.

My current IP range is set to 10.0.0.101 - 10.0.0.220

## Specifications

  - OpenMPTCProuter version: v0.60-6.1
  - OpenMPTCProuter VPS version: Haven't yet installed it
  - OpenMPTCProuter VPS provider: Hostinger
  - OpenMPTCProuter platform: RPI4B

viorizz, 2024-06-18:
Also might be interesting to add that I scanned my network with 3 different tools and still can't get anything the raspberry doesn't show up nowhere. I made a fresh install on the RPI, it's brand new received it today. Both leds are constant, and the network leds are blinking. I'm wondering if it's something I did wrong. Mabye I installed the RPI wrong. Did I need to install Raspberry os and on top install OMR ? I don't really know. But I could really use some advice here.

Ysurac, 2024-06-19:
The default IP of OpenMPTCProuter is always 192.168.100.1. You need to set IP on your client on 192.168.100.0/24 network to access it or use DHCP.

viorizz, 2024-06-20:
I figured it out. I though I had to connect the RPI to my network but that was a dumb way of thinking. So it's all good now.

closed 2024-06-20 by viorizz