Ysurac/openmptcprouter #3421: Fibocom FM350-GL Modem Not Recognized on Raspberry Pi 4B with OpenMPTCProuter

AlekZo, 2024-06-21:
**Expected Behavior**
The Fibocom FM350-GL modem should be recognized and function correctly with the OpenMPTCProuter on my Raspberry Pi 4B, similar to its performance with OpenWRT.

**Current Behavior**
When using OpenMPTCProuter, the Fibocom FM350-GL modem is not recognized and does not work. The same setup works perfectly with OpenWRT. I have checked the logs in Status->System log but did not find any relevant entries indicating recognition or issues with the modem.

**Specifications**
- OpenMPTCProuter version: openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz (2024/04/09)
- OpenMPTCProuter VPS version: Not applicable
- OpenMPTCProuter VPS provider: Not applicable
- OpenMPTCProuter platform: Raspberry Pi 4B

**Additional Hardware:**
- Fibocom FM350-GL modem
- Waveshare 5G Dongle (USB 3 Adapter)

AlekZo, 2024-06-21:
I can provide more info if needed. Please help.

Ysurac, 2024-06-21:
What protocol are you using ? ModemManager ?
Via SSH, ```mmcli -L``` return only the Waveshare modem ?
Do you have a USB powered hub ?

AlekZo, 2024-06-21:
> What protocol are you using ? ModemManager ? Via SSH, `mmcli -L` return only the Waveshare modem ? Do you have a USB powered hub ?

**What protocol are you using?**
I use:
- fm350-modem
- luci-proto-fm350

I created a "Fibocom" interface and set "/dev/ttyUSB4" as the Modem Port. I populated the APN and IPv4 as PDP, set custom DNS to 1.1.1.1 and 8.8.8.8, and added it to the WAN firewall.

**Via SSH, does mmcli -L return only the Waveshare modem?**
```
root@OpenMPTCProuter:~# mmcli -L
No modems were found
```
I have an m.2 FM350-GL modem with a Waveshare USB adapter (this one: https://www.waveshare.com/usb-to-m.2-b-key.htm).

**Do you have a USB-powered hub?**
I have a 30 cm cable with a USB female (power + data) and two male ports (one for power and data, and another for auxiliary power supply) (the same as shown in the Cable Upgrade screenshot here: https://www.waveshare.com/usb-to-m.2-b-key.htm).

**Additional information:**
In OpenWRT, it worked with the same protocol (fm350-modem and luci-proto-fm350) and without an auxiliary power supply. On OpenMPTCProuter, I tried both with the auxiliary power supply (plugged into a USB 2.0 port on my Raspberry Pi) and without the auxiliary power supply.

AlekZo, 2024-06-21:
I have noticed the following in the log "usb 2-2: no of_node; not parsing pinctrl DT":
``
[  130.371315] usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
[  130.402537] usb 2-2: New USB device found, idVendor=0e8d, idProduct=7127, bcdDevice= 0.01
[  130.410710] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  130.417845] usb 2-2: Product: FM350-GL
[  130.421590] usb 2-2: Manufacturer: Fibocom Wireless Inc.
[  130.427084] usb 2-2: no of_node; not parsing pinctrl DT
``

I also noticed that the Fibocom FM350-GL is missing in the `77-mm-fibocom-port-types.rules` file.

How can I add it? How do I know the code I need to insert there? There is also a comment at the beginning of the file:
```
# Do not edit this file; it will be overwritten on update.
```
So, I think my change would not be applied.

AlekZo, 2024-06-21:
I added this to the `77-mm-fibocom-port-types.rules` file. But it did not worked
```
# Fibocom FM350-GL
# ttyUSB0 (if #6): MD AT
# ttyUSB1 (if #3): AP GNSS
# ttyUSB2 (if #2): AP Log
# ttyUSB3 (if #4): AP META
# ttyUSB4 (if #9): Debug
# ttyUSB5 (if #7): MD Meta
# ttyUSB6 (if #8): NPT

ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="06", SUBSYSTEM=="tty", ENV{ID_MM_PORT_TYPE_AT_PRIMARY}="1"
ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="03", SUBSYSTEM=="tty", ENV{ID_MM_PORT_TYPE_GPS}="1"
ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="02", SUBSYSTEM=="tty", ENV{ID_MM_PORT_TYPE_AT_SECONDARY}="1"
ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="04", SUBSYSTEM=="tty", ENV{ID_MM_PORT_TYPE_DIAG}="1"
ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="09", SUBSYSTEM=="tty", ENV{ID_MM_PORT_IGNORE}="1"
ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="07", SUBSYSTEM=="tty", ENV{ID_MM_PORT_TYPE_QCDM}="1"
ATTRS{idVendor}=="0e8d", ATTRS{idProduct}=="7127", ENV{.MM_USBIFNUM}=="08", SUBSYSTEM=="tty", ENV{ID_MM_PORT_TYPE_NMEA}="1"
```

Ysurac, 2024-06-21:
There is a big topic on OpenWRT forum: https://forum.openwrt.org/t/fibocom-fm350-gl-support/142682
Maybe this can help...
I really don't know anything about this modem.

cr08, 2024-06-27:
I've had this card for a hot minute now. I've only briefly tried it on OMPTCP so can't comment on specifics but I have tried it with OpenWRT proper and other forks. The link posted just above me is a great resource. I'll add two more for packages that exist to better manage this card:

https://forum.openwrt.org/t/fibocom-fm350-gl-support/142682/133
https://github.com/koshev-msk/modemfeed > There's fm350 named packages here and they have precompiled packages for a couple different mainline OWRT versions. @Ysurac might need to advise which would be the closest to what OMPTCP uses but I think when I VERY briefly tried this I pulled from the 23.05 repo and it didn't throw any immediate errors.

The FM350 is handled a bit different from most other cards I've seen. TL;DR version: A number of AT commands need to be run in specific order to assign the correct PDP/APN credentials, activate and connect the card, then pull the IP info it has. There's a separate RNDIS eth device with no DHCP and the IP info pulled via AT commands is used to manually configure this interface. The above packages will do this all automatically. QMI/MBIM/ModemManager don't support it. I think MM is being worked on but at last check it only supports the card in PCIe mode currently but not USB.

If you have any luck getting it to work reliably, would love to hear about your experience and any tips. I've been working on my own little project off and on and currently I've been sticking closely to stock OpenWRT with Speedify added on but Speedify feels like it has a bit of a performance penalty. I'd love to give OMPTCP a go again but the UDP situation currently stops me from pursuing it further.

cr08, 2024-07-05:
Ok. Went in and did some more tinkering myself. ModemManager definitely seems to conflict. And unfortunately just toggling the 'Disable ModemManager' option in advanced settings doesn't seem to be enough. Once I removed the package entirely the FM350 stayed attached to the appropriate ttyUSB ports and allowed either of the linked FM350 packages to do their work.

Not a game breaking thing so far and might not even be OMTCP related, but the 464xlat package in my case seems to continue to try and bring up its own interface when a v4 address is already being provided natively. I had similar 'quirks' with this myself on other OWRT forks. Doesn't seem to affect anything other than flooding logs with interface up/down lines.

closed 2024-10-10 by github-actions[bot]

AshesNIL, 2024-10-21:
@cr08 hey brother was wondering if u got it working and if u did how the speeds look like ? I was thinking of doing the exact same project but I'm scared that module won't work so I searched and found your issue

cr08, 2024-10-21:
Hey! So yeah, essentially in the interim using the card in USB mode you have to disable ModemManager and use one of the available tools linked above. But they do work even with OMPTCP at least the last time I tested with it.

What happens with MM in place is it tries to poll one of the tty ports on the card that it shouldn't and it causes the card to power cycle. So the two can't coexist. If you rely on MM for other cards, you have to decide what to drop. For most cards you should be able to use the dedicated QMI or MBIM protocols without MM though.

There is a patch someone on the OWRT forums posted for MM which not only resolves this but also adds the ability for MM to work with the card itself. Keep in mind they state it's just for testing at the moment and not to consider it stable. But in my VERY brief usage it seems to behave. This would likely need to be something added as a custom patch in OMPTCP for now. I'm now aware of how long it'll take to get upstreamed in its current state

https://github.com/PMGZED/fm350gl-usb-mm-modul

My own personal project has been all over the place trying to focus on something that has been stable and it's still in the build and testing phase. Unfortunately with all the 5G cards out there being really power hungry, getting a proper power setup has been a chore and has caused some confusion with the stability testing. I've since moved back to the FM350-GL's and currently run them on a Pi 5 with a PCIe splitter and a couple of Waveshare's 5G Hat+ boards which basically provide a dedicated USB hub fed from PCIe and feed power from the GPIO header vs the PCIe cable. Knock on wood the power supply and stability seems to be good but the whole thing REALLY wants a stable 5.2+v feed. Feeding from a power bank that hovers around 5.0 even with 3-5A of capacity it wasn't happy.

That all said I'm not sure if OMPTCP supports the Pi 5 yet considering mainline OpenWRT still only has it in snapshots. So in the interim I've been trying to build my own simplified fork focusing on trying to use Zerotier for bonding since they support it via their 'Multipath' feature. But I may revisit. I haven't looked at OMPTCP in a while and Pi 5 support may already be on the agenda. If not, I may see if there's a way I can create my own test build on the latest OpenWRT snapshots. As noted above, one of the original roadblocks that stopped me from using OMPTCP is the UDP aggregation issues and my primary use is for video feeds which most broadcasting tools aim for UDP use. But I'm now considering going to a redundant feed (send the same data over two links) to save my own sanity as Zerotier may have a similar issue aggregating UDP. And moving to a redundant feed should, in theory, be more feasible for OMPTCP to handle.

AshesNIL, 2024-10-21:
Hey again, thanks for your return, I guess I'm gonna try to do this project then I want something just for internet access ( media, gaming and developing )