Ysurac/openmptcprouter #3422: Document OMR-bypass settings for SkyQ Apps

mario-minati, 2024-06-22:
**Is your enhancement proposal related to a problem? Please describe.**
After the initial setup of OMR the settings for streaming services felt like a rocky road. Especially there were no hints for Sky Q available.

**Describe the solution you'd like**
After looking into the DNS requests I could see from a Samsung TV and an iPad related to starting the Sky Q apps I created a list of domains that might be related with this service (here in germany) and bypassed them to one of our wan interfaces.

So I'd like to share tthis list as it might be helpfull to anyone:

| Domain                         |
|--------------------------------|
| sky.com                        |
| skycdp.com                     |
| skyanywhere.com                |
| epgsky.com                     |
| asnapieu.com                   |
| akamai.net                     |
| akamaiedge.net                 |
| adobedtm.com                   |
| akadns.net                     |
| skycdn.it                      |

I think we could create a wiki page to collect the bypass informations of all the different streaming services.

**Describe alternatives you've considered**
I also checked in the nDPI sources and the ndpi_content_match_c.inc file but here was only one occurence of Sky which is not related to any service definition: https://github.com/vel21ripn/nDPI/blob/flow_info-4/src/lib/ndpi_content_match.c.inc#L1685

But for sure there could be added a new service definition to that upstream package and after that an updated OMR build could be released.

**Additional context**
I'm willing to help :-)

Ysurac, 2024-06-22:
I added all domains with sky in it to a custom omr-bypass service. This will be available in next release. Thanks

mario-minati, 2024-06-22:
That's great, thank you :-)

closed 2024-06-22 by mario-minati