Ysurac/openmptcprouter #3423: Amazon EC2 Instance Problem with fast.com

chubx, 2024-06-22:
So, thanks to this OpenSource Project, it am now able to bond my Starlink with a backup connection. One main problem though is...

**Quick Details**:

- I have OMR installed on an **Amazon EC2 Instance** in a particular region.
- Port Speed for the VPS instance is more than enough to handle my bonded speed.
- The OMR router is using the default configuration from a squashfs image, with minor manual configuration to change the bonding interfaces from MACVLAN to **Normal**, then attach each interface to a VLAN.
- The OMR router has been tested with a DigitalOcean instance with no issues.

**MY Issue**:

When I tried the OMR router with the **EC2** instance, I get errors when speed testing using **fast.com**. See reference image below.

[image]

However, when testing using **speedtest.net**, I don't experience the issue I am having with ``fast.com. I've tried several VPS instances within the same EC2 regions, but all were the same, and had the same error with fast.com. 

I did tried on another EC2 regions, but all went well.

Can anyone help me further debug the issue, eventually help me solving the issue please?

Ysurac, 2024-06-23:
It's possible that fast.com bas Amazon IPs.

Network-Traditions, 2024-06-27:
Just an option to consider, I don't have direct knowledge this is actually the case, but many internet service providers block and/or throttle Netflix traffic.  Fast.com is a speedtest that uses Netflix servers.  Our system has never had a problem accessing fast.com and I would expect OpenMPTCProuter's obfuscation and encrypted traffic should make it difficult, but likely not impossible to decipher if a connection on one of its WAN connections is directed at a blacklisted or throttled IP address.  That being said, fast.com always reports an upload speed significantly higher than speedtest.net and [Waveform.com's Bufferbloat Test,](https://www.waveform.com/tools/bufferbloat), but it's not completely outside a reasonable result, just always significantly higher.  Additionally, you could also experience blocking as a result of your VPS public internet address.  We are currently battling that issue with our IP address provided by our Oracle Cloud VPS.  It turns out our static IP address is listed in Cloudflare Security as part of a malicious network so many Cloudflare protected websites block our address in addition to being constantly presented with CAPTCHA and security prompts.  If Oracle cannot offer a solution, we will have to keep acquiring new IP addresses for our VPS until we find one that doesn't have a negative reputation listing.

closed 2024-09-30 by github-actions[bot]