Ysurac/openmptcprouter #3424: When migrating to the new stable 6.0 version form the 5.4 version I notice a break in VOIP Traffic

JRTechServices, 2024-06-22:
## Version 5.4 MPTCP unbreakable VOIP Traffic works very well

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: Linode 
  - OpenMPTCProuter platform:

Ysurac, 2024-06-23:
You can try another VPN in System->OpenMPTCProuter, Wizard tab, "Advanced Settings" checkbox

Network-Traditions, 2024-06-27:
Have a look at #3392.  This is consistent with our recently launched system:

- Specifications
OpenMPTCProuter version: 0.60-6.1
OpenMPTCProuter VPS version: 0.1030 6.1.0-21-cloud-arm64
OpenMPTCProuter VPS provider: Oracle Cloud Ampere ARM Instance
OpenMPTCProuter platform: Debian 12 x86_64
WAN1 (Master) Starlink with 35ms average latency
WAN2 T-Mobile Business Internet with static public ip address at 100ms+ latency

We lose our VOIP connection when the "Master" connection is interrupted, but it survives non-master WAN interruptions.  Additionally, our Asterisk based VOIP system uses "Direct Media" connections.  To make it work with the current status of OpenMPTCProuter, we had to create individual forwards for each RTP/UDP port we use in our configuration.  Since we don't have high concurrent call volume on our system, this has been a work-around for now.  Prior to our migration to the system documented above, our 5.4 system used the "redundant" scheduler for optimum VOIP (i.e. latency/jitter stability) performance.  Since our deployment has realized significant performance and reliability improvements utilizing an Ampere ARM based VPS we've moved forward and are looking forward to many of the OpenMPTCProuter issues being resolved by upstream development.  Hope this helps!

JRTechServices, 2024-07-15:
Thank you very much for your reply. This helps a lot! I was going to do some additional testing.

closed 2024-10-20 by github-actions[bot]