Ysurac/openmptcprouter #3426: 10Gbe network interfaces not working on the Qnap Qhora 301w

T0p4Ze, 2024-06-25:
## Expected Behavior

## Current Behavior

Hi I can't use the 10 Gbe network interfaces yet in dsmeg I can see the firmware loaded
`[    4.644201] Aquantia AQR113C 90000.mdio-1:08: FW 5.4, Build 4, Provisioning 1`
`[    4.646964] Aquantia AQR113C 90000.mdio-1:00: FW 5.4, Build 4, Provisioning 1`
The 1Gbe interfaces work, I followed the [commit](https://github.com/openwrt/openwrt/commit/bd17683261e1dde1fdd4223745852de1ce8cfc89) but for some reason the 10Gbe don't want to work

thank you for your help, I've been looking for a solution for several days alone but in vain
## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 6.1.0-21-amd64 0.1030
  - OpenMPTCProuter VPS provider: Sapinet
  - OpenMPTCProuter platform: QNAP 301w

Ysurac, 2024-06-26:
Can you try the 6.6 snapshot release ? https://snapshots.openmptcprouter.com/6.6/

T0p4Ze, 2024-06-27:
> Can you try the 6.6 snapshot release ? https://snapshots.openmptcprouter.com/6.6/

So I tested some progress but not everywhere
The 10 Gbe lan works but I have very low bandwidth and very high latency compared to 1 Gbe which is very correct
The wan is partially detected but no internet
[image]

T0p4Ze, 2024-06-27:
> Can you try the 6.6 snapshot release ? https://snapshots.openmptcprouter.com/6.6/

So I tested some progress but not everywhere
The 10 Gbe lan works but I have very low bandwidth and very high latency compared to1 Gbe which is correct but which does not reach the speed that I had with the old version
The 10 Gbe wan is partially detected but no internet 
[image]
/

Ysurac, 2024-06-27:
What is the result of ```ip a``` and ```ethtool 10g-1``` via SSH on the router ?

T0p4Ze, 2024-06-27:
> What is the result of `ip a` and `ethtool 10g-1` via SSH on the router ?

root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo
       valid_lft forever preferred_lft forever
2: miireg: <> mtu 0 qdisc noop state DOWN group default qlen 1000
    link/netrom
3: lan4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 24:5e:be:52:90:93 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.2/24 brd 192.168.8.255 scope global lan4
       valid_lft forever preferred_lft forever
4: lan3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 24:5e:be:52:90:94 brd ff:ff:ff:ff:ff:ff
5: lan2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 24:5e:be:52:90:95 brd ff:ff:ff:ff:ff:ff
6: lan1: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 24:5e:be:52:90:96 brd ff:ff:ff:ff:ff:ff
7: 10g-1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 24:5e:be:52:90:97 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 brd 192.168.0.255 scope global 10g-1
       valid_lft forever preferred_lft forever
8: 10g-2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 24:5e:be:52:90:98 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global 10g-2
       valid_lft forever preferred_lft forever
    inet6 fe80::265e:beff:fe52:9098/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
9: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 2ca:3c92:a5df::
10: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
11: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
12: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr 8a92:be58:2249::
15: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
51: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1420 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none
    inet 10.255.252.2/24 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::a266:b621:f393:a3ec/64 scope link stable-privacy proto kernel_ll
       valid_lft forever preferred_lft forever
54: ifb4lan2: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 1e:1b:22:37:0c:c3 brd ff:ff:ff:ff:ff:ff
57: ifb4lan3: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 0e:5e:ad:bc:8d:28 brd ff:ff:ff:ff:ff:ff
60: ifb410g-1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether c6:5e:08:d3:8d:a1 brd ff:ff:ff:ff:ff:ff

T0p4Ze, 2024-06-27:
I forgot to post `ethtool 10g-1` @Ysurac 

root@OpenMPTCProuter:~# ethtool 10g-1
Settings for 10g-1:
        Supported ports: [ ]
        Supported link modes:   100baseT/Half 100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                1000baseKX/Full
                                10000baseKX4/Full
                                10000baseKR/Full
                                2500baseT/Full
                                5000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  100baseT/Half 100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                1000baseKX/Full
                                10000baseKX4/Full
                                10000baseKR/Full
                                2500baseT/Full
                                5000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  100baseT/Half 100baseT/Full
                                             1000baseT/Full
                                             10000baseT/Full
                                             2500baseT/Full
                                             5000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 10000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 8
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Link detected: yes

Ysurac, 2024-06-28:
All seems ok here.
What do you have when you do a ```ping -B -I 10g-1 192.168.0.254``` ?

T0p4Ze, 2024-06-28:
> ping -B -I 10g-1 192.168.0.254

root@OpenMPTCProuter:~# ping -B -I 10g-1 192.168.0.254
ping: bind: Address not available

For information, I tried 10g-1 and 10g-2 on openwrt and it works

Ysurac, 2024-06-28:
Strange that ping is not able to bind to the interface, maybe due to the "-".
What is the result if you try ```ping -B -I 192.168.0.1 192.168.0.254``` and if this doesn't work ```ping -I 10g-1 192.168.0.254``` ?

T0p4Ze, 2024-06-28:
root@OpenMPTCProuter:~# ping -B -I 192.168.0.1 192.168.0.254
PING 192.168.0.254 (192.168.0.254) from 192.168.0.1 : 56(84) bytes of data.
64 bytes from 192.168.0.254: icmp_seq=1 ttl=64 time=0.282 ms
64 bytes from 192.168.0.254: icmp_seq=2 ttl=64 time=0.226 ms
64 bytes from 192.168.0.254: icmp_seq=3 ttl=64 time=0.217 ms
64 bytes from 192.168.0.254: icmp_seq=4 ttl=64 time=0.217 ms
64 bytes from 192.168.0.254: icmp_seq=5 ttl=64 time=0.289 ms
64 bytes from 192.168.0.254: icmp_seq=6 ttl=64 time=0.240 ms
64 bytes from 192.168.0.254: icmp_seq=7 ttl=64 time=0.210 ms
64 bytes from 192.168.0.254: icmp_seq=8 ttl=64 time=0.261 ms
64 bytes from 192.168.0.254: icmp_seq=9 ttl=64 time=0.232 ms
64 bytes from 192.168.0.254: icmp_seq=10 ttl=64 time=0.255 ms
^C
--- 192.168.0.254 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9389ms
rtt min/avg/max/mdev = 0.210/0.242/0.289/0.026 ms

root@OpenMPTCProuter:~# ping -I 10g-1 192.168.0.254
PING 192.168.0.254 (192.168.0.254) from 10.0.0.0 10g-1: 56(84) bytes of data.
64 bytes from 192.168.0.254: icmp_seq=1 ttl=64 time=0.271 ms
64 bytes from 192.168.0.254: icmp_seq=2 ttl=64 time=0.265 ms
64 bytes from 192.168.0.254: icmp_seq=3 ttl=64 time=0.191 ms
64 bytes from 192.168.0.254: icmp_seq=4 ttl=64 time=0.257 ms
64 bytes from 192.168.0.254: icmp_seq=5 ttl=64 time=0.242 ms
64 bytes from 192.168.0.254: icmp_seq=6 ttl=64 time=0.255 ms
64 bytes from 192.168.0.254: icmp_seq=7 ttl=64 time=0.243 ms
64 bytes from 192.168.0.254: icmp_seq=8 ttl=64 time=0.243 ms
64 bytes from 192.168.0.254: icmp_seq=9 ttl=64 time=0.256 ms
64 bytes from 192.168.0.254: icmp_seq=10 ttl=64 time=0.251 ms
^C
--- 192.168.0.254 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9334ms
rtt min/avg/max/mdev = 0.191/0.247/0.271/0.020 ms

Ysurac, 2024-06-29:
So the problem seems to really be the name of the interface...
I need to find a real way to fix that, but for now you can rename them (this doesn't survice a reboot):
```
ip link set 10g-1 down
ip link set 10g-1 name 10g1
ip link set 10g1 up
```
And then set the device in Network->Interface

T0p4Ze, 2024-06-30:
> So the problem seems to really be the name of the interface... I need to find a real way to fix that, but for now you can rename them (this doesn't survice a reboot):
> 
> ```
> ip link set 10g-1 down
> ip link set 10g-1 name 10g1
> ip link set 10g1 up
> ```
> 
> And then set the device in Network->Interface

Bad news, the problem continues despite the fix

Ysurac, 2024-06-30:
You still can't do a ```ping -B -I ....``` ?

T0p4Ze, 2024-06-30:
Same result as before

root@OpenMPTCProuter:~# ping -B -I 10g1 192.168.0.254
ping: bind: Address not available

root@OpenMPTCProuter:~# ping -B -I 192.168.0.1 192.168.0.254
PING 192.168.0.254 (192.168.0.254) from 192.168.0.1 : 56(84) bytes of data.
64 bytes from 192.168.0.254: icmp_seq=1 ttl=64 time=0.290 ms
64 bytes from 192.168.0.254: icmp_seq=2 ttl=64 time=0.234 ms
64 bytes from 192.168.0.254: icmp_seq=3 ttl=64 time=0.246 ms
64 bytes from 192.168.0.254: icmp_seq=4 ttl=64 time=0.265 ms
64 bytes from 192.168.0.254: icmp_seq=5 ttl=64 time=0.288 ms
^C
--- 192.168.0.254 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4188ms
rtt min/avg/max/mdev = 0.234/0.264/0.290/0.022 ms

root@OpenMPTCProuter:~# ping -I 10g1 192.168.0.254
PING 192.168.0.254 (192.168.0.254) from 10.0.0.0 10g1: 56(84) bytes of data.
64 bytes from 192.168.0.254: icmp_seq=1 ttl=64 time=0.316 ms
64 bytes from 192.168.0.254: icmp_seq=2 ttl=64 time=0.258 ms
64 bytes from 192.168.0.254: icmp_seq=3 ttl=64 time=0.263 ms
64 bytes from 192.168.0.254: icmp_seq=4 ttl=64 time=0.319 ms
64 bytes from 192.168.0.254: icmp_seq=5 ttl=64 time=0.300 ms
^C
--- 192.168.0.254 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4168ms
rtt min/avg/max/mdev = 0.258/0.291/0.319/0.025 ms

T0p4Ze, 2024-07-09:
Hi I tried the new snapshot 6.6 from July 9, 2024 I have a little progress: 10g-1 wan partially works but ping is still not possible and the internet speed is very low (I don't know if this comes from the router or vps)
[image]

Ysurac, 2024-07-10:
Can you try to disable MPTCP on 4G box and check speed again ?
What is your VPS provider ?

T0p4Ze, 2024-07-11:
[image]
Same result unfortunately and my vps provider is Sapinet
But I tried with my rpi 3 and I get a good result so for me the vps is not the problem
[image]

Ysurac, 2024-07-11:
What is the proxy used ?
Do you have anything strange in System->kernel log ?

T0p4Ze, 2024-07-12:
I use ShadowSocks-Rust 2022 as proxy, this is used by default I think

I didn't see anything strange in the logs but if in doubt I'll post them anyway.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.33 (runner@fv-az885-336) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0+26695-820823198d) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Wed Jun 19 08:24:49 2024
[    0.000000] Machine model: QNAP 301w
[    0.000000] OF: reserved mem: 0x0000000040000000..0x0000000040ffffff (16384 KiB) nomap non-reusable nss@40000000
[    0.000000] OF: reserved mem: 0x000000004a400000..0x000000004a5fffff (2048 KiB) nomap non-reusable tzapp@4a400000
[    0.000000] OF: reserved mem: 0x000000004a600000..0x000000004a9fffff (4096 KiB) nomap non-reusable bootloader@4a600000
[    0.000000] OF: reserved mem: 0x000000004aa00000..0x000000004aafffff (1024 KiB) nomap non-reusable sbl@4aa00000
[    0.000000] OF: reserved mem: 0x000000004ab00000..0x000000004abfffff (1024 KiB) nomap non-reusable smem@4ab00000
[    0.000000] OF: reserved mem: 0x000000004ac00000..0x000000004affffff (4096 KiB) nomap non-reusable memory@4ac00000
[    0.000000] OF: reserved mem: 0x000000004b000000..0x0000000050efffff (97280 KiB) nomap non-reusable wcnss@4b000000
[    0.000000] OF: reserved mem: 0x0000000050f00000..0x0000000050ffffff (1024 KiB) nomap non-reusable q6_etr_dump@50f00000
[    0.000000] OF: reserved mem: 0x0000000051000000..0x00000000510fffff (1024 KiB) nomap non-reusable m3_dump@51000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
[    0.000000]   node   0: [mem 0x0000000051100000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 28 pages/cpu s75432 r8192 d31064 u114688
[    0.000000] pcpu-alloc: s75432 r8192 d31064 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 root=PARTUUID=9e34f1ab-add2-e59c-db36-61d414c02cd4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007ea80000-0x000000007eb80000] (1MB)
[    0.000000] Memory: 879592K/1048576K available (9216K kernel code, 1304K rwdata, 6408K rodata, 2048K init, 306K bss, 168984K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000504] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000518] pid_max: default: 32768 minimum: 301
[    0.005499] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.005513] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.009820] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.010086] rcu: Hierarchical SRCU implementation.
[    0.010090] rcu: 	Max phase no-delay instances is 1000.
[    0.011211] smp: Bringing up secondary CPUs ...
[    0.011877] Detected VIPT I-cache on CPU1
[    0.011983] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012710] Detected VIPT I-cache on CPU2
[    0.012782] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[... 1353 lines cut ...]
Jul 12 00:22:25 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Jul 12 00:22:36 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
[... cut ...]
Jul 12 00:25:39 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Jul 12 00:25:50 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Jul 12 00:26:01 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Jul 12 00:26:12 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Jul 12 00:26:22 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules

No, I'm not on the latest snapshot of July 10, I'll try the update tomorrow

Ysurac, 2024-07-12:
It seems that there is a problem with Shadowsocks-Rust here. Rules are not set.
Can you give me the result of ```uci show shadowsocks-rust``` ?

T0p4Ze, 2024-07-12:
root@OpenMPTCProuter:~# uci show shadowsocks-rust
shadowsocks-rust.hi1=ss_redir
shadowsocks-rust.hi1.server='sss0'
shadowsocks-rust.hi1.local_address='::'
shadowsocks-rust.hi1.local_port='1100'
shadowsocks-rust.hi1.mode='tcp_and_udp'
shadowsocks-rust.hi1.timeout='1000'
shadowsocks-rust.hi1.fast_open='0'
shadowsocks-rust.hi1.verbose='0'
shadowsocks-rust.hi1.syslog='1'
shadowsocks-rust.hi1.reuse_port='0'
shadowsocks-rust.hi1.mptcp='1'
shadowsocks-rust.hi1.ipv6_first='1'
shadowsocks-rust.hi1.no_delay='0'
shadowsocks-rust.hi2=ss_redir
shadowsocks-rust.hi2.server='sss1'
shadowsocks-rust.hi2.local_address='::'
shadowsocks-rust.hi2.local_port='1101'
shadowsocks-rust.hi2.mode='tcp_and_udp'
shadowsocks-rust.hi2.timeout='1000'
shadowsocks-rust.hi2.fast_open='0'
shadowsocks-rust.hi2.verbose='0'
shadowsocks-rust.hi2.syslog='1'
shadowsocks-rust.hi2.reuse_port='0'
shadowsocks-rust.hi2.mptcp='1'
shadowsocks-rust.hi2.ipv6_first='1'
shadowsocks-rust.hi2.no_delay='0'
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='all'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.ss_rules.ifnames='lan1' '10g-2'
shadowsocks-rust.sss0=server
shadowsocks-rust.sss0.disabled='0'
shadowsocks-rust.sss0.server='185.44.xxxxxx'
shadowsocks-rust.sss0.server_port='65280'
shadowsocks-rust.sss0.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.obfs='0'
shadowsocks-rust.sss0.password='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
shadowsocks-rust.sss1=server
shadowsocks-rust.sss1.disabled='1'
shadowsocks-rust.sss1.server='185.44.xxxxxx'
shadowsocks-rust.sss1.server_port='65280'
shadowsocks-rust.sss1.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss1.obfs='0'
shadowsocks-rust.sss1.password='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
shadowsocks-rust.dns=ss_tunnel
shadowsocks-rust.dns.disabled='1'
shadowsocks-rust.dns.mode='tcp_and_udp'
shadowsocks-rust.dns.server='sss0'
shadowsocks-rust.dns.local_port='5353'
shadowsocks-rust.dns.tunnel_address='8.8.8.8:53'
shadowsocks-rust.tracker_sss0=ss_local
shadowsocks-rust.tracker_sss0.server='sss0'
shadowsocks-rust.tracker_sss0.local_address='127.0.0.1'
shadowsocks-rust.tracker_sss0.local_port='1111'
shadowsocks-rust.tracker_sss0.mode='tcp_and_udp'
shadowsocks-rust.tracker_sss0.timeout='600'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'

Ysurac, 2024-07-12:
The problem is maybe related to the "10g-2" interface...
Can you try a ```uci set openmptcprouter.settings.restrict_to_lan="0" && uci commit openmptcprouter``` and wait 5 minutes and check if there is no more the "Reload Shadowsocks Rust rules" message in log ?

T0p4Ze, 2024-07-12:
> The problem is maybe related to the "10g-2" interface... Can you try a `uci set openmptcprouter.settings.restrict_to_lan="0" && uci commit openmptcprouter` and wait 5 minutes and check if there is no more the "Reload Shadowsocks Rust rules" message in log ?

The command fixed the problem after I don't know if I can increase the speed since I have an 8Gbe FFTH connection (4G is used as backup), I read on the Openwrt forum that the SOC limited the flow because no source to manage the NSS but in cases it remains very satisfactory

[image]

Ysurac, 2024-07-12:
ok, it's better. I don't know if it's because there is no NSS support or if the CPU is slow. 
I need to check that, it's far slower than a RPI5 here...

closed 2024-10-15 by github-actions[bot]