Ysurac/openmptcprouter #3427: ipv6 non-mptcp-able traffic goes out directly via interfaces

rdmitry0911, 2024-06-26:
## Expected Behavior

if omr-vps has ipv6 only public address then
  - all ipv4 traffic should go out via omr-vps tunnel 
  - all ipv6 mptcp-able traffic should go out via omr-vps
  - all ipv6 non-mptcp-able traffic should go out via omr-vps tunnel 
## Current Behavior

  - all ipv4 traffic goes out via omr-vps tunnel (**good**)
  - all ipv6 mptcp-able traffic goes out via omr-vps (**good**)
  - all ipv6 non-mptcp-able traffic goes out directly via master interface (**wrong**)
```
root@OpenMPTCProuter:~# traceroute6 -n google.com
traceroute to google.com (2a00:1450:4001:828::200e), 30 hops max, 72 byte packets
 1  2a0d:XXXX:cf10::1  0.904 ms  0.787 ms  0.730 ms
 2  2a0d:XXXX:da8::1  58.468 ms  60.348 ms  61.558 ms
 3  2620:134:b0fe:252::100  55.900 ms  55.749 ms  60.920 ms
 4  *^C
root@OpenMPTCProuter:~# 
```
  -  Even though ping via interfaces works fine
```
root@OpenMPTCProuter:~# ping -I eth1 google.com
PING google.com(tzfraa-aa-in-x0e.1e100.net (2a00:1450:4001:81d::200e)) from 2a0d:xxxx::83c2 eth1: 56 data bytes
64 bytes from fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e): icmp_seq=1 ttl=58 time=53.3 ms
64 bytes from fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e): icmp_seq=2 ttl=58 time=49.0 ms
64 bytes from fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e): icmp_seq=3 ttl=58 time=57.4 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 48.959/53.201/57.360/3.430 ms

root@OpenMPTCProuter:~# ping -4 -I eth1 google.com
PING google.com (216.58.206.78) from 192.168.1.24 eth1: 56(84) bytes of data.
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=1 ttl=58 time=59.7 ms
64 bytes from lhr35s11-in-f14.1e100.net (216.58.206.78): icmp_seq=2 ttl=58 time=56.0 ms
64 bytes from lhr35s11-in-f14.1e100.net (216.58.206.78): icmp_seq=3 ttl=58 time=58.0 ms
^C
--- google.com ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5006ms
rtt min/avg/max/mdev = 52.856/56.854/59.678/2.238 ms

root@OpenMPTCProuter:~# ping -4 -I eth2 google.com
PING google.com (216.58.206.78) from 192.168.10.199 eth2: 56(84) bytes of data.
64 bytes from lhr35s11-in-f14.1e100.net (216.58.206.78): icmp_seq=1 ttl=57 time=53.9 ms
64 bytes from lhr35s11-in-f14.1e100.net (216.58.206.78): icmp_seq=2 ttl=57 time=49.2 ms
64 bytes from lhr35s11-in-f14.1e100.net (216.58.206.78): icmp_seq=3 ttl=57 time=58.0 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2000ms
rtt min/avg/max/mdev = 49.215/53.678/57.964/3.573 ms

root@OpenMPTCProuter:~# ping -I eth2 google.com
PING google.com(fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e)) from 2a02:xxxx::92ea eth2: 56 data bytes
64 bytes from fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e): icmp_seq=1 ttl=56 time=56.3 ms
64 bytes from fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e): icmp_seq=2 ttl=56 time=54.7 ms
64 bytes from fra15s18-in-x0e.1e100.net (2a00:1450:4001:81d::200e): icmp_seq=3 ttl=56 time=54.0 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 53.978/54.975/56.280/0.964 ms

```

openmptcprouter/status doesn't look good

[image]

## Possible Solution

## Steps to Reproduce the Problem

  1. Install fresh omr build
  2. configure wan1(eth1, Multipath TCP master) and wan2(eth2, Multipath TCP enabled) as dhcpv6 
  3. configure wan3(eth1, Multipath TCP disabled) and wan4(eth2, Multipath TCP disabled) as dhcp

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  v0.61-snapshot-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0.-18-amd64
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform:  x86_64
  - Country:  Greece

AndreaLeidi, 2024-06-26:
Same here. I have no IPv6 on WAN interfaces (just one address /64 on VPS). TCP/UDP ok and getting 10/10 on test-ipv6.com but ICMPv6 (ping, traceroute) doesn't go out, 6in4 tunnel dead, apparently no route for this kind of traffic (?). 

See also #3425 

GT/Shadowsocks-libev on x86_64 (no AES-NI), OMR v0.61-snapshot-5.4 r0+16862-170d9e447d (same with the stable build, both upgrade and fresh install). Firewall rules untouched, OMR and VPS.

from OMR

root@link:~# ping -6 google.com
PING google.com(mil07s17-in-x0e.1e100.net (2a00:1450:4002:414::200e)) 56 data bytes
From link.###.it (fd56:90f4:dd4c::1) icmp_seq=1 Time exceeded: Hop limit
From link.###.it (fd56:90f4:dd4c::1) icmp_seq=2 Time exceeded: Hop limit
From link.###.it (fd56:90f4:dd4c::1) icmp_seq=3 Time exceeded: Hop limit
^C
--- google.com ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 3004ms

from LAN

$ ping -6 google.com
PING google.com (2a00:1450:4002:411::200e) 56 data bytes
^C
--- google.com ping statistics ---
5 packets transmitted, 0 received, 100% packet loss, time 4113ms

traceroute -6
(device at #2 is not any gateway, is my LAN wifi access point!)

$ traceroute -6 google.com
traceroute to google.com (2a00:1450:4002:411::200e), 30 hops max, 80 byte packets
 1  link.###.it (fd56:90f4:dd4c::1)  0.495 ms  0.477 ms  0.468 ms
 2  * fd56:90f4:dd4c:0:f692:bfff:fe29:9573 (fd56:90f4:dd4c:0:f692:bfff:fe29:9573)  0.845 ms  0.737 ms
 3  link.###.it (fd56:90f4:dd4c::1)  0.905 ms  1.032 ms  1.088 ms
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * link.###.it (fd56:90f4:dd4c::1)  2.715 ms  2.868 ms
10  * * *
11  * link.###.it (fd56:90f4:dd4c::1)  3.016 ms *
12  * * *
13  link.###.it (fd56:90f4:dd4c::1)  2.639 ms  2.706 ms  2.698 ms

6in4 no traffic

[image]

Ysurac, 2024-06-26:
I would need the result of ```ip -6 r```

Ysurac, 2024-06-26:
A new image is compiling that should fix the issue.

AndreaLeidi, 2024-06-26:
Thanks a lot @Ysurac for your help!

rdmitry0911, 2024-06-26:
> A new image is compiling that should fix the issue.

With [this patch](https://github.com/Ysurac/openmptcprouter-feeds/commit/56e83988e59761175a5251b5586282f9c2e90464) the tunnel doesn't work. Even thou I see tun0 in ifconfig, I can't ping the other end of the tunnel:
```
root@OpenMPTCProuter:~# ping 10.255.255.1
PING 10.255.255.1 (10.255.255.1) 56(84) bytes of data.
^C
--- 10.255.255.1 ping statistics ---
6 packets transmitted, 0 received, 100% packet loss, time 5219ms

```
I see this in logread |tail -n 35
```
root@OpenMPTCProuter:~# logread |tail -n 35
Jun 26 14:19:56 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:20:00 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 14:20:00 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 14:20:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 26 14:20:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 26 14:20:06 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:20:16 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:20:17 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 26 14:20:19 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 14:20:19 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 14:20:26 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:20:36 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:20:46 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:20:56 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:21:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 26 14:21:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 26 14:21:05 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 14:21:05 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 14:21:06 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:21:16 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:21:17 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 26 14:21:26 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:21:28 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 14:21:28 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 14:21:36 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:21:46 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 14:21:46 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 14:21:46 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:21:56 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:22:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 26 14:22:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 26 14:22:06 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:22:16 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
Jun 26 14:22:17 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 26 14:22:26 OpenMPTCProuter user.notice omr6in4: addr: 10.255.255.2 - peer: 10.255.255.1
```
```
root@OpenMPTCProuter:~# ifconfig tun0
tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.255.255.2  P-t-P:10.255.255.1  Mask:255.255.255.255
          inet6 addr: fe80::cb35:832f:a63f:bd8a/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:34 errors:0 dropped:0 overruns:0 frame:0
          TX packets:772 errors:0 dropped:2121 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:2826 (2.7 KiB)  TX bytes:96473 (94.2 KiB)
```

Ysurac, 2024-06-26:
I forgot to remove some debug log...
You are using Glorytun TCP VPN, if the 10.255.255.1 end doesn't answer, it should not be related to IPv6 and OMR6IN4.
Can you try OpenVPN TCP ? I had many issues with Glorytun TCP on 6.1 kernel...

rdmitry0911, 2024-06-26:
OpenVPN TCP desn't work either. I think it is a problem with the patch as without this patch tunnel does work both with Glorytun TCP and with OpenVPN TCP
```
root@OpenMPTCProuter:~# logread |tail -n 30
Jun 26 15:33:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 26 15:33:01 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Jun 26 15:33:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jun 26 15:33:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jun 26 15:33:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jun 26 15:33:09 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 26 15:33:12 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (2a01:4f8:212:25cf:ff00::199)
Jun 26 15:33:50 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 15:33:50 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 15:33:59 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 15:33:59 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 15:34:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 26 15:34:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 26 15:34:02 OpenMPTCProuter daemon.notice netifd: Interface 'wan2_4' is setting up now
Jun 26 15:34:02 OpenMPTCProuter daemon.notice netifd: Interface 'wan2_4' is now down
Jun 26 15:34:03 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 26 15:34:04 OpenMPTCProuter daemon.warn openvpn(omr)[31377]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Jun 26 15:34:04 OpenMPTCProuter daemon.warn openvpn(omr)[31377]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jun 26 15:34:04 OpenMPTCProuter daemon.notice openvpn(omr)[31377]: TCP/UDP: Preserving recently used remote address: [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301
Jun 26 15:34:04 OpenMPTCProuter daemon.notice openvpn(omr)[31377]: Attempting to establish TCP connection with [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301
Jun 26 15:34:06 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (2a01:4f8:212:25cf:ff00::199)
Jun 26 15:34:06 OpenMPTCProuter daemon.err openvpn(omr)[31377]: TCP: connect to [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301 failed: Host is unreachable
Jun 26 15:34:06 OpenMPTCProuter daemon.notice openvpn(omr)[31377]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Jun 26 15:34:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 15:34:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 26 15:35:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 26 15:35:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 26 15:35:03 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 26 15:35:13 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 26 15:35:13 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down

```

AndreaLeidi, 2024-06-27:
After installing the latest snapshot (June 27), the whole IPv6 traffic is now routed through the 6in4 tunnel, bringing it back to life. Apparently no proxy for TCP and UDP v6.
This fixes the ping / traceroute issues and I still have 10/10 on IPv6 support but OMR-bypass is not working anymore.

------------------------------------------------------------------------------
   (r0+16862-170d9e447d)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.61-snapshot
 TARGET:      x86_64
 ARCH:        x86_64

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Thu Jun 27 01:24:55 UTC 2024
-----------------------------------------------------

:~# ip -6 r
default from fd00::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
fd00::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
fd56:90f4:dd4c::/64 dev eth0 proto static metric 9999 pref medium
fd56:90f4:dd4c::/60 dev eth0 metric 69999 pref medium
unreachable fd56:90f4:dd4c::/48 dev lo proto static metric 2147483647 pref medium
fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth4 proto kernel metric 256 pref medium
fe80::/64 dev eth3 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth1 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth2 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth3 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth4 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium  #### ADDED
default via fd56:90f4:dd4c:: dev eth0 metric 69999 pref medium

PREVIOUSLY WAS:

:~# ip -6 r
default from fd00::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
fd00::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
fd56:90f4:dd4c::/64 dev eth0 proto static metric 9999 pref medium
fd56:90f4:dd4c::/60 dev eth0 metric 69999 pref medium
unreachable fd56:90f4:dd4c::/48 dev lo proto static metric 2147483647 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth3 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev eth4 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth1 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth2 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth3 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth4 proto kernel metric 256 pref medium
default via fd56:90f4:dd4c:: dev eth0 metric 69999 pref medium

AndreaLeidi, 2024-06-27:
default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium 

doesn't survive to router reboots...

rdmitry0911, 2024-06-27:
> After installing the latest snapshot (June 27), the whole IPv6 traffic is now routed through the 6in4 tunnel, bringing it back to life. Apparently no proxy for TCP and UDP v6.

That's the problem I think why my tunnel doesn't work. my omr-vps has only ipv6 public address. With routing of all ipv6 traffic through the 6in4 tunnel there is no way to access my omr-vps public IP

Ysurac, 2024-06-27:
@rdmitry0911 A default route to the VPS IPv6 should be created so the 6in4 should not make any problem. I will test if it's working (I've more tested IPv4 behind IPv6 only connections for now, so without 6in4).

@AndreaLeidi The route ```default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium``` is created by ```/etc/init.d/omr-6in4``` and seems to survive reboot on my tests since last changes.
For omr-bypass, what is not working ? domains bypass ?
What is the proxy you use ? TCP should always use the proxy, UDP by default use the 6in4 tunnel.

rdmitry0911, 2024-06-27:
> @rdmitry0911 A default route to the VPS IPv6 should be created so the 6in4 should not make any problem. I will test if it's working (I've more tested IPv4 behind IPv6 only connections for now, so without 6in4).

With the [latest commit](https://github.com/Ysurac/openmptcprouter-feeds/commit/82ecc58e28c72e30652ccb447a35ae48c61c29aa) there is no default route to the VPS IPv6. here is my routing table with ip -6 ro sh
```
default from fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
2a02:1388:2163:c95a::6da dev eth2 metric 66 pref medium
2a02:1388:2163:c95a::/64 dev eth2 proto static metric 256 pref medium
unreachable 2a02:1388:2163:c95a::/64 dev lo proto static metric 2147483647 pref medium
2a0d:3344:1a87:cf00::/56 from 2a0d:3344:1a87:cf10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf00::/56 from 2a0d:3344:1a87:cf10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf00::/56 from fd6f:7e1:2556:10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf00::/56 from fd6f:7e1:2556:10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf10::b44 dev eth1 metric 63 pref medium
2a0d:3344:1a87:cf10::/64 dev eth1 proto static metric 256 pref medium
unreachable 2a0d:3344:1a87:cf10::/64 dev lo proto static metric 2147483647 pref medium
fd24:16af:d717::/48 from 2a02:1388:2163:c95a::6da via fe80::3c51:e0ff:fec4:4eb2 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::/48 from 2a02:1388:2163:c95a::/64 via fe80::3c51:e0ff:fec4:4eb2 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::/48 from fd24:16af:d717::6da via fe80::3c51:e0ff:fec4:4eb2 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::/48 from fd24:16af:d717::/64 via fe80::3c51:e0ff:fec4:4eb2 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::/64 dev eth2 metric 65 pref medium
fd24:16af:d717::/64 dev eth2 proto static metric 256 pref medium
unreachable fd24:16af:d717::/64 dev lo proto static metric 2147483647 pref medium
fd6d:d2f0:12d8::/64 dev eth0 proto static metric 1024 pref medium
fd6d:d2f0:12d8::/64 dev eth0 proto static metric 9999 pref medium
fd6d:d2f0:12d8::/60 dev eth0 proto kernel metric 256 expires 6769sec pref medium
fd6d:d2f0:12d8::/60 dev eth0 metric 69999 pref medium
unreachable fd6d:d2f0:12d8::/48 dev lo proto static metric 2147483647 pref medium
fd6f:7e1:2556::/48 from 2a0d:3344:1a87:cf10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from 2a0d:3344:1a87:cf10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from fd6f:7e1:2556:10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from fd6f:7e1:2556:10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556:10::/64 dev eth1 metric 64 pref medium
fd6f:7e1:2556:10::/64 dev eth1 proto static metric 256 pref medium
unreachable fd6f:7e1:2556:10::/64 dev lo proto static metric 2147483647 pref medium
fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium
default via fe80::7624:9fff:feb7:80cb dev eth1 metric 63 pref medium
default via 2a02:1388:2163:c95a:: dev eth2 metric 65 pref medium
default via fe80::3c51:e0ff:fec4:4eb2 dev eth2 metric 66 pref medium
default via fd6d:d2f0:12d8:: dev eth0 metric 69999 pref medium
```
There is nothing related to 2a01:4f8:212:25cf:ff00::199 which is my vps public IP
May be I have to use another build?

rdmitry0911, 2024-06-28:
As far as I understand the routing logic when aggregation is enables should be like this:

**ipv4-tcp goes:**

1. to vps if vps has ipv4 public ip
2. to the ipv6 vps tunnel if vps does not have ipv4 public ip

**ipv4-noncp goes:**

1. to the ipv4 vps tunnel if vps has ipv4 public ip
2. to the ipv6 vps tunnel if vps does not have ipv4 public ip

**ipv6-tcp goes:**

1. to vps if vps has ipv6 public ip
2. to the ipv4 vps tunnel if vps does not have ipv6 public ip

**ipv6-noncp goes:**

1. to the ipv6 vps tunnel if vps has ipv6 public ip
2. to the ipv4 vps tunnel if vps does not have ipv6 public ip

Otherwise all the traffic should go via master interface

Can someone tell me how this logic is implemented through the scripts? I can find 6in4 scripts, but there are no 4in6 ones.

Ysurac, 2024-06-28:
It's not the logic.
TCP IPv4 and IPv6 use the proxy and everything else use the VPN. 
6in4 is over the VPN and is used for anything that is not TCP and IPv6.

So the VPN, used for anything that is not TCP, is always in IPv4 and 6in4 over it, even when using VPS IPv6.
The Proxy, for TCP, is always used for TCP IPv4 and IPv6.

AndreaLeidi, 2024-06-28:
> The route `default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium` is created by `/etc/init.d/omr-6in4` and seems to survive reboot on my tests since last changes.

@Ysurac you're right, I'm sorry. I found openmptcprouter.settings.disable_6in4='1' on my router. Set it = 0 makes your latest changes working as expected, reboot after reboot. It's strange, I only use LUCI to config and this key is not linked to any control. 

> For omr-bypass, what is not working ? domains bypass ? What is the proxy you use ? TCP should always use the proxy, UDP by default use the 6in4 tunnel.

Proxy - I'm with Shadowsocks-libev, my router CPU doesn't have AES-NI.
OMR-bypass is by domain since Cloudflare is blocking my VPS IPv6 address on many sites.

I'm gonna run an untouched OMR image on virtualbox to compare each uci key with those of my router, in search of possible inconsistencies.

rdmitry0911, 2024-06-28:
> is always in IPv4 and 6in4 over it, even when using VPS IPv6.

What is the reason to have 6in4 over ipv4 vpn when ipv6 vpn is an option?

Ysurac, 2024-06-28:
@rdmitry0911 It's easier to manage only a 6in4 for now.
This may change in the future.

Ysurac, 2024-06-28:
The "openmptcprouter.settings.disable_6in4" disable 6in4 tunnel so if set to 1, it's normal that 6in4 tunnel is not set up.
If IPv6 is enabled in the wizard, you should have the tunnel.

rdmitry0911, 2024-06-28:
> The "openmptcprouter.settings.disable_6in4" disable 6in4 tunnel so if set to 1, it's normal that 6in4 tunnel is not set up. If IPv6 is enabled in the wizard, you should have the tunnel.

The problem is not with 6in4-omr6in4 tunnel, it exists, the problem is with the main tunnel for non-tcp traffic between omr and vps.  This one doesn't exist.

I have this in /etc/config/openmptcprouter
```
config settings 'settings'
        option enabled '1'
        option sipalg '1'
        option vpn 'openvpn'
        option master 'balancing'
        option disable_ipv6 '0'
        option check_ipv4_website 'http://ip.openmptcprouter.com'
        option check_ipv6_website 'http://ipv6.openmptcprouter.com'
        option status_vps_timeout '3'
        option status_getip_timeout '1'
        option enable_nodelay '1'
        option scaling_governor 'performance'
        option menu 'OpenMPTCProuter'
        option disableintfrename '1'
        option tracebox '0'
        option version '0.61-snapshot-6.1'
        option allmptcpovervpn '0'
        option serial 'Not'
        option mptcpovervpn 'wireguard'
        option dns64 '0'
        option proxy 'shadowsocks-rust'
        option ha '0'
        option encryption 'aes-256-gcm'
        option disable_fastopen '0'
        option external_check '1'
        option restrict_to_lan '0'
        option debug '0'
        option disablegwping '0'
        option disableloopdetection '0'
        option disableserverhttptest '0'
        option defaultgw '1'
        option modemmanager '1'
        option disableserverping '0'
        option shadowsocksudp '0'
        option ndpi '1'
        option sfe_enabled '0'
        option sfe_bridge '0'
        option vnstat_backup '0'
        option apilc '1719588022'
        option firstboot '0'
        option sysupgrade_lc '1719518285'
        option disable_6in4 '0'
```
and here is ifconfig output:
```
6in4-omr6in4 Link encap:IPv6-in-IPv4  
          inet6 addr: fe80::a00:2/126 Scope:Link
          inet6 addr: fe80::aff:fc03/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1 errors:10484 dropped:0 overruns:0 carrier:10484
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:46 (46.0 B)

eth0      Link encap:Ethernet  HWaddr BC:24:11:9C:08:0B  
          inet addr:192.168.99.1  Bcast:192.168.99.255  Mask:255.255.255.0
          inet6 addr: fe80::be24:11ff:fe9c:80b/64 Scope:Link
          inet6 addr: fd6d:d2f0:12d8::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:2002 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2906 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:187882 (183.4 KiB)  TX bytes:697890 (681.5 KiB)

[... 86 lines cut ...]
fd24:16af:d717::/64 dev eth2 proto static metric 256 pref medium
unreachable fd24:16af:d717::/64 dev lo proto static metric 2147483647 pref medium
fd6d:d2f0:12d8::/64 dev eth0 proto static metric 9999 pref medium
fd6d:d2f0:12d8::/60 dev eth0 metric 69999 pref medium
unreachable fd6d:d2f0:12d8::/48 dev lo proto static metric 2147483647 pref medium
fd6f:7e1:2556::/48 from 2a0d:3344:1a87:cf10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from 2a0d:3344:1a87:cf10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from fd6f:7e1:2556:10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from fd6f:7e1:2556:10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556:10::/64 dev eth1 metric 64 pref medium
fd6f:7e1:2556:10::/64 dev eth1 proto static metric 256 pref medium
unreachable fd6f:7e1:2556:10::/64 dev lo proto static metric 2147483647 pref medium
fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1201 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium
default via fe80::7624:9fff:feb7:80cb dev eth1 metric 63 pref medium
default via 2a02:1388:2163:c95a:: dev eth2 metric 65 pref medium
default via fe80::3c51:e0ff:fec4:4eb2 dev eth2 metric 66 pref medium
default via fd6d:d2f0:12d8:: dev eth0 metric 69999 pref medium

```

Ysurac, 2024-06-29:
This is the problem: ```OMR-Tracker-Server: Master server vps down (2a01:4f8:212:25cf:ff00::199)```
The VPS doesn't answer to ping in test.
Can you try a ```ping -B -I eth1 2a01:4f8:212:25cf:ff00::199``` and same with eth2 ?
Both interface have many IPv6, do you know why ?

rdmitry0911, 2024-06-29:
> The VPS doesn't answer to ping in test.

This didn't happen with the old version. 

> Can you try a `ping -B -I eth1 2a01:4f8:212:25cf:ff00::199` and same with eth2 ?

yes, 2a01:4f8:212:25cf:ff00::199 is reachablew via both interfaces

> Both interface have many IPv6, do you know why ?

I have no idea. This is how dhcpv6 works. This happens all the time with the new firmware and with the old one. The router sitting on omr eth2 interface is also running openwrt ans is under mine control. It also has several ipv6 addresses on its wan interface like this:
```
wwan0     Link encap:Ethernet  HWaddr E6:90:C9:AD:69:CA  
          inet addr:10.63.201.223  Bcast:10.63.201.255  Mask:255.255.255.192
          inet6 addr: 2a02:1388:2163:c95a:e490:c9ff:fead:69ca/64 Scope:Global
          inet6 addr: fe80::e490:c9ff:fead:69ca/64 Scope:Link
          UP BROADCAST RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:45271390 errors:0 dropped:2459 overruns:0 frame:0
          TX packets:32529836 errors:0 dropped:11 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:45589037975 (42.4 GiB)  TX bytes:5771839966 (5.3 GiB)

```

rdmitry0911, 2024-06-30:
It seems, that for any reason the route to vps in new firmware is via 6in4-omr6in4 This is the problem why the omr-vps tunnel soesn't start. My vps doesn't have ipv4 public address.

tcpdump -n -i any host 2a01:4f8:212:25cf:ff00::199
```
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
06:45:22.102351 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.60203 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 267495658, win 42700, options [mss 1220,sackOK,TS val 1706537960 ecr 0,nop,wscale 9], length 0
06:45:22.920762 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.17991 > 2a01:4f8:212:25cf:ff00::199.65280: Flags [S], seq 4209595465, win 42700, options [mss 1220,sackOK,TS val 1706538778 ecr 0,nop,wscale 9,mptcp 4 capable v1], length 0
06:45:23.140833 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.60203 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 267495658, win 42700, options [mss 1220,sackOK,TS val 1706538999 ecr 0,nop,wscale 9], length 0
06:45:23.950827 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.17991 > 2a01:4f8:212:25cf:ff00::199.65280: Flags [S], seq 4209595465, win 42700, options [mss 1220,sackOK,TS val 1706539809 ecr 0,nop,wscale 9,mptcp 4 capable v1], length 0
06:45:24.142949 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.30549 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1398342312, win 42700, options [mss 1220,sackOK,TS val 1706540001 ecr 0,nop,wscale 9], length 0
06:45:25.230829 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.30549 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1398342312, win 42700, options [mss 1220,sackOK,TS val 1706541089 ecr 0,nop,wscale 9], length 0
06:45:26.233016 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.30563 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2621848121, win 42700, options [mss 1220,sackOK,TS val 1706542091 ecr 0,nop,wscale 9], length 0
06:45:27.310832 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.30563 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2621848121, win 42700, options [mss 1220,sackOK,TS val 1706543169 ecr 0,nop,wscale 9], length 0
06:45:28.312956 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.30575 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 891390944, win 42700, options [mss 1220,sackOK,TS val 1706544171 ecr 0,nop,wscale 9], length 0
06:45:29.380821 6in4-omr6in4 Out IP6 2a02:1388:2163:c95a:1ebf:ceff:fe6c:92ea.30575 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 891390944, win 42700, options [mss 1220,sackOK,TS val 1706545239 ecr 0,nop,wscale 9], length 0
^C
10 packets captured
10 packets received by filter
0 packets dropped by kernel

```

Ysurac, 2024-06-30:
There is something that is not working correctly.
For the test if server IP answer or not, a ```curl -s -k --interface eth1 https://[2a01:4f8:212:25cf:ff00::199]:65500``` is done on each interface where Multipath is enabled. Can you give me result of previous command and the result of ```multipath``` command ?

rdmitry0911, 2024-06-30:
> There is something that is not working correctly. For the test if server IP answer or not, a `curl -s -k --interface eth1 https://[2a01:4f8:212:25cf:ff00::199]:65500` is done on each interface where [...]

Here are the results:
```
root@OpenMPTCProuter:~# curl -s -k --interface eth1 https://[2a01:4f8:212:25cf:ff00::199]:65500
"Welcome to OpenMPTCProuter Server part"root@OpenMPTCProuter:~# 
```
```
root@OpenMPTCProuter:~# curl -s -k --interface eth2 https://[2a01:4f8:212:25cf:ff00::199]:65500
"Welcome to OpenMPTCProuter Server part"root@OpenMPTCProuter:~# 
```
```
root@OpenMPTCProuter:~# multipath 
6in4-omr6in4 is deactivated
erspan0 is deactivated
eth0 is deactivated
eth1 is deactivated
eth2 is in default mode
gre0 is deactivated
gretap0 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
```

Ysurac, 2024-06-30:
Only MPTCP is enabled only on eth2, strange.
I would need more system log.

rdmitry0911, 2024-06-30:
> Only MPTCP is enabled only on eth2, strange. I would need more system log.

Here are last lines of dmesg
```
Jun 30 18:14:09 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:14:09 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:14:20 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:14:20 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:14:49 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:14:49 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:15:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 30 18:15:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 30 18:15:03 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 30 18:15:04 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (2a01:4f8:212:25cf:ff00::199)
Jun 30 18:15:33 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:15:33 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:15:38 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:15:38 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:16:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 30 18:16:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 30 18:16:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 30 18:16:47 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:16:47 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:16:52 OpenMPTCProuter daemon.warn openvpn(omr)[5673]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Jun 30 18:16:52 OpenMPTCProuter daemon.warn openvpn(omr)[5673]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jun 30 18:16:52 OpenMPTCProuter daemon.notice openvpn(omr)[5673]: TCP/UDP: Preserving recently used remote address: [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301
Jun 30 18:16:52 OpenMPTCProuter daemon.notice openvpn(omr)[5673]: Attempting to establish TCP connection with [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301
Jun 30 18:16:53 OpenMPTCProuter daemon.err openvpn(omr)[5673]: TCP: connect to [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301 failed: Host is unreachable
Jun 30 18:16:53 OpenMPTCProuter daemon.notice openvpn(omr)[5673]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Jun 30 18:17:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 30 18:17:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 30 18:17:03 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 30 18:17:04 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (2a01:4f8:212:25cf:ff00::199)
Jun 30 18:17:06 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:17:06 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:17:24 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:17:24 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:18:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 30 18:18:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 30 18:18:03 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 30 18:18:20 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:18:20 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:19:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 30 18:19:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 30 18:19:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
Jun 30 18:19:04 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (2a01:4f8:212:25cf:ff00::199)
Jun 30 18:19:13 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:19:13 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:19:27 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:19:27 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:19:44 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:19:44 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:20:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
[... cut ...]
Jun 30 18:25:49 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is setting up now
Jun 30 18:25:49 OpenMPTCProuter daemon.notice netifd: Interface 'wan1_4' is now down
Jun 30 18:26:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to off from eth2 is in default mode
Jun 30 18:26:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Jun 30 18:26:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on omr-vps.ddns.net, no server API answer on omr-vps.ddns.net)
```

rdmitry0911, 2024-07-01:
And just to be sure all the information is provided, here is my /etc/config/network 
```
root@OpenMPTCProuter:~# cat /etc/config/network 

config interface 'loopback'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option multipath 'off'
	option device 'lo'
	option metric '7'

config globals 'globals'
	option ula_prefix 'fd6d:d2f0:12d8::/48'
	option multipath 'enable'
	option mptcp_path_manager 'fullmesh'
	option mptcp_scheduler 'blest'
	option congestion 'bbr'
	option mptcp_checksum '0'
	option mptcp_debug '0'
	option mptcp_syn_retries '4'
	option mptcp_subflows '8'
	option mptcp_add_addr_accepted '1'
	option mptcp_add_addr_timeout '120'
	option mptcp_pm_type '0'
	option mptcp_disable_initial_config '0'
	option mptcp_force_multipath '1'
	option mptcpd_enable '0'
	option mptcp_fullmesh_num_subflows '1'
	option mptcp_fullmesh_create_on_err '1'
	option mptcp_ndiffports_num_subflows '1'
	option packet_steering '0'

config interface 'lan'
	option proto 'static'
	option ipaddr '192.168.99.1'
	option netmask '255.255.255.0'
	option device 'eth0'
	option ifname 'eth0'
	option ipv6 '1'
	option delegate '0'
	option addlatency '0'
	option txqueuelen '2000'
	option multipath 'off'
	option ip4table 'lan'
	option label 'lan'
	option defaultroute '0'
	option peerdns '0'
	option ip6assign '60'
	option metric '9999'

config rule 'lan_rule'
	option lookup 'lan'
	option priority '100'

config interface 'wan1'
	option device 'eth1'
	option proto 'dhcpv6'
	option ip4table 'wan'
	option multipath 'on'
	option defaultroute '0'
	option delegate '0'
	option addlatency '0'
	option peerdns '0'
	option ipv6 '1'
	option label 'starlink6'
	option reqaddress 'try'
	option reqprefix 'no'
	option iface_map '0'
	option iface_dslite '0'
	option iface_464xlate '0'
	option modalias 'usb:v0BDAp8156d3104dc00dsc00dp00icFFiscFFip00in00-1c:bf:ce:6c:83:c2'
	option product 'bda/8156/3104'
	option metric '3'

config interface 'wan3'
	option device 'eth1'
	option proto 'dhcp'
	option ip4table 'wan'
	option multipath 'off'
	option defaultroute '0'
	option peerdns '0'
	option label 'starlink'
	option ipv6 '0'
	option delegate '0'
	option metric '4'
	option modalias 'usb:v0BDAp8156d3104dc00dsc00dp00icFFiscFFip00in00-1c:bf:ce:6c:83:c2'
	option product 'bda/8156/3104'

config interface 'wan4'
	option device 'eth2'
	option proto 'dhcp'
	option ip4table 'wan'
	option multipath 'off'
	option defaultroute '0'
	option peerdns '0'
	option label 'cosmote'
	option ipv6 '0'
	option delegate '0'
	option metric '5'
	option modalias 'usb:v0BDAp8156d3104dc00dsc00dp00icFFiscFFip00in00-1c:bf:ce:6c:92:ea'
	option product 'bda/8156/3104'

config device 'wan1_dev'
	option name 'eth1'
	option txqueuelen '1000'

config interface 'wan2'
	option device 'eth2'
	option proto 'dhcpv6'
	option ip4table 'wan'
	option multipath 'master'
	option defaultroute '0'
	option delegate '0'
	option addlatency '0'
	option peerdns '0'
	option ipv6 '1'
	option label 'cosmote6'
	option reqaddress 'try'
	option reqprefix 'no'
	option iface_map '0'
	option iface_dslite '0'
	option iface_464xlate '0'
	option modalias 'usb:v0BDAp8156d3104dc00dsc00dp00icFFiscFFip00in00-1c:bf:ce:6c:92:ea'
	option product 'bda/8156/3104'
	option metric '6'

config device 'wan2_dev'
	option name 'eth2'
	option txqueuelen '1000'

config interface 'omrvpn'
	option device 'tun0'
	option proto 'none'
	option ip4table 'vpn'
	option multipath 'off'
	option leasetime '12h'
	option type 'tunnel'
	option txqueuelen '100'
	option metric '1500'

config device 'tun0'
	option name 'tun0'
	option mtu '1500'

config interface 'omr6in4'
	option proto '6in4'
	option ip4table 'vpn'
	option multipath 'off'
	option ipaddr '10.255.252.2'
	option peeraddr '10.255.252.1'
	option gateway 'fe80::a00:1/126'
	option ip6addr 'fe80::a00:2/126'
	option force_link '1'
	option metric '1201'

config rule6 'eth0_fw_rule6'
	option priority '1'
	option mark '0x65399999'
	option lookup '9999'

config rule6 'eth1_fw_rule6'
	option priority '1'
	option mark '0x65394'
	option lookup '4'

config rule6 'eth2_fw_rule6'
	option priority '1'
	option mark '0x65396'
	option lookup '6'

config rule6 'tun0_fw_rule6'
	option priority '1'
	option mark '0x65391500'
	option lookup '1500'

config rule6 'all_fw_rule6'
	option priority '1'
	option mark '0x6539'
	option lookup '6991337'

config device 'lan_dev'
	option name 'eth0'

config rule6 '6in4_omr6in4_fw_rule6'
	option priority '1'
	option mark '0x65391201'
	option lookup '1201'

config device 'wan4_dev'
	option name 'wan4'
	option txqueuelen '1000'

config device 'wan3_dev'
	option name 'wan3'
	option txqueuelen '1000'

config rule6 'wan3_fw_rule6'
	option priority '1'
	option mark '0x65396'
	option lookup '6'

config rule6 'wan4_fw_rule6'
	option priority '1'
	option mark '0x65397'
	option lookup '7'

config rule 'eth0_fw_rule'
	option priority '1'
	option mark '0x45399999'
	option lookup '9999'

config rule 'eth1_fw_rule'
	option priority '1'
	option mark '0x45394'
	option lookup '4'

config rule 'eth2_fw_rule'
	option priority '1'
	option mark '0x45396'
	option lookup '6'

config rule 'tun0_fw_rule'
	option priority '1'
	option mark '0x45391500'
	option lookup '1500'

config rule '6in4_omr6in4_fw_rule'
	option priority '1'
	option mark '0x45391201'
	option lookup '1201'

config rule 'all_fw_rule'
	option priority '1'
	option mark '0x4539'
	option lookup '991337'
```

rdmitry0911, 2024-07-01:
> You have the same device used multiples times, some with multipath enabled, some not. Enable Multipath on all interfaces using same device.

After your suggestion I think I don't quite understand the logic of enabling and disabling multipath. I thought only if an interface is capable to deliver tcp packets to the proxy on vps, this interface should be marked as multipath enabled/master. Otherwise multipath should be set disabled. As far as wan3/wan4 in my case are ipv4 only interfaces, they can not deliver anything to my ipv6-only vps. That's why I set multipath for them disabled. Nevertheless, I marked wan3/wan4 as enabled. Still no omr <-> vps tunnel. here are all the details:

curl -s -k --interface eth1 https://[2a01:4f8:212:25cf:ff00::199]:65500 
returns nothing 

curl -s -k --interface eth2 https://[2a01:4f8:212:25cf:ff00::199]:65500 
returns nothing 

root@OpenMPTCProuter:~# multipath 
```
6in4-omr6in4 is deactivated
erspan0 is deactivated
eth0 is deactivated
eth1 is in default mode
eth2 is in default mode
gre0 is deactivated
gretap0 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
```
tcpdump -i any host 2a01:4f8:212:25cf:ff00::199
```
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
17:50:58.675665 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.37767 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1569893613, win 42700, options [mss 1220,sackOK,TS val 2391015466 ecr 0,nop,wscale 9], length 0
17:50:58.915654 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.64901 > 2a01:4f8:212:25cf:ff00::199.65280: Flags [S], seq 832328486, win 42700, options [mss 1220,sackOK,TS val 2391015706 ecr 0,nop,wscale 9,mptcp 4 capable v1], length 0
17:50:59.677810 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.37781 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1694646978, win 42700, options [mss 1220,sackOK,TS val 2391016468 ecr 0,nop,wscale 9], length 0
17:51:00.745671 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.37781 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1694646978, win 42700, options [mss 1220,sackOK,TS val 2391017536 ecr 0,nop,wscale 9], length 0
17:51:01.001495 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.37791 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1927437529, win 42700, options [mss 1220,sackOK,TS val 2391017791 ecr 0,nop,wscale 9], length 0
17:51:01.395674 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.57919 > 2a01:4f8:212:25cf:ff00::199.65280: Flags [S], seq 1759627579, win 42700, options [mss 1220,sackOK,TS val 2391018186 ecr 0,nop,wscale 9,mptcp 4 capable v1], length 0
17:51:01.747810 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.31735 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3609110699, win 42700, options [mss 1220,sackOK,TS val 2391018538 ecr 0,nop,wscale 9], length 0
17:51:02.025687 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.37791 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1927437529, win 42700, options [mss 1220,sackOK,TS val 2391018816 ecr 0,nop,wscale 9], length 0
17:51:02.027001 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan > 2a01:4f8:212:25cf:ff00::199: ICMP6, echo request, id 51194, seq 1, length 64
17:51:02.029258 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.31751 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2545369886, win 42700, options [mss 1220,sackOK,TS val 2391018819 ecr 0,nop,wscale 9], length 0
17:51:02.425666 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.57919 > 2a01:4f8:212:25cf:ff00::199.65280: Flags [S], seq 1759627579, win 42700, options [mss 1220,sackOK,TS val 2391019216 ecr 0,nop,wscale 9,mptcp 4 capable v1], length 0
17:51:02.835674 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.31735 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3609110699, win 42700, options [mss 1220,sackOK,TS val 2391019626 ecr 0,nop,wscale 9], length 0
17:51:03.065675 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.31751 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2545369886, win 42700, options [mss 1220,sackOK,TS val 2391019856 ecr 0,nop,wscale 9], length 0
17:51:06.010488 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.31765 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 321937217, win 42700, options [mss 1220,sackOK,TS val 2391022800 ecr 0,nop,wscale 9], length 0
17:51:07.075683 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.31765 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 321937217, win 42700, options [mss 1220,sackOK,TS val 2391023866 ecr 0,nop,wscale 9], length 0
^C
15 packets captured
16 packets received by filter
0 packets dropped by kernel

```
last lines of logred:
```
Jul  1 17:44:45 OpenMPTCProuter daemon.warn openvpn(omr)[12040]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jul  1 17:44:45 OpenMPTCProuter daemon.notice openvpn(omr)[12040]: TCP/UDP: Preserving recently used remote address: [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301
Jul  1 17:44:45 OpenMPTCProuter daemon.notice openvpn(omr)[12040]: Attempting to establish TCP connection with [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301
Jul  1 17:44:46 OpenMPTCProuter daemon.err openvpn(omr)[12040]: TCP: connect to [AF_INET6]2a01:4f8:212:25cf:ff00::199:65301 failed: Host is unreachable
Jul  1 17:44:46 OpenMPTCProuter daemon.notice openvpn(omr)[12040]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
[... cut ...]
Jul  1 17:45:51 OpenMPTCProuter daemon.notice openvpn(omr)[12040]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
[... 215 lines cut ...]
config rule 'eth1_fw_rule'
	option priority '1'
	option mark '0x45394'
	option lookup '4'

config rule 'eth2_fw_rule'
	option priority '1'
	option mark '0x45396'
	option lookup '6'

config rule 'tun0_fw_rule'
	option priority '1'
	option mark '0x45391500'
	option lookup '1500'

config rule '6in4_omr6in4_fw_rule'
	option priority '1'
	option mark '0x45391'
	option lookup '1'

config rule 'all_fw_rule'
	option priority '1'
	option mark '0x4539'
	option lookup '991337'
```

Ysurac, 2024-07-01:
Due to historical reason in OpenMPTCProuter multipath work by device (it was the case in 5.4, not more needed in 6.1 but still the case for now).
Without IPv4, omr-tracker can be set in IPv6 only.
Still not able to reach correctly the server, I will make some tests on my side.

Ysurac, 2024-07-01:
Can you try ```curl -s -k https://[2a01:4f8:212:25cf:ff00::199%eth1]:65500``` ? same with eth2 ?

rdmitry0911, 2024-07-01:
> Can you try `curl -s -k https://[2a01:4f8:212:25cf:ff00::199%eth1]:65500` ? same with eth2 ?

I did it and posted the results in my last message above. These commands produce nothing. All the traffic goes via 6in4 tunnel

rdmitry0911, 2024-07-01:
routing table looks like this:
ip -6 ro sh ta all
```
fd6d:d2f0:12d8::/60 dev eth0 table 69999 metric 1024 pref medium
default via fd6d:d2f0:12d8:: dev eth0 table 69999 metric 1024 pref medium
default from fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1 pref medium
unreachable 2a02:1388:2099:913f::/64 dev lo proto static metric 2147483647 pref medium
unreachable 2a0d:3344:1a87:cf10::/64 dev lo proto static metric 2147483647 pref medium
unreachable fd24:16af:d717::/64 dev lo proto static metric 2147483647 pref medium
fd6d:d2f0:12d8::/64 dev eth0 proto static metric 9999 pref medium
fd6d:d2f0:12d8::/60 dev eth0 metric 69999 pref medium
unreachable fd6d:d2f0:12d8::/48 dev lo proto static metric 2147483647 pref medium
unreachable fd6f:7e1:2556:10::/64 dev lo proto static metric 2147483647 pref medium
fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium
default via fd6d:d2f0:12d8:: dev eth0 metric 69999 pref medium
local ::1 dev lo table local proto kernel metric 0 pref medium
anycast 2a0d:3344:1a87:cf10:: dev eth1 table local proto kernel metric 0 pref medium
local 2a0d:3344:1a87:cf10::b44 dev eth1 table local proto kernel metric 0 pref medium
local 2a0d:3344:1a87:cf10:1ebf:ceff:fe6c:83c2 dev eth1 table local proto kernel metric 0 pref medium
anycast fd6d:d2f0:12d8:: dev eth0 table local proto kernel metric 0 pref medium
local fd6d:d2f0:12d8::1 dev eth0 table local proto kernel metric 0 pref medium
anycast fe80:: dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth2 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth1 table local proto kernel metric 0 pref medium
anycast fe80::a00:0 dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
local fe80::a00:2 dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
local fe80::aff:fc02 dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
local fe80::1ebf:ceff:fe6c:83c2 dev eth1 table local proto kernel metric 0 pref medium
local fe80::1ebf:ceff:fe6c:92ea dev eth2 table local proto kernel metric 0 pref medium
local fe80::be24:11ff:fe9c:80b dev eth0 table local proto kernel metric 0 pref medium
multicast ff00::/8 dev 6in4-omr6in4 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth0 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth2 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth1 table local proto kernel metric 256 pref medium

```

Ysurac, 2024-07-02:
I'm not able to find the result of the curl command, it's not the same as the version with --interface.

rdmitry0911, 2024-07-02:
> I'm not able to find the result of the curl command, it's not the same as the version with --interface.

I didn't realized that, checked right now, the results are empty strings. These comands produce nothing.
And also I see this:

root@OpenMPTCProuter:~# tcpdump -i any host 2a01:4f8:212:25cf:ff00::199 and port 65500
```
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
11:58:09.918481 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19827 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2870705765, win 42700, options [mss 1220,sackOK,TS val 1668859972 ecr 0,nop,wscale 9], length 0
11:58:10.934200 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19827 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2870705765, win 42700, options [mss 1220,sackOK,TS val 1668860988 ecr 0,nop,wscale 9], length 0
11:58:10.981936 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19833 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3227699731, win 42700, options [mss 1220,sackOK,TS val 1668861035 ecr 0,nop,wscale 9], length 0
11:58:11.936252 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19837 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3428566329, win 42700, options [mss 1220,sackOK,TS val 1668861990 ecr 0,nop,wscale 9], length 0
11:58:12.044216 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19833 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3227699731, win 42700, options [mss 1220,sackOK,TS val 1668862098 ecr 0,nop,wscale 9], length 0
11:58:12.047206 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19851 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2720543020, win 42700, options [mss 1220,sackOK,TS val 1668862101 ecr 0,nop,wscale 9], length 0
11:58:13.004218 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19837 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3428566329, win 42700, options [mss 1220,sackOK,TS val 1668863058 ecr 0,nop,wscale 9], length 0
11:58:13.084194 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19851 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 2720543020, win 42700, options [mss 1220,sackOK,TS val 1668863138 ecr 0,nop,wscale 9], length 0
11:58:14.006299 6in4-omr6in4 Out IP6 OpenMPTCProuter.lan.19859 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 1531559051, win 42700, options [mss 1220,sackOK,TS val 1668864060 ecr 0,nop,wscale 9], length 0
^C
9 packets captured
10 packets received by filter
0 packets dropped by kernel

```

Ysurac, 2024-07-02:
What is the result of ```ip r show table 3``` ?

rdmitry0911, 2024-07-02:
> What is the result of `ip r show table 3` ?

```
default via 192.168.1.1 dev eth1 
192.168.1.0/24 dev eth1 scope link 
```

Ysurac, 2024-07-02:
Sorry, in fact I want ```ip -6 r show table 63```

rdmitry0911, 2024-07-02:
Here are all the routing tables in the system:

root@OpenMPTCProuter:~# ip_all_tables 
```
Table: local (IPv4)
local 127.0.0.0/8 dev lo proto kernel scope host src 127.0.0.1 
local 127.0.0.1 dev lo proto kernel scope host src 127.0.0.1 
broadcast 127.255.255.255 dev lo proto kernel scope link src 127.0.0.1 
local 192.168.1.29 dev eth1 proto kernel scope host src 192.168.1.29 
broadcast 192.168.1.255 dev eth1 proto kernel scope link src 192.168.1.29 
local 192.168.10.199 dev eth2 proto kernel scope host src 192.168.10.199 
broadcast 192.168.10.255 dev eth2 proto kernel scope link src 192.168.10.199 
local 192.168.99.1 dev eth0 proto kernel scope host src 192.168.99.1 
broadcast 192.168.99.255 dev eth0 proto kernel scope link src 192.168.99.1 

Table: main (IPv4)
default via 192.168.1.1 dev eth1 metric 1 
default via 192.168.1.1 dev eth1 metric 3 
default via 192.168.10.1 dev eth2 metric 6 
default via 192.168.1.1 dev eth1 metric 8 
default via 192.168.10.1 dev eth2 metric 9 
127.0.0.0/8 dev lo proto static scope link metric 7 
192.168.1.0/24 dev eth1 scope link metric 3 
192.168.1.0/24 dev eth1 scope link metric 8 
192.168.10.0/24 dev eth2 scope link metric 6 
192.168.10.0/24 dev eth2 scope link metric 9 

Table: unspec (IPv4)
default via 192.168.1.1 dev eth1 table 3 
192.168.1.0/24 dev eth1 table 3 scope link 
default via 192.168.10.1 dev eth2 table 6 
192.168.10.0/24 dev eth2 table 6 scope link 
default via 192.168.1.1 dev eth1 table 8 
192.168.1.0/24 dev eth1 table 8 scope link 
default via 192.168.10.1 dev eth2 table 9 
192.168.10.0/24 dev eth2 table 9 scope link 
192.168.99.0/24 dev eth0 table lan proto static scope link metric 9999 
local default dev lo table 100 scope host 
default via 192.168.1.1 dev eth1 table 991337 
default via 192.168.1.1 dev eth1 metric 1 
default via 192.168.1.1 dev eth1 metric 3 
default via 192.168.10.1 dev eth2 metric 6 
default via 192.168.1.1 dev eth1 metric 8 
default via 192.168.10.1 dev eth2 metric 9 
127.0.0.0/8 dev lo proto static scope link metric 7 
192.168.1.0/24 dev eth1 scope link metric 3 
192.168.1.0/24 dev eth1 scope link metric 8 
192.168.10.0/24 dev eth2 scope link metric 6 
192.168.10.0/24 dev eth2 scope link metric 9 
local 127.0.0.0/8 dev lo table local proto kernel scope host src 127.0.0.1 
local 127.0.0.1 dev lo table local proto kernel scope host src 127.0.0.1 
broadcast 127.255.255.255 dev lo table local proto kernel scope link src 127.0.0.1 
local 192.168.1.29 dev eth1 table local proto kernel scope host src 192.168.1.29 
broadcast 192.168.1.255 dev eth1 table local proto kernel scope link src 192.168.1.29 
local 192.168.10.199 dev eth2 table local proto kernel scope host src 192.168.10.199 
broadcast 192.168.10.255 dev eth2 table local proto kernel scope link src 192.168.10.199 
local 192.168.99.1 dev eth0 table local proto kernel scope host src 192.168.99.1 
broadcast 192.168.99.255 dev eth0 table local proto kernel scope link src 192.168.99.1 
fd24:16af:d717::6da dev eth2 table 66 metric 1024 pref medium
fd24:16af:d717::/64 dev eth2 table 66 metric 1024 pref medium
default via fe80::7414:ceff:fea1:20e8 dev eth2 table 66 metric 1024 pref medium
2a02:1388:2143:e1e8::/64 dev eth2 table 69 metric 1024 pref medium
fd24:16af:d717::/64 dev eth2 table 69 metric 1024 pref medium
default via 2a02:1388:2143:e1e8:: dev eth2 table 69 metric 1024 pref medium
local default dev lo table 100 metric 1024 pref medium
fd6d:d2f0:12d8::/60 dev eth0 table 69999 metric 1024 pref medium
default via fd6d:d2f0:12d8:: dev eth0 table 69999 metric 1024 pref medium
default from fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1 pref medium
2a02:1388:2143:e1e8::/64 dev eth2 metric 69 pref medium
2a02:1388:2143:e1e8::/64 dev eth2 proto static metric 256 pref medium
[... 224 lines cut ...]
unreachable 2a02:1388:2143:e1e8::/64 dev lo proto static metric 2147483647 pref medium
unreachable 2a0d:3344:1a87:cf10::/64 dev lo proto static metric 2147483647 pref medium
fd24:16af:d717::/48 from 2a02:1388:2143:e1e8::/64 via fe80::7414:ceff:fea1:20e8 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::/48 from fd24:16af:d717::6da via fe80::7414:ceff:fea1:20e8 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::/48 from fd24:16af:d717::/64 via fe80::7414:ceff:fea1:20e8 dev eth2 proto static metric 512 pref medium
fd24:16af:d717::6da dev eth2 metric 66 pref medium
fd24:16af:d717::/64 dev eth2 metric 66 pref medium
fd24:16af:d717::/64 dev eth2 metric 69 pref medium
fd24:16af:d717::/64 dev eth2 proto static metric 256 pref medium
unreachable fd24:16af:d717::/64 dev lo proto static metric 2147483647 pref medium
fd6d:d2f0:12d8::/64 dev eth0 proto static metric 9999 pref medium
fd6d:d2f0:12d8::/60 dev eth0 metric 69999 pref medium
unreachable fd6d:d2f0:12d8::/48 dev lo proto static metric 2147483647 pref medium
unreachable fd6f:7e1:2556:10::/64 dev lo proto static metric 2147483647 pref medium
fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
fe80::a00:0/126 dev 6in4-omr6in4 proto static metric 1 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1 pref medium
default via fe80::7414:ceff:fea1:20e8 dev eth2 metric 66 pref medium
default via 2a02:1388:2143:e1e8:: dev eth2 metric 69 pref medium
default via fd6d:d2f0:12d8:: dev eth0 metric 69999 pref medium

```

Ysurac, 2024-07-02:
ok eth1 routes are totally missing, but at least eth2 should be working... Not sure the good route is used.
Can you do something like ```ip route get <vpsipv6> dev eth2``` ?

Ysurac, 2024-07-02:
This doesn't seems correct to me to be via fe80...
There is really too many global IPv6 on each interfaces... This seems to confuse routing part.
What is the result of ```ifstatus cosmote6``` ?

rdmitry0911, 2024-07-02:
> This doesn't seems correct to me to be via fe80... There is really too many global IPv6 on each interfaces... This seems to confuse routing part. What is the result of `ifstatus cosmote6` ?

ifstatus cosmote6 produces nothing
here is ifstatus wan2

```
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 5765,
	"l3_device": "eth2",
	"proto": "dhcpv6",
	"device": "eth2",
	"metric": 6,
	"dns_metric": 0,
	"delegation": false,
	"ipv4-address": [
		
	],
	"ipv6-address": [
		{
			"address": "fd24:16af:d717::6da",
			"mask": 128,
			"preferred": 37432,
			"valid": 37432
		},
		{
			"address": "fd24:16af:d717:0:1ebf:ceff:fe6c:92ea",
			"mask": 64
		},
		{
			"address": "2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea",
			"mask": 64
		}
	],
	"ipv6-prefix": [
		
	],
	"ipv6-prefix-assignment": [
		
	],
	"route": [
		{
			"target": "2a02:1388:2143:e1e8::",
			"mask": 64,
			"nexthop": "::",
			"metric": 256,
			"source": "::/0"
		},
		{
			"target": "fd24:16af:d717::",
			"mask": 64,
			"nexthop": "::",
			"metric": 256,
			"source": "::/0"
		},
		{
			"target": "fd24:16af:d717::",
			"mask": 48,
			"nexthop": "fe80::7414:ceff:fea1:20e8",
			"metric": 512,
			"valid": 1675,
			"source": "2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea/64"
		},
		{
			"target": "fd24:16af:d717::",
			"mask": 48,
			"nexthop": "fe80::7414:ceff:fea1:20e8",
			"metric": 512,
			"valid": 1675,
			"source": "fd24:16af:d717:0:1ebf:ceff:fe6c:92ea/64"
		},
		{
			"target": "fd24:16af:d717::",
			"mask": 48,
			"nexthop": "fe80::7414:ceff:fea1:20e8",
			"metric": 512,
			"valid": 1675,
			"source": "fd24:16af:d717::6da/128"
		}
	],
	"dns-server": [
		
	],
	"dns-search": [
		
	],
	"neighbors": [
		
	],
	"inactive": {
		"ipv4-address": [
			
		],
		"ipv6-address": [
			
		],
		"route": [
			{
				"target": "::",
				"mask": 0,
				"nexthop": "fe80::7414:ceff:fea1:20e8",
				"metric": 512,
				"valid": 1675,
				"source": "2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea/64"
			},
			{
				"target": "::",
				"mask": 0,
				"nexthop": "fe80::7414:ceff:fea1:20e8",
				"metric": 512,
				"valid": 1675,
				"source": "fd24:16af:d717:0:1ebf:ceff:fe6c:92ea/64"
			},
			{
				"target": "::",
				"mask": 0,
				"nexthop": "fe80::7414:ceff:fea1:20e8",
				"metric": 512,
				"valid": 1675,
				"source": "fd24:16af:d717::6da/128"
			}
		],
		"dns-server": [
			"fd24:16af:d717::1"
		],
		"dns-search": [
			
		],
		"neighbors": [
			
		]
	},
	"data": {
		"passthru": "00170010fd2416afd71700000000000000000001"
	}
}

```

Ysurac, 2024-07-02:
So it seems correct route is set.
There is something really strange in your config. What is exactly wan1 and wan2 ?

rdmitry0911, 2024-07-02:
> So it seems correct route is set. There is something really strange in your config. What is exactly wan1 and wan2 ?

I wouldn't say routing is correct as far as all the packets to vps go to 6in4 tunnel.. 
wan1/wan2 are just ethrnet ports connected to the routers. 

root@OpenMPTCProuter:~# ethtool eth1
```
Settings for eth1:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	                        2500baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: Symmetric
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: d
	Wake-on: d
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
```
root@OpenMPTCProuter:~# ethtool eth2
```
Settings for eth2:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	                        2500baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: d
	Wake-on: d
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
```

Ysurac, 2024-07-02:
I mean WAN2 routes are correct. All use 6in4 because connection to VPS is not really working, so no routes are set.
What are the routers connected and how are they connected ?
Can you ping fe80::7414:ceff:fea1:20e8 ?

rdmitry0911, 2024-07-03:
> I mean WAN2 routes are correct. All use 6in4 because connection to VPS is not really working, so no routes are set. What are the routers connected and how are they connected ? Can you ping fe80::7414:ceff:fea1:20e8 ?

One router, connected to eth2 is mofi router running openwrt 23.05.2. This router is under mine control. Another one is standard starlink router connected to eth1. I have no access to its firmware. 

With my previous config when only wan1/wan2 (ipv6) were set to support multipath and wan3/wan4 (ipv4) were set to not support multipath, only non-mptcp-able traffic went out directly via interfaces (wrong way) All other traffic went right routes and vps tunnel was working. Imho something's wrong in omr routing logic that prevents it to work with ipv6 only vps

ping fe80::7414:ceff:fea1:20e8
```
PING fe80::7414:ceff:fea1:20e8(fe80::7414:ceff:fea1:20e8) 56 data bytes
From fe80::be24:11ff:fe9c:80b%eth0 icmp_seq=1 Destination unreachable: Address unreachable
From fe80::be24:11ff:fe9c:80b%eth0 icmp_seq=2 Destination unreachable: Address unreachable
From fe80::be24:11ff:fe9c:80b%eth0 icmp_seq=5 Destination unreachable: Address unreachable
From fe80::be24:11ff:fe9c:80b%eth0 icmp_seq=8 Destination unreachable: Address unreachable
From fe80::be24:11ff:fe9c:80b%eth0 icmp_seq=9 Destination unreachable: Address unreachable
^C
--- fe80::7414:ceff:fea1:20e8 ping statistics ---
10 packets transmitted, 0 received, +5 errors, 100% packet loss, time 9316ms
```

Ysurac, 2024-07-03:
Possible that there is a problem with IPv6 routes, but for now I detected no problems.
Can you try a ```ping -B -I eth2 fe80::7414:ceff:fea1:20e8``` ? here it's using eth0 to exit.

rdmitry0911, 2024-07-03:
> Possible that there is a problem with IPv6 routes, but for now I detected no problems. Can you try a `ping -B -I eth2 fe80::7414:ceff:fea1:20e8` ? here it's using eth0 to exit.

root@OpenMPTCProuter:~# ping -B -I eth2 fe80::7414:ceff:fea1:20e8

```
ping: Warning: source address might be selected on device other than: eth2
PING fe80::7414:ceff:fea1:20e8(fe80::7414:ceff:fea1:20e8) from :: eth2: 56 data bytes
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=1 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=2 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=3 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=4 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=5 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=6 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=7 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=8 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=9 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=10 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:92ea%eth2 icmp_seq=11 Destination unreachable: Address unreachable
^C
--- fe80::7414:ceff:fea1:20e8 ping statistics ---
14 packets transmitted, 0 received, +11 errors, 100% packet loss, time 13476ms

```
root@OpenMPTCProuter:~# ping -B -I eth1 fe80::7414:ceff:fea1:20e8

```
ping: Warning: source address might be selected on device other than: eth1
PING fe80::7414:ceff:fea1:20e8(fe80::7414:ceff:fea1:20e8) from :: eth1: 56 data bytes
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=1 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=2 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=3 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=4 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=5 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=6 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=7 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=8 Destination unreachable: Address unreachable
From fe80::1ebf:ceff:fe6c:83c2%eth1 icmp_seq=9 Destination unreachable: Address unreachable
^C
--- fe80::7414:ceff:fea1:20e8 ping statistics ---
10 packets transmitted, 0 received, +9 errors, 100% packet loss, time 9350ms

```

Ysurac, 2024-07-03:
So not possible to ping the gateway, this explain why there is no traffic... There is something wrong with the data received via DHCPv6

rdmitry0911, 2024-07-03:
However pings to vps go thru:

root@OpenMPTCProuter:~# ping -B -I eth1 2a01:4f8:212:25cf:ff00::199
```
PING 2a01:4f8:212:25cf:ff00::199(2a01:4f8:212:25cf:ff00::199) from 2a0d:3344:1a87:cf10:1ebf:ceff:fe6c:83c2 eth1: 56 data bytes
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=1 ttl=55 time=54.4 ms
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=2 ttl=55 time=59.8 ms
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=3 ttl=55 time=65.3 ms
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=4 ttl=55 time=57.4 ms
^C
--- 2a01:4f8:212:25cf:ff00::199 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3003ms
rtt min/avg/max/mdev = 54.385/59.225/65.318/4.003 ms

```
root@OpenMPTCProuter:~# ping -B -I eth2 2a01:4f8:212:25cf:ff00::199
```
PING 2a01:4f8:212:25cf:ff00::199(2a01:4f8:212:25cf:ff00::199) from 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea eth2: 56 data bytes
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=1 ttl=52 time=68.0 ms
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=2 ttl=52 time=81.7 ms
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=3 ttl=52 time=83.7 ms
64 bytes from 2a01:4f8:212:25cf:ff00::199: icmp_seq=4 ttl=52 time=62.4 ms
^C
--- 2a01:4f8:212:25cf:ff00::199 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 62.363/73.934/83.672/8.992 ms

```

rdmitry0911, 2024-07-03:
> So not possible to ping the gateway, this explain why there is no traffic... There is something wrong with the data received via DHCPv6

It is possible to ping vps though

rdmitry0911, 2024-07-03:
My mofi router that provides internet to omr via eth2 has lan interface with the following ipv6 addresses:

root@OpenWrt-mofi:~# ifconfig 
```
br-lan    Link encap:Ethernet  HWaddr 8E:3E:4B:AB:9C:45  
          inet addr:192.168.10.1  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::8c3e:4bff:feab:9c45/64 Scope:Link
          inet6 addr: fd24:16af:d717::1/60 Scope:Global
          inet6 addr: 2a02:1388:2143:e1e8::1/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4083622 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3599748 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:948509960 (904.5 MiB)  TX bytes:2921114299 (2.7 GiB)

```
I can ping all of them from omr:

root@OpenMPTCProuter:~# ping -B -I eth2 2a02:1388:2143:e1e8::1
```
PING 2a02:1388:2143:e1e8::1(2a02:1388:2143:e1e8::1) from 2a02:1388:2143:e1e8::6da eth2: 56 data bytes
64 bytes from 2a02:1388:2143:e1e8::1: icmp_seq=1 ttl=64 time=0.950 ms
64 bytes from 2a02:1388:2143:e1e8::1: icmp_seq=2 ttl=64 time=0.843 ms
^C
--- 2a02:1388:2143:e1e8::1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1048ms
rtt min/avg/max/mdev = 0.843/0.896/0.950/0.053 ms

```
root@OpenMPTCProuter:~# ping -B -I eth2 fd24:16af:d717::1
```
PING fd24:16af:d717::1(fd24:16af:d717::1) from fd24:16af:d717::6da eth2: 56 data bytes
64 bytes from fd24:16af:d717::1: icmp_seq=1 ttl=64 time=0.831 ms
64 bytes from fd24:16af:d717::1: icmp_seq=2 ttl=64 time=0.800 ms
64 bytes from fd24:16af:d717::1: icmp_seq=3 ttl=64 time=1.36 ms
^C
--- fd24:16af:d717::1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2082ms
rtt min/avg/max/mdev = 0.800/0.995/1.356/0.255 ms

```
root@OpenMPTCProuter:~# ping -B -I eth2 fe80::8c3e:4bff:feab:9c45
```
ping: Warning: source address might be selected on device other than: eth2
PING fe80::8c3e:4bff:feab:9c45(fe80::8c3e:4bff:feab:9c45) from :: eth2: 56 data bytes
64 bytes from fe80::8c3e:4bff:feab:9c45%eth2: icmp_seq=1 ttl=64 time=0.772 ms
64 bytes from fe80::8c3e:4bff:feab:9c45%eth2: icmp_seq=2 ttl=64 time=0.495 ms
64 bytes from fe80::8c3e:4bff:feab:9c45%eth2: icmp_seq=3 ttl=64 time=0.783 ms
64 bytes from fe80::8c3e:4bff:feab:9c45%eth2: icmp_seq=4 ttl=64 time=0.801 ms
^C
--- fe80::8c3e:4bff:feab:9c45 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3097ms
rtt min/avg/max/mdev = 0.495/0.712/0.801/0.126 ms

```

rdmitry0911, 2024-07-03:
I've just checked and it turned out that ifstatus eth2 and if status eth1 shows different nexthops in comparison with what it was yesterday.
Now they are:
root@OpenMPTCProuter:~# ifstatus wan1
```
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 12570,
	"l3_device": "eth1",
	"proto": "dhcpv6",
	"device": "eth1",
	"metric": 3,
	"dns_metric": 0,
	"delegation": false,
	"ipv4-address": [
		
	],
	"ipv6-address": [
		{
			"address": "2a0d:3344:1a87:cf10::b44",
			"mask": 128,
			"preferred": 126,
			"valid": 276
		},
		{
			"address": "fd6f:7e1:2556:10::b44",
			"mask": 128
		},
		{
			"address": "2a0d:3344:1a87:cf10:1ebf:ceff:fe6c:83c2",
			"mask": 64,
			"preferred": 126,
			"valid": 7177
		},
		{
			"address": "fd6f:7e1:2556:10:1ebf:ceff:fe6c:83c2",
			"mask": 64
		}
	],
	"ipv6-prefix": [
		
	],
	"ipv6-prefix-assignment": [
		
	],
	"route": [
		{
			"target": "2a0d:3344:1a87:cf10::",
			"mask": 64,
			"nexthop": "::",
			"metric": 256,
			"valid": 7177,
			"source": "::/0"
		},
		{
			"target": "fd6f:7e1:2556:10::",
			"mask": 64,
			"nexthop": "::",
			"metric": 256,
			"source": "::/0"
		},
		{
			"target": "2a0d:3344:1a87:cf00::",
			"mask": 56,
			"nexthop": "fe80::7624:9fff:feb7:80cb",
			"metric": 512,
			"valid": 276,
[... 274 lines cut ...]
	},
	"data": {
		"passthru": "00170010fd2416afd71700000000000000000001"
	}
}
```
All nexthops are pingable via corresponding interfaces as well as vps via both interfaces.
However, vps tunnel does not exist and traffic to vps goes via 6in4

root@OpenMPTCProuter:~# tcpdump -i any host 2a01:4f8:212:25cf:ff00::199 and port 65500
```
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
10:09:19.668330 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.47007 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3088357221, win 42700, options [mss 1220,sackOK,TS val 3414298098 ecr 0,nop,wscale 9], length 0
10:09:20.670444 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.47023 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3375828759, win 42700, options [mss 1220,sackOK,TS val 3414299100 ecr 0,nop,wscale 9], length 0
10:09:21.748332 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.47023 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 3375828759, win 42700, options [mss 1220,sackOK,TS val 3414300178 ecr 0,nop,wscale 9], length 0
10:09:27.784659 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.52791 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 765059734, win 42700, options [mss 1220,sackOK,TS val 3414306214 ecr 0,nop,wscale 9], length 0
10:09:28.798329 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.52791 > 2a01:4f8:212:25cf:ff00::199.65500: Flags [S], seq 765059734, win 42700, options [mss 1220,sackOK,TS val 3414307228 ecr 0,nop,wscale 9], length 0
^C
5 packets captured
7 packets received by filter
0 packets dropped by kernel

```

Ysurac, 2024-07-03:
Still no result with ```curl -s -k https://[2a01:4f8:212:25cf:ff00::199%eth1]:65500``` ? 
tcpdump traffic is normal as the direct route to the VPS is not set. And default route to the VPS is not set if it doesn't answer to the previous curl command.

rdmitry0911, 2024-07-04:
> Still no result with `curl -s -k https://[2a01:4f8:212:25cf:ff00::199%eth1]:65500` ? tcpdump traffic is normal as the direct route to the VPS is not set. And default route to the VPS is not set if it doesn't answer to the previous curl command.

No. As far as curl traffic goes via 6in4-omr6in4 according to tcpdump, it can't reach vps. 
I made a clear test, to avoid interference with other traffic. I tried this curl command

curl -s -k https://[2a01:4f8:212:25cf:ff00::199%eth1]:65223

to check which route traffic of this command will use. As you can see it uses 6in4-omr6in4 which is wrong imho

root@OpenMPTCProuter:~# tcpdump -i any host 2a01:4f8:212:25cf:ff00::199 and port 65223
```
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
07:47:10.998150 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.29921 > 2a01:4f8:212:25cf:ff00::199.65223: Flags [S], seq 1197739813, win 42700, options [mss 1220,sackOK,TS val 3893644424 ecr 0,nop,wscale 9], length 0
07:47:12.030588 6in4-omr6in4 Out IP6 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea.29921 > 2a01:4f8:212:25cf:ff00::199.65223: Flags [S], seq 1197739813, win 42700, options [mss 1220,sackOK,TS val 3893645457 ecr 0,nop,wscale 9], length 0
```
root@OpenMPTCProuter:~# ping -B -I 6in4-omr6in4 2a01:4f8:212:25cf:ff00::199
```
ping: Warning: source address might be selected on device other than: 6in4-omr6in4
PING 2a01:4f8:212:25cf:ff00::199(2a01:4f8:212:25cf:ff00::199) from 2a02:1388:2143:e1e8:1ebf:ceff:fe6c:92ea 6in4-omr6in4: 56 data bytes
^C
--- 2a01:4f8:212:25cf:ff00::199 ping statistics ---
15 packets transmitted, 0 received, 100% packet loss, time 14588ms

```

Ysurac, 2024-07-04:
It should bind to the eth1 interface, so use direct route. You do tcpdump at the same time you check with the curl command ? And the port is 65500 not 65223

rdmitry0911, 2024-07-04:
> It should bind to the eth1 interface, so use direct route. 

It is not. The traffic in tcpdump occures only on 6in4-omr6in4 interface and nowhere else. It doesn't go via eth1. 

> You do tcpdump at the same time you check with the curl command ? And the port is 65500 not 65223

I made port 65223 to be sure, that I capture only the traffic from my curl command and nothing else.

rdmitry0911, 2024-07-04:
> It should bind to the eth1 interface, so use direct route.

I think this is the problem. 6in4-omr6in4 interface is of type 6in4 which means a tunnel with encapsulation of ipv6 traffic into ipv4 packets at the entrance and decapsulation at the exit. In our case the entrance point is 10.255.252.2 and the exit point is 10.255.252.1 which is located in vps. So, to make this 6in4 tunnel work a tunnel between omr and vps should exist in advance. However the logic of this omr-vps tunnel creation includes a working 6in4 tunnel which creates a cycle. 6in4 tunnel can't start, as it needs omr-vps tunnel and omr-vps tunnel can't start as it needs 6in4 tunnel

Ysurac, 2024-07-04:
I just put a change to start the tunnel only when then VPN is UP. It's compiling, we will see if this change something or not.

rdmitry0911, 2024-07-04:
> I just put a change to start the tunnel only when then VPN is UP. It's compiling, we will see if this change something or not.

I think it might not be enough. We don't need 6in4 tunnel at all if vps has ipv6 only public ip.

Ysurac, 2024-07-04:
We need it as the VPN is IPv4 only.

[33 more replies]

closed 2024-10-31 by github-actions[bot]