Ysurac/openmptcprouter #3428: Suggestion for a modem

YB1200, 2024-06-27:
Hello Ysurac,
Can you recommend a 5G modem that can be used with openMPTCProuter?

Ysurac, 2024-06-27:
USB key or Mini PCIe format ?
For USB key, I like Huawei Hi-Link, because it's see a an ethernet interface, so it just work.
For Mini PCIe, Quectel is what I'm using.

YB1200, 2024-06-27:
Yes, Mini PCIe format.
Can you give me the reference of your Quectel ?

YB1200, 2024-07-05:
Hello Ysurac,
Just out of curiosity, how do you configure your 4G Quectel modem with OMR? Are there any instructions for that?

Ysurac, 2024-07-05:
I use ModemManager and it's should just work. You may have to set APN and PIN if there is any on your SIM card.
eSIM are not supported for now.

FabrizioBocci, 2024-07-18:
@Ysurac could you tell me which Huawei HiLink model is the USB 5G modem?