Ysurac/openmptcprouter #3429: ZBT-WG1602 V04 16m Support Request

Jarkaruss, 2024-06-27:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Hi I've been trying to build openmptcprouter for my ZBT WG-1602 V04 16m router and it's not going well. Any chance of adding it to the supported machine to do list? This should benefit Zephyr because this router features dual sim card slots and should be the perfect platform for openmptcprouter to be contained within a single hardware unit. It's also very affordable for R and D purposes.

**Describe why you are asking for this support?**

I'm struggling to achieve it myself!

**Additional context**

https://www.zbtwifi.com/productinfo/1064441.html

Model: WG1602

CPU: Dual core, 880GHz Frequency

RAM: 256MB (MAX 512MB)

FLASH:16MB (MAX 32MB)

Software: Openwrt & ZBT OS

Ysurac, 2024-06-27:
I only support hardware that I own.
This device have a small flash (and quite slow CPU).
You followed https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform ? This device is supported by OpenWRT so it should also work with OpenMPTCProuter (you will need to remove some default packages due to flash size... 16MB will not be possible I think but should be ok for 32MB version...)

Jarkaruss, 2024-06-27:
Thank you for the speedy response!

Yes I followed that link and I'm currently trying to follow that with the edited kernel settings but I'm guessing it's going to come out too big as I've done it so will try to remove some packages. 

Also thank you for the advice regarding the CPU. We'll test this and consider something faster!