Ysurac/openmptcprouter #3430: Iperf UDP using MPTCP

Raquel1L, 2024-06-28:
## Expected Behavior

I want to test the network performance from my laptop to a Google Cloud VM (where the openMPTCProuter VPS is installed) using an RPI 4 (where the openMPTCProuter router is installed) as a router.

To do this, I ran the Iperf UDP client on the laptop and the iperf server on the VM and, in parallel, did a tcpdump on the VM. So the idea is to generate traffic from the laptop to the VM.

In short, the command I ran on the laptop was: 
        iperf -c <server_ip> -u -b 5M -t 10

and on the GC VM I ran
	iperf -s -u
	sudo tcpdump -i any -w packets.pcap port 5001

What I expected to see in the capture I made was only the MPTCP protocol.

## Current Behavior
However, the capture shows UDP packets, which means that MPTCP didn't encapsulate the traffic coming from iperf UDP, as I thought it would.

[image]

What could be wrong for this to happen? How can I get MPTCP to work with Iperf?

## Specifications

  - OpenMPTCProuter version: v0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: RPI4

Ysurac, 2024-06-28:
MPTCP is only for TCP, to aggregate UDP a TCP VPN is used.
Server IP have a direct route from the router so direct UDP traffic between router and server is not aggregated (UDP traffic from client use the VPN, so is aggregated).

Raquel1L, 2024-06-30:
Thank you very much for your reply! 

I just tried Iperf TCP and the only protocol that appears in the packet capture made in the VM is TCP (applying the filter for packets only related to iperf). My question now is: shouldn't it be MPTCP?

This time, in short, the command I ran on the laptop was:
iperf -c <server_ip> -p 65000 -b 5M -t 10

and on the GC VM I ran
iperf -s -p 65000
sudo tcpdump -i any -w packets.pcap port 65000

[image]

closed 2024-10-04 by github-actions[bot]