Ysurac/openmptcprouter #3431: MPTCP connection becomes stale

mario-minati, 2024-06-28:
## Expected Behavior
We would expect that both OMR sides could run for a long time without problems.

## Current Behavior
We discovered problems with the VPS part of OMR after running for about 14 days.

The problem occured as a massiv speed degradation and failure of most reuqest which were routed to the VPS, 
Those packages which were bypassed through OMR-bypass settings had no problems.

The graphs of our VPS look strange:
[image]

There are no problems with free disk space, but there are hundreds and hundreds processes like this:
```
root     1952539  0.0  0.0   2596   108 ?        S    May22   0:00  \_ /bin/sh -c ss -M | grep 195.50.149.xxx
root     1952540  0.0  0.0   6632   612 ?        S    May22   0:00  |   \_ ss -M
root     1952541  0.0  0.0   6444   144 ?        S    May22   0:00  |   \_ grep 195.50.149.xxx
```
Every day has around 20 processes like this. 195.50.149.xxx is the fixed ip of our master internet connection.
This includes every single day since the last reboot.

If helpfull we can post the complete output of `ps aufx`.

After a reboot everything was working smoothly again.

## Possible Solution
Cleanup of old sh processes?

## Steps to Reproduce the Problem

  1. Run OMR VPS for multipe days.

## Context (Environment)
Our network overview:
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

Ysurac, 2024-06-29:
Strange, I don't have this issue on my servers.
What is the result of  ```ss -M | grep 195.50.149.xxx``` if you run it manually ?
I added a timeout as a workaround solution for now in snapshot script ( https://github.com/Ysurac/openmptcprouter/wiki/Snapshots )

mario-minati, 2024-06-29:
The output doesn't show any obvious problems:
https://pastebin.com/zj718dZh

I'll do an update later this weekend. Thanks again for the quick reply.

mario-minati, 2024-06-29:
I just saw that these shells are created by the omr-admin script:

```
root         827  0.1  4.5 195768 90108 ?        Ssl  Jun27   3:50 python3 /usr/local/bin/omr-admin.py --host=::
root         828  0.2  7.9 333972 156572 ?       Ssl  Jun27   6:33 python3 /usr/local/bin/omr-admin.py
root       80479  0.0  0.0   2596   932 ?        S    Jun27   0:00  \_ /bin/sh -c ss -M | grep 195.50.149.178
root       80481  0.0  0.1   6444  2024 ?        S    Jun27   0:00  |   \_ grep 195.50.149.178
root      295841  0.0  0.0   2596   944 ?        S    Jun27   0:00  \_ /bin/sh -c ss -M | grep 195.50.149.178
root      295842  0.0  0.1   6632  3424 ?        S    Jun27   0:00  |   \_ ss -M
root      295843  0.0  0.1   6444  2008 ?        S    Jun27   0:00  |   \_ grep 195.50.149.178
```

Maybe this is related to having the OMR web admin always open in one tab of my browser and showing the status page which is always refreshing.

mario-minati, 2024-07-03:
I updated our system with your timeout change https://github.com/Ysurac/openmptcprouter-vps-admin/commit/1e36d2aa4be71fd348f954cbe4a4dcaf906ba1f4

After a reboot theses stale processes still pile up:

```
root         848  0.1  4.5 195772 90108 ?        Ssl  Jul02   0:49 python3 /usr/local/bin/omr-admin.py --host=::
root         857  0.2  6.3 305184 124184 ?       Ssl  Jul02   1:29 python3 /usr/local/bin/omr-admin.py
root      185600  0.0  0.0   2596   940 ?        S    Jul02   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root      185602  0.0  0.1   6444  2012 ?        S    Jul02   0:00  |   \_ grep 195.50.149.xxx
root      400925  0.0  0.0   2596   940 ?        S    Jul02   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root      400927  0.0  0.0   6444  1952 ?        S    Jul02   0:00  |   \_ grep 195.50.149.xxx
root      617621  0.0  0.0   2596   924 ?        S    Jul02   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root      617623  0.0  0.1   6444  2016 ?        S    Jul02   0:00  |   \_ grep 195.50.149.xxx
root      834113  0.0  0.0   2596   920 ?        S    Jul02   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root      834115  0.0  0.1   6444  1980 ?        S    Jul02   0:00  |   \_ grep 195.50.149.xxx
root     1049497  0.0  0.0   2596   936 ?        S    00:19   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root     1049499  0.0  0.1   6444  2012 ?        S    00:19   0:00  |   \_ grep 195.50.149.xxx
root     1264807  0.0  0.0   2596   908 ?        S    01:19   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root     1264809  0.0  0.1   6444  1968 ?        S    01:19   0:00  |   \_ grep 195.50.149.xxx
root     1479438  0.0  0.0   2596   912 ?        S    02:18   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root     1479440  0.0  0.1   6444  2068 ?        S    02:18   0:00  |   \_ grep 195.50.149.xxx
root     1694578  0.0  0.0   2596   900 ?        S    03:18   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root     1694580  0.0  0.1   6444  2016 ?        S    03:18   0:00  |   \_ grep 195.50.149.xxx
root     1910575  0.0  0.0   2596   952 ?        S    04:18   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root     1910577  0.0  0.1   6444  2008 ?        S    04:18   0:00  |   \_ grep 195.50.149.xxx
root     2125877  0.0  0.0   2596   896 ?        S    05:18   0:00  \_ /bin/sh -c timeout 2 ss -M | grep 195.50.149.xxx
root     2125879  0.0  0.1   6444  2164 ?        S    05:18   0:00  |   \_ grep 195.50.149.xxx
root     2250696  0.0  0.0      0     0 ?        Z    05:52   0:00  \_ [sh] <defunct>
```

What else can we debug?

mario-minati, 2024-07-03:
After digging a little bit I came up this post about problems with os.popen:
https://thraxil.org/users/anders/posts/2008/03/13/Subprocess-Hanging-PIPE-is-your-enemy/

According this there is a limit of 65536 characters with that command. Looking into the results of the ss command that would match with our problem. The output has 178kB.

What do you think?

Ysurac, 2024-07-05:
Can you try latest script ? Still using PIPE but with a grep -q and a kill. May solve or not the issue...

closed 2024-10-09 by github-actions[bot]