Ysurac/openmptcprouter #3432: WAN5 not passing any traffic at all.

mattfaraday, 2024-07-01:
## Expected Behavior
Adding a new WAN interface with DHCPv4 I would expect that traffic would be passed down this interface as well and it would add to the aggregate speed. 

## Current Behavior
The interface is not being used for anything, the speed shows as kilobytes (It's capable of over 200mbit when I speed test the link directly - as in not via OMR - but OMR just seems to ignore this link. In the system log I see 

Jul  1 22:02:31 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan5 (eth5) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 192.168.130.52 error (114.114.114.114,1.1.1.1,4.2.2.1)
Jul  1 22:02:31 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 198.244.188.165 dev eth5

I don't understand the problem with IPv6 because the ISP does not give an IPv6 IP only IPv4 and this WAN5 interface is configured as DHCPv4 (I have to, i don't know what IP it will get). The interface seems to me to be configured in the same way the other WAN interfaces are configured

config interface 'wan5'
        option proto 'dhcp'
        option device 'eth5'
        option multipath 'on'
        option addlatency '0'
        option metric '14'
        option ipv6 '0'
        option defaultroute '0'
        option peerdns '0'
        option delegate '0'

## Steps to Reproduce the Problem

  1. Install OMR. 
  2. Add a WAN5 with DHCP , Enable Multipath TCP on it 
  3. Test speed. Observe no data sent on WAN5 

## Context (Environment)
I would like to use WAN5. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60-6.1 r0+24843-acf40
  - OpenMPTCProuter VPS version: Version：6.1.0-18-cloud-amd64 0.1030
  - OpenMPTCProuter VPS provider: OVH  (UK) 
  - OpenMPTCProuter platform: x86_64 - Virtualised - Proxmox. 
  - Country: UK

Thank you for any assistance.

Ysurac, 2024-07-02:
What is the result of ```ping -B -I eth5 1.1.1.1``` ?

mattfaraday, 2024-07-03:
Hi. 
As requested:

root@OpenMPTCProuter:~# ping -B -I eth5 1.1.1.1
PING 1.1.1.1 (1.1.1.1) from 192.168.10.101 eth5: 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=55 time=18.9 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=55 time=15.5 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=55 time=18.9 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=55 time=18.0 ms

I'm going to be re-installing it either today or tomorrow so I will update if it's still an issue.

mattfaraday, 2024-07-03:
Hi

I re-installed just now. 
WAN 5 still not passing any traffic, on a newly re-installed openmptcprouter and also a new VPS too. 

[image]

Ysurac, 2024-07-04:
Seems that there is problems on this interface: Link connectivity loss lean that the cable is removed from the interface.
As it's a proxmox, check on it what can be the problem.

mattfaraday, 2024-07-04:
Hi

The proxmox interface is configured the same as the others, i think those logs are because i had just booted it perhaps. 
but even if i create a new interface, it's the same problem. The proxmox server has a LACP trunk and all the interfaces are actually VLANs.

[image]

mattfaraday, 2024-07-06:
ok i have some more time to mess with this, it looks like the tracker is marking the interface as down. 
i have also noticed with this particular ISP that connections take a few seconds to actually work on it. like if you go to whatismyip.com it will take it about 5 or 6 seconds to actually start working. 

is there a way to not use the tracker on this interface and to mark it as always up ?

mattfaraday, 2024-07-06:
Progress! 

I swapped the SIM in the 5G modem and it works just fine now .

[image]

So this is a problem with Vodafone UK 5G. I put a different SIM in and bam , it works fine. 
But now I don't know why it doesn't work with Vodafone. I thought maybe they don't like MLVPN traffic, so I turned on the option to do it over VPN but then it just says the interface is down permanently with this option.

Ysurac, 2024-07-06:
I also have some strange result with Vodafone, sometimes it's working correclty, and sometimes not at all...
The modem is an external modem or a USB/PCIE with ModemManager ?

mattfaraday, 2024-07-06:
external. Huawei 5G CPE Pro 2 
i guess we can probably consider this closed then, I can just not use Vodafone as its not a contract but it's a shame because here it does 300+ mbps. Thank you for all your help

closed 2024-07-06 by mattfaraday