Ysurac/openmptcprouter #3433: MPTCP aggregation does not occur when using iperf

Raquel1L, 2024-07-02:
## Expected Behavior
I've used Iperf with TCP, and the VM's packet capture shows only TCP packets. Shouldn't it be showing MPTCP?

The command I ran on the laptop was:
iperf -c <server_ip> -p 59990 -b 5M -t 10

and on the GC VM I ran
iperf -s -p 59990
sudo tcpdump -i any -w packets.pcap

## Current Behavior
How can I use Iperf from a laptop to the VM with MPTCP working as expected?

[image]

## Specifications

  - OpenMPTCProuter version: v0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: RPI4

Ysurac, 2024-07-02:
mptcp is not enabled by default on iperf you need to use the "--mptcp" flag. Not sure if this was available on v0.59.1

taylord0ng, 2024-07-04:
> mptcp is not enabled by default on iperf you need to use the "--mptcp" flag. Not sure if this was available on v0.59.1

when use "--mptcp" flag return "iperf3: unrecognized option: mptcp", this way is not correct?
any i used "--multipath" flag, the traffic also transfer on one ethernet
OpenMPTCProuter version: v0.60-6.1 x86-64
[image]

Ysurac, 2024-07-04:
You need to have the flag enabled on both side.

closed 2024-10-08 by github-actions[bot]