Ysurac/openmptcprouter #3435: RPI4B is in Read-Only mode, anyway to fix it ?

viorizz, 2024-07-02:
## Expected Behavior

RPI4B is supposed to run normally without read-only on the FS

## Current Behavior

RPI4B is running with FS in read-only :

[image]

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version:   v0.1030 6.1.0-21-cloud-amd64
  - OpenMPTCProuter VPS provider: Infomaniak
  - OpenMPTCProuter platform:  RPI4B

## Explanations
I think it might have happend because I unplugged the board. But now I can't make any modifications to the router. And I don't understand why that's happening. If I have to reinstall the whole router. Is there a way I can keep those config file in some way ?

I don't really want to go and redo all my config.

Thanks.

fareign, 2024-07-03:
use squashfs-factory firmware

w453y, 2024-08-08:
That happens to me too, when I unplug it directly. But it can be fixed by running the following command.
SSH into the router.
`e2fsck /dev/mmcblk0p2`
then reboot it from the terminal, which will solve the issue.

closed 2024-11-11 by github-actions[bot]