Ysurac/openmptcprouter #3436: TCP and UDP Traffic Management

YB1200, 2024-07-03:
Will the aggregated traffic (TCP & UDP) be transferred by the tunnel created by the default VPN, or will TCP be transferred through the tunnel created for Shadowsocks and UDP through the tunnel created by the default VPN?
Because as I noticed is the creation of gt-tun0 with the IP address 10.255.255.1 and it's for the default VPN.

Ysurac, 2024-07-03:
By default, TCP use the proxy (ShadowSocks-Rust in latest release) and everything else use the VPN (OpenVPN TCP in latest release).
gt-tun0 is Glorytun TCP on VPS side, all VPNs server are created and have interfaces. It was the default VPN in previous releases.

YB1200, 2024-07-04:
Thanks for your reply. Regarding the hardware, the QNAP QHora-301W router seems powerful in terms of CPU. What is the maximum bandwidth that it can aggregate? Additionally, does the Wi-Fi module work with OMR? Would you recommend it?

Ysurac, 2024-07-05:
No idea, I didn't make much tests on it, and it seems there is issues with 10G interfaces from others issues.

closed 2024-10-09 by github-actions[bot]