Ysurac/openmptcprouter #3437: What's difference between Shadowsocks and Shadowsocks-Rust 2022

longxi0412, 2024-07-04:
What's difference between Shadowsocks and Shadowsocks-Rust 2022
VPS Setting/Advanced Setting/Proxy Setting/
Shadowsocks can visit internet(it can't visit the internet if i chane the Shadowsock to Shadowsocks-Rust 2022)

[image]
[image]

Ysurac, 2024-07-04:
Shadowsocks-Rust use new Shadowsocks 2022 protocol.
It should work, direct output mean an IP from one of  your connection is used.
You can also try Snapshot 0.61 that may fix (or not) the issue.

longxi0412, 2024-07-05:
> Shadowsocks-Rust use new Shadowsocks 2022 protocol. It should work, direct output mean an IP from one of your connection is used. You can also try Snapshot 0.61 that may fix (or not) the issue.

Thanks for your reply,

mrkaffeine92, 2024-07-17:
> Shadowsocks-Rust use new Shadowsocks 2022 protocol. It should work, direct output mean an IP from one of your connection is used. You can also try Snapshot 0.61 that may fix (or not) the issue.

I made the decision to try 0.61 and am very pleased with it so far.

closed 2024-10-21 by github-actions[bot]