Ysurac/openmptcprouter #3439: Set MPTCP master last

dfadev, 2024-07-05:
## Expected Behavior
MPTCP master must be set last so subflows are created prior to master being set.

## Current Behavior

```
Jul  5 00:31:25 router user.notice MPTCP: Set modem0 to on
Jul  5 00:31:25 router user.notice MPTCP: Set modem1 to on
Jul  5 00:31:25 router user.notice MPTCP: Set eth2 to master  <--- this must happen last
Jul  5 00:31:25 router user.notice MPTCP: Set eth3.100 to on
Jul  5 00:31:25 router user.notice MPTCP: Set eth3.200 to on
```

In this scenario it's possible `eth3.100` and `eth3.200` won't be used for MPTCP.

## Possible Solution

Change scripts to set the master last.

## Steps to Reproduce the Problem

Unfortunately I don't have a good repro other than keep restarting an interface until an existing TCP socket stops using MPTCP on the interface that was restarted.  Rebooting a modem triggers it sometimes.  Then try this to fix:
```
/etc/init.d/mptcp reload <broken interface>
/etc/init.d/mptcp reload <master interface>
```

Existing TCP sockets should then start using the previously broken interface.

## Context (Environment)

Long running TCP sockets, then a modem reboots and the interface comes back up but MPTCP no longer makes use of it for currently connected TCP sockets even if the interface is set to on.  New TCP sockets will use it though.  Modems are connected via VLAN through a switch.

## Specifications

  - OpenMPTCProuter version: v0.60rc2-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 6.1.0-21-amd64 0.1031-test
  - OpenMPTCProuter VPS provider: madcityservers
  - OpenMPTCProuter platform: x86_64
  - Country: US

Ysurac, 2024-07-05:
I don't really understand what you mean, "master" mode is only to set an interface as default gateway and mptcp init script do exactly same configuration with interfaces defined as "master" or as "on".
In the future, master interface will be remove to simply set any interface as default gateway even if multipath is not enabled on it.

dfadev, 2024-07-05:
You're right I must be seeing something else.  Thanks for the sanity check :)

closed 2024-07-05 by dfadev