Ysurac/openmptcprouter #3440: Are VPSs with Ubuntu still compatible with the snapshot?

vempire-ghost, 2024-07-05:
I have one of my VPSs running Ubuntu and after updating to the latest OMR script using the command `wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="5.4" sh`, OMR works normally, but these two screens stopped showing data.
[image]
[image]

OMR is running the snapshot from today, 04/07.

Ysurac, 2024-07-05:
VPS script is not tested with Ubuntu anymore and even less with 5.4 kernel.
You can check that VPS API is working: go to https://<yourvpsip>:65500/ and you should get, after accepting the self signed certificate, a welcome message. Or via SSH on the router: ```curl -k https://<yourvpsip>:65500/```

Ysurac, 2024-07-05:
5.4 will still work with 0.61, but it's a legacy version, OpenWRT packages and kernel are not updated.
omr-vps-admin should still work with 5.4
It's more Ubuntu that is not tested at all ;)

vempire-ghost, 2024-07-05:
When I run the command through the browser, this message is displayed: "Welcome to OpenMPTCProuter Server part."
This is the only VPS I am using with Ubuntu because it is somewhat troublesome to put Debian on an Oracle Cloud VPS, but as soon as possible, I will be migrating to Debian on this VPS.

When I tried to update, I got this error: "failed: Temporary failure in name resolution." I had to edit the /etc/systemd/resolved.conf file and add a DNS to get it working again. Could this have something to do with it?

Ysurac, 2024-07-05:
No if you run again the script after the error.
You can check using ```journalctl +u omr-admin``` if there is any error on VPS side, if not the problem is on router side.

vempire-ghost, 2024-07-05:
This is the log from the last execution of the omr admin.

Jul 05 03:01:55 instance-2 systemd[1]: Started OMR-Admin.
Jul 05 03:03:15 instance-2 omr-admin.py[608]: ERROR:    Exception in ASGI application
Jul 05 03:03:15 instance-2 omr-admin.py[608]: Traceback (most recent call last):
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/uvicorn/protocols/http/httptools_impl.py", line 399, in run_asgi
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     result = await app(  # type: ignore[func-returns-value]
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/uvicorn/middleware/proxy_headers.py", line 70, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     return await self.app(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/fastapi/applications.py", line 1054, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await super().__call__(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/applications.py", line 123, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await self.middleware_stack(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/middleware/errors.py", line 186, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     raise exc
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/middleware/errors.py", line 164, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await self.app(scope, receive, _send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/middleware/exceptions.py", line 65, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/_exception_handler.py", line 64, in wrapped_app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     raise exc
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/_exception_handler.py", line 53, in wrapped_app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await app(scope, receive, sender)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/routing.py", line 756, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await self.middleware_stack(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/routing.py", line 776, in app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await route.handle(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/routing.py", line 297, in handle
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await self.app(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/routing.py", line 77, in app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await wrap_app_handling_exceptions(app, request)(scope, receive, send)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/_exception_handler.py", line 64, in wrapped_app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     raise exc
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/_exception_handler.py", line 53, in wrapped_app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     await app(scope, receive, sender)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/starlette/routing.py", line 72, in app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     response = await func(request)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/fastapi/routing.py", line 278, in app
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     raw_response = await run_endpoint_function(
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/fastapi/routing.py", line 191, in run_endpoint_function
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     return await dependant.call(**values)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/bin/omr-admin.py", line 1374, in status
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     vps_cpu_freq = psutil.cpu_freq().current
Jul 05 03:03:15 instance-2 omr-admin.py[608]: AttributeError: 'NoneType' object has no attribute 'current'
Jul 05 03:03:15 instance-2 omr-admin.py[608]: ERROR:    Exception in ASGI application
Jul 05 03:03:15 instance-2 omr-admin.py[608]: Traceback (most recent call last):
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/uvicorn/protocols/http/httptools_impl.py", line 399, in run_asgi
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     result = await app(  # type: ignore[func-returns-value]
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/uvicorn/middleware/proxy_headers.py", line 70, in __call__
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     return await self.app(scope, receive, send)
[... cut ...]
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/local/lib/python3.8/dist-packages/fastapi/routing.py", line 191, in run_endpoint_function
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     return await dependant.call(**values)
Jul 05 03:03:15 instance-2 omr-admin.py[608]:   File "/usr/bin/omr-admin.py", line 1374, in status
Jul 05 03:03:15 instance-2 omr-admin.py[608]:     vps_cpu_freq = psutil.cpu_freq().current
Jul 05 03:03:15 instance-2 omr-admin.py[608]: AttributeError: 'NoneType' object has no attribute 'current'

Ysurac, 2024-07-05:
It's a bug due to an older version of psutil on your install. I fixed it in the omr-admin script. You can run the VPS install script again to update and this should fix your issue.

vempire-ghost, 2024-07-06:
I ran the script and now it worked perfectly, thank you.

closed 2024-07-06 by vempire-ghost