Ysurac/openmptcprouter #3441: Error in the MPTCP tab of luci on OMR 6.6

vempire-ghost, 2024-07-05:
## Expected Behavior
Luci MPTCP tab works

## Current Behavior
Show a runtime error

## Context (Environment)
I am starting to test snapshot 6.6 now that it has a redundant scheduler. I downloaded the most recent version, but when I run it and access the address http://192.168.100.1/cgi-bin/luci/admin/network/mptcp, only an error is shown.

Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/model/cbi/mptcp.lua:72: bad argument #1 to 'gmatch' (string expected, got nil)

In error(), file [C]
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
  called from function  ((tail call))

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:794:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:795:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:831:78)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-snapshot-6.6 r0+26695-8
  - OpenMPTCProuter VPS version: 1031-teste
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-07-06:
Refresh the page and this will work.

vempire-ghost, 2024-07-06:
I tried to update normally and with ctrl pressed, clear the cache, and use private mode in both Edge and Firefox, but the page did not load. I will wait for your correction.

vempire-ghost, 2024-08-06:
Hello Ysurac, I still can't access the MPTCP configuration page in snapshot 6.6.

Ysurac, 2024-08-06:
There is a bug, but a browser refresh always worked for me.

vempire-ghost, 2024-08-06:
Refreshing the page sometimes changes the error but doesn't resolve it here. Another thing is that in snapshot 6.6, the new ping configuration option for the omr-tracker server is not appearing. Is this snapshot being updated?

closed 2024-11-10 by github-actions[bot]