Ysurac/openmptcprouter #3443: Website rejects connection on first load

Pixelramp, 2024-07-09:
## Expected Behavior
- Website should load without any issues

## Current Behavior
- Many websites (not all) are rejecting the connection. After refreshing the website, it loads without any issues.
- Some images are not being loaded
## Specifications

  - OpenMPTCProuter version: v0.61-snapshot-6.6 r0+26695-820823198d
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-18-cloud-amd64
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: N100

What i've tried:
 Changing DNS to Google and Cloudflare

Ysurac, 2024-07-09:
What is the proxy used ?

Pixelramp, 2024-07-09:
> What is the proxy used ?

Shadowsocks-Rust 2022

Ysurac, 2024-07-09:
I just fixed an issue on firewall rules for Shadowsocks-Rust, this may be the problem.
A new image is compiling.

Pixelramp, 2024-07-09:
I switched to only using Shadowsocks, and the problem is no longer present. Are there any disadvantages to using Shadowsocks, or can I just continue using it?

I'll try out the updated image later.

> I just fixed an issue on firewall rules for Shadowsocks-Rust, this may be the problem. A new image is compiling.

Ysurac, 2024-07-09:
Shadowsocks-Rust use latest Shadowsocks protocol (more secure) and is multi threated (so far better if you have more than one core available). It also give a better speed in most case.

Pixelramp, 2024-07-10:
Updated to the new image, seems to be fixed. I'll report back if there is any problerm remaining. Thanks for the quick fix!

closed 2024-07-10 by Pixelramp