Ysurac/openmptcprouter #3444: Help understanding proxy type and variation

Brazzo978, 2024-07-09:
Hi in 0.60 a lot of proxy option were added , can you please provide some explaination on what is the difference within them , i see that there is shadowsock and its rust variant , then V2ray and Xray and their variant , but searching on the net i cant find most of those protocols Thanks 

## Specifications

  - OpenMPTCProuter version: 0.61 Snapshot
  - OpenMPTCProuter VPS version: 1031-test
  - OpenMPTCProuter VPS provider: local
  - OpenMPTCProuter platform: vmware

Ysurac, 2024-07-15:
Shadowsocks-libev use Shadowsocks protocol, it's no more in development.
Shadowsocks-Rust use Shadowsocks 2022 protocol dev is active and it support multi thread. Seems to give better speed than Shadowsocks libev. it's the default since v0.60

V2Ray (https://www.v2fly.org/en_US/) and XRay (https://xtls.github.io/en/) support multiples protocols. XRay Vless Reality use port 443 and should be able to bypass some firewalls. Development seems more active on XRay, it's why both are available for now. V2Ray and XRay support transport of UDP over TCP and can be used to redirect ports instead of the VPN.

Brazzo978, 2024-07-16:
So correct me if i am wrong : 
**Vless** (for both v2ray and Xray) should be the best performance wise , cause it doesnt have integrated encryption and is really light .
**Vless reality** is even faster beceause it uses XTLS  thats a very efficent encryption protocol .
**Vmess** seems to be the stock version with focus on encryption and obfuscation .
**socks**   i didnt understand fully what it does , it says its usefull on simple network routing "??".
**trojan** is just obfuscating all traffic as https to avoid almost all firewall 
**Xray Shadowsocks**  uses shadowsock over Xray ????

Ysurac, 2024-07-16:
Vless is encrypted by default in OpenMPTCProuter using TLS. All Proxy/VPN have encryption enabled by default.
For the speed, you need to test, not always the same that is faster.
Socks is a simple socks proxy.
XRay Shadowsocks is using XRay with Shadowsocks 2022 protocol for client/server

vempire-ghost, 2024-07-16:
> Vless is encrypted by default in OpenMPTCProuter using TLS. All Proxy/VPN have encryption enabled by default. For the speed, you need to test, not always the same that is faster. Socks is a simple socks proxy. XRay Shadowsocks is using XRay with Shadowsocks 2022 protocol for client/server

Can I disable this encryption in Vless if I want?

Ysurac, 2024-07-23:
No, for now encryption can't be disabled on XRay or V2Ray for any protocols.

closed 2024-07-23 by Brazzo978