Ysurac/openmptcprouter #3445: Cannot write to ‘-’ (Broken pipe)

mvmaksnet, 2024-07-11:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

mvmaksnet, 2024-07-11:
Don't know what happens with creating issue ticket...
Problem is when I tried to install VPS server on new debian 12 bookworm.
Problem is:

E: Sub-process /usr/bin/dpkg returned an error code (1)
-                    89%[================>   ]  79.16K  --.-KB/s    in 1.8s

Cannot write to ‘-’ (Broken pipe).

Ysurac, 2024-07-15:
What is the command you use ?
Did you try:
 ```
wget https://www.openmptcprouter.com/server/debian-x86_64.sh
KERNEL="6.1" sh debian-x86_64.sh
```

mvmaksnet, 2024-07-16:
Yes.

Ysurac, 2024-07-23:
Sorry for late answer, I would need more log in this case.

closed 2024-10-26 by github-actions[bot]