Ysurac/openmptcprouter #3447: Internet provider not working in bonding

mvmaksnet, 2024-07-15:
Hello.
I have problem that one mobile internet provider don't work in multipath bonding.
A1 and Telekom Srbija works but Yettel provider simply just don't work when I do speed test.
I started speed test on on PC wich is connected to openmptcp router and what I see in bandwith stats page that bonding is working only on A1 and Telekom, but Yettel not.
Is there any setting that I need to configure when this is happening?

Ysurac, 2024-07-15:
This issue is not in the good categorie. I would need a real issue with the template used: I need router and version version used with kernel,...

mvmaksnet, 2024-07-15:
Ok. I will make new ticket issue. Thanks.

closed 2024-07-15 by Ysurac