Ysurac/openmptcprouter #3448: 1 Internet operator don't work in bodning - 2 other works

mvmaksnet, 2024-07-15:
Current setup

I have made mptcp router and set properly with 3 connections which are GREEN on status page. I am using this router for browsing and streaming video.

Expected Behavior
I expected that 3 diferent operators works together to get better bonding experience, but one of them just won't bond. 

Current Behavior
During my testing (Speedtest, rtmp streaming, srt streaming) I encountered the problem that one operator Yettel won't work in bonding. Telekom and A1 operators are bonding normally. So I think that this operator locked something or I need something to change in settings for this specific operator.

Specifications
OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022
OpenMPTCProuter VPS version: 0.1030 6.1.00-amd64
OpenMPTCProuter VPS provider: Maksnet
OpenMPTCProuter platform: x86_64
Country: Serbia

Ysurac, 2024-07-15:
Can you take a screenshot of System->OpenMPTCProuter, Status page ?

mvmaksnet, 2024-07-15:
[image]

Ysurac, 2024-07-15:
Did you try to set yeltel as master and maybe with only one another connection to test ?

mvmaksnet, 2024-07-15:
[image]

I have added new 4th connection and it works but A1 operator, Wan3 is yettel and not working in bonding.

mvmaksnet, 2024-07-15:
I have made 4th connection with A1 operator and it is bonding, but wan3 yettel operator is not bonded. Here is example of screenshot of bandwidth stats while do speedtest upload test.
Also tried make wan3 as master but it was the same (without bonding)

[image]

Ysurac, 2024-07-15:
It's in System->OpenMPTCProuter, wizard tab, the the box on the connection yettel

mvmaksnet, 2024-07-15:
I tried this but no success.
[image]

Ysurac, 2024-07-15:
Maybe I fixed MPTCP over VPN only in snapshot image... I will have to do a new release soon.

mvmaksnet, 2024-07-15:
Ok. Please inform me when you do that. Thanks.
I have one more question not related to this issue.
If I give MPTCP router to my client how can I login in router without having anydesk on clients laptop.
Should I configure portforwarding to make this work? How can I do that?
public_ip:port

Ysurac, 2024-07-15:
By doing something like that https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding and in Network->Firewall set to accept traffic that come from VPN (by default only outgoing traffic is allowed).

mvmaksnet, 2024-07-15:
I have made it work but I couldn't choose specific 192.168.100.1 IP address but only option is ANY.
[image]

Ysurac, 2024-07-15:
You can enter IP manually, there is internal IP address only when DHCP is used.

mvmaksnet, 2024-07-16:
Please inform me when you make new release so I can use yettel operator. Thanks for help.

mvmaksnet, 2024-07-16:
Do mptcp router have speedtest integrated so I don't need to use speedtest browser test?

Ysurac, 2024-07-23:
```omr-test-speed``` via SSH can be used to get aggregated speed.

dragon2611, 2024-07-31:
> Maybe I fixed MPTCP over VPN only in snapshot image... I will have to do a new release soon.

Can you confirm which snapshot,

I tried openmptcprouter v0.61-snapshot-6.6 r0+26695-820823198d and whilst I see wireguard traffic at the VPS it's coming from WAN1 not WAN3. (This is more than was happening with stable where the tunnel wasn't coming up at all).

I suspect the Cellular carrier O2 (Telefonica) might be messing with MPTCP, it's previously been blocked on them and although it's not reporting blocked that WAN doesn't seem to be being used either, very little traffic on it (kilobits)

closed 2024-11-04 by github-actions[bot]

mvmaksnet, 2025-03-06:
Are threre any news about this topic?

Ysurac, 2025-03-06:
@mvmaksnet You tested latest Release Candidate ?