Ysurac/openmptcprouter #3449: DSCP not working

YB1200, 2024-07-15:
## Expected Behavior
OMR-DSCP works correctly.
## Current Behavior
I am trying to test OMR-DSCP, but when I generate UDP traffic, it is supposed to be treated as high priority. However, when I check in Wireshark, I see that the class is CS0 instead of CS7. I also tried to test DSCP with domains, but the same problem persists. All the traffic is set to CS0.

## Specifications

  - OpenMPTCProuter version:  v6.0-6.1
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: BPI-R64

Ysurac, 2024-07-15:
I would need, via SSH on the router, the result of ```uci show dscp```, ```uci show dhcp```, ```uci show firewall``` and ```nft list ruleset```

YB1200, 2024-07-15:
`uci show dscp`
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs7'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs4'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs4'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs4'
dscp.@classify[3].dest_port='65500'
dscp.@classify[3].comment='OMR API'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='tcp'
dscp.@classify[4].class='cs7'
dscp.@classify[4].dest_port='65001,65301,65401,65011'
dscp.@classify[4].comment='OMR vpn'
dscp.@classify[5]=classify
dscp.@classify[5].direction='both'
dscp.@classify[5].proto='udp'
dscp.@classify[5].class='cs7'
dscp.@classify[5].dest_port='65001,65301'
dscp.@classify[5].comment='OMR vpn'
dscp.@classify[6]=classify
dscp.@classify[6].direction='both'
dscp.@classify[6].proto='tcp'
dscp.@classify[6].class='cs6'
dscp.@classify[6].dest_port='65101,65228'
dscp.@classify[6].comment='OMR proxy'
dscp.@domains[0]=domains
dscp.@domains[0].name='googlevideo.com'
dscp.@domains[0].class='cs4'
dscp.@domains[1]=domains
dscp.@domains[1].name='nflxvideo.net'
dscp.@domains[1].class='cs4'
dscp.@domains[2]=domains
dscp.@domains[2].name='s3.ll.dash.row.aiv-cdn.net'
dscp.@domains[2].class='cs4'
dscp.@domains[3]=domains
dscp.@domains[3].name='d25xi40x97liuc.cloudfront.net'
dscp.@domains[3].class='cs4'
dscp.@domains[4]=domains
dscp.@domains[4].name='aiv-delivery.net'
dscp.@domains[4].class='cs4'
dscp.@domains[5]=domains
dscp.@domains[5].name='fbcdn.net'
dscp.@domains[5].class='cs4'
dscp.@domains[6]=domains
dscp.@domains[6].name='ttvnw.net'
dscp.@domains[6].class='cs4'
dscp.@domains[7]=domains
dscp.@domains[7].name='vevo.com'
dscp.@domains[7].class='cs4'
dscp.@domains[8]=domains
dscp.@domains[8].name='audio-fa.scdn.com'
dscp.@domains[8].class='cs4'
dscp.@domains[9]=domains
[... 1738 lines cut ...]
		iifname { "br-lan", "wl0-ap0" } ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65001, 65011, 65301, 65401 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule5"
		iifname { "br-lan", "wl0-ap0" } ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport 0-65535 udp dport { 65001, 65301 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule6"
		iifname { "br-lan", "wl0-ap0" } ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65101, 65228 } counter packets 0 bytes 0 ip dscp set cs6 comment "!fw4: omr_dscp_rule7"
	}

	chain mangle_output {
		type route hook output priority mangle; policy accept;
	}

	chain mangle_forward {
		type filter hook forward priority mangle; policy accept;
		iifname { "br-lan", "wl0-ap0" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
		iifname { "eth0", "lan1", "lan3", "lan4" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		iifname "tun0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
	}

	chain upnp_forward {
	}

	chain upnp_prerouting {
	}

	chain upnp_postrouting {
	}
}

YB1200, 2024-07-15:
Even when I tried to ping any IP address, it was supposed to have the class CS7 with the default configuration of OMR-DSCP, but I see the class as CS0. It's really weird.

Ysurac, 2024-07-16:
You are checking DSCP between which points ?

YB1200, 2024-07-16:
From OMR to the application.
For example, I want to prioritize google meet traffic : it uses WebRTC based on UDP. How can do that with DSCP ?

Ysurac, 2024-07-23:
You can try latest snapshot this may be better. I've made some changes in DSCP, a side was missing since conversion to nftables.

YB1200, 2024-07-23:
I have just updated to the latest snapshot and also upgraded the server to the most recent version, but I encountered the following error:
[image]

I obtained information by running `curl -k https://vps-ip:65500/` from both the router and the VPS. 

It worked fine when I was using the stable release. I have now switched to the latest snapshot to test DSCP again.

Ysurac, 2024-07-23:
Even with the "Can't get public IP..." message, it's working correctly or not ? It seems that it correctly output from the server.
For both, I would need the result of ```logread``` command via SSH or Status->System Log (it's same info).

YB1200, 2024-07-23:
No, it didn't work. The aggregation is not working. It seems that Shadowsocks is down.

System log: 
Jul 23 12:15:10 OpenMPTCProuter daemon.warn openvpn(omr)[21123]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jul 23 12:15:10 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TCP/UDP: Preserving recently used remote address: [AF_INET]193.169.104.0:65301
Jul 23 12:15:10 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: Attempting to establish TCP connection with [AF_INET]193.169.104.0:65301
Jul 23 12:15:10 OpenMPTCProuter daemon.err openvpn(omr)[21123]: TCP: connect to [AF_INET]193.169.104.0:65301 failed: Connection refused
Jul 23 12:15:10 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Jul 23 12:15:14 OpenMPTCProuter daemon.warn openvpn(omr)[21123]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Jul 23 12:15:14 OpenMPTCProuter daemon.warn openvpn(omr)[21123]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jul 23 12:15:14 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TCP/UDP: Preserving recently used remote address: [AF_INET]193.169.104.0:65301
Jul 23 12:15:14 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: Attempting to establish TCP connection with [AF_INET]193.169.104.0:65301
Jul 23 12:15:14 OpenMPTCProuter daemon.err openvpn(omr)[21123]: TCP: connect to [AF_INET]193.169.104.0:65301 failed: Connection refused
Jul 23 12:15:14 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Jul 23 12:15:18 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (193.169.104.0)
Jul 23 12:15:21 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 23 12:15:22 OpenMPTCProuter daemon.warn openvpn(omr)[21123]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Jul 23 12:15:22 OpenMPTCProuter daemon.warn openvpn(omr)[21123]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jul 23 12:15:22 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TCP/UDP: Preserving recently used remote address: [AF_INET]193.169.104.0:65301
Jul 23 12:15:22 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: Attempting to establish TCP connection with [AF_INET]193.169.104.0:65301
Jul 23 12:15:22 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TCP connection established with [AF_INET]193.169.104.0:65301
Jul 23 12:15:22 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TCPv4_CLIENT link local: (not bound)
Jul 23 12:15:22 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TCPv4_CLIENT link remote: [AF_INET]193.169.104.0:65301
Jul 23 12:15:23 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: [server] Peer Connection Initiated with [AF_INET]193.169.104.0:65301
Jul 23 12:15:23 OpenMPTCProuter daemon.warn openvpn(omr)[21123]: sitnl_route_save: rtnl: can't get ifname for index 0: No such device or address (errno=6)
Jul 23 12:15:23 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: TUN/TAP device tun0 opened
Jul 23 12:15:23 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: net_iface_mtu_set: mtu 1420 for tun0
Jul 23 12:15:23 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: net_iface_up: set tun0 up
Jul 23 12:15:23 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Jul 23 12:15:23 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Jul 23 12:15:23 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Jul 23 12:15:23 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Jul 23 12:15:23 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: net_addr_v4_add: 10.255.252.2/24 dev tun0
Jul 23 12:15:23 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Jul 23 12:15:23 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Jul 23 12:15:23 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='tun0'
Jul 23 12:15:23 OpenMPTCProuter daemon.info ModemManager[27116]: hotplug: add network interface tun0: event processed
Jul 23 12:15:25 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 23 12:15:27 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Jul 23 12:15:27 OpenMPTCProuter daemon.err openvpn(omr)[21123]: event_wait : Interrupted system call (fd=-1,code=4)
Jul 23 12:15:27 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Jul 23 12:15:27 OpenMPTCProuter daemon.notice openvpn(omr)[21123]: net_addr_v4_del: 10.255.252.2 dev tun0
Jul 23 12:15:27 OpenMPTCProuter daemon.notice ttyd[30963]: [2024/07/23 12:15:27:9493] N: rops_handle_POLLIN_netlink: DELADDR
Jul 23 12:15:27 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Jul 23 12:15:27 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Jul 23 12:15:27 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
[... cut ...]
Jul 23 12:18:49 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jul 23 12:18:49 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)
Jul 23 12:18:57 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)
Jul 23 12:19:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)

YB1200, 2024-07-23:
Here are the logs:

-- Boot 413a4db027e54897aa09291495e528ed --
Jul 23 12:53:07 Main-Server systemd[1]: Started omr-admin.service - OMR-Admin.
Jul 23 14:11:50 Main-Server systemd[1]: Stopping omr-admin.service - OMR-Admin...
Jul 23 14:11:50 Main-Server systemd[1]: omr-admin.service: Deactivated successfully.
Jul 23 14:11:50 Main-Server systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jul 23 14:11:50 Main-Server systemd[1]: omr-admin.service: Consumed 14.132s CPU time.
Jul 23 14:11:54 Main-Server systemd[1]: Started omr-admin.service - OMR-Admin.
Jul 23 14:11:54 Main-Server omr-admin.py[288767]: Traceback (most recent call last):
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:   File "/usr/local/bin/omr-admin.py", line 42, in <module>
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:     from netjsonconfig import OpenWrt
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:   File "/usr/local/lib/python3.11/dist-packages/netjsonconfig/__init__.py", line 5, in <module>
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:     from .backends.openvpn.openvpn import OpenVpn  # noqa
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:   File "/usr/local/lib/python3.11/dist-packages/netjsonconfig/backends/openvpn/openvpn.py", line 2, in <module>
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:     from ..base.backend import BaseVpnBackend
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:   File "/usr/local/lib/python3.11/dist-packages/netjsonconfig/backends/base/backend.py", line 10, in <module>
Jul 23 14:11:54 Main-Server omr-admin.py[288767]:     from jsonschema import Draft4Validator, draft4_format_checker
Jul 23 14:11:54 Main-Server omr-admin.py[288767]: ImportError: cannot import name 'draft4_format_checker' from 'jsonschema' (/usr/local/lib/python3.11/dist-packages/jsonschema/__init__.py)
Jul 23 14:11:54 Main-Server systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Jul 23 14:11:54 Main-Server systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jul 23 14:11:54 Main-Server systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 1.
Jul 23 14:11:54 Main-Server systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jul 23 14:11:54 Main-Server systemd[1]: Started omr-admin.service - OMR-Admin.
Jul 23 14:14:43 Main-Server systemd[1]: Stopping omr-admin.service - OMR-Admin...
Jul 23 14:14:44 Main-Server systemd[1]: omr-admin.service: Deactivated successfully.
Jul 23 14:14:44 Main-Server systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jul 23 14:14:44 Main-Server systemd[1]: omr-admin.service: Consumed 1.434s CPU time.
Jul 23 14:14:44 Main-Server systemd[1]: Started omr-admin.service - OMR-Admin.
Jul 23 14:14:54 Main-Server systemd[1]: Stopping omr-admin.service - OMR-Admin...
Jul 23 14:14:54 Main-Server systemd[1]: omr-admin.service: Deactivated successfully.
Jul 23 14:14:54 Main-Server systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jul 23 14:14:54 Main-Server systemd[1]: omr-admin.service: Consumed 1.145s CPU time.
-- Boot ac9312d409274676b413feedbc2d023c --
Jul 23 14:15:18 Main-Server systemd[1]: Started omr-admin.service - OMR-Admin.

Ysurac, 2024-07-23:
You can try new VPS script, it should solve previous omr-admin crash.

YB1200, 2024-07-23:
Still the same problem: can't get vps token. 
From the vps side, yes the issue is corrected.

Ysurac, 2024-07-23:
You still have "no server API answer on" in router log ? The ```curl -k https://vps-ip:65500/``` still work from router ?
Can you check if you have same key in System->OpenMPTCProuter, Wizard tab and in ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` on VPS ?

YB1200, 2024-07-23:
I got error from `curl -k https://vps-ip:65500/ `: 
curl: (7) Failed to connect to X.X.X.X port 65500 after 53 ms: Error

Yes, same on router and VPS.

Ysurac, 2024-07-23:
Can you do the curl command from VPS and from another connection ?
* If it's work: 
I would need the result of ```ip r``` and ```ip a``` from the router. And ```curl -k --interface lan1 https://vps-ip:65500/``` (same for lan2 and lan3).
* If it's not working:
```systemctl restart omr-admin``` on the VPS and try again. ```journalctl -u omr-admin``` to check if no errors in logs.

YB1200, 2024-07-23:
result of `ip r` : 
default via 10.255.255.1 dev tun0 
default metric 1 
	nexthop via 192.168.1.1 dev lan1 weight 100 
	nexthop via 192.168.3.1 dev lan2 weight 1 
	nexthop via 192.168.2.1 dev lan3 weight 1 
default via 192.168.1.1 dev lan1 metric 9 
default via 192.168.3.1 dev lan2 metric 10 
default via 192.168.2.1 dev lan3 metric 11 
default via 10.255.255.1 dev tun0 metric 1500 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1500 
127.0.0.0/8 dev lo proto static scope link metric 7 
192.168.1.0/24 dev lan1 scope link metric 9 
192.168.2.0/24 dev lan3 scope link metric 11 
192.168.3.0/24 dev lan2 scope link metric 10 

Result of `ip a `:
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP group default qlen 1000
    link/ether 46:07:a4:20:c5:8e brd ff:ff:ff:ff:ff:ff
    inet6 fe80::4407:a4ff:fe20:c58e/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2e:54:0e:fb:78:61 brd ff:ff:ff:ff:ff:ff
4: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 46:07:a4:20:c5:8e brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global wan
       valid_lft forever preferred_lft forever
    inet6 fe80::4407:a4ff:fe20:c58e/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
5: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 46:07:a4:20:c5:8e brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.42/24 brd 192.168.1.255 scope global lan1
       valid_lft forever preferred_lft forever
6: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 46:07:a4:20:c5:8e brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.42/24 brd 192.168.3.255 scope global lan2
       valid_lft forever preferred_lft forever
7: lan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 46:07:a4:20:c5:8e brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.42/24 brd 192.168.2.255 scope global lan3
       valid_lft forever preferred_lft forever
8: lan4@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 46:07:a4:20:c5:8e brd ff:ff:ff:ff:ff:ff
9: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 2ae1:9ac0:cbf4::
10: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
11: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
12: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr c6d4:41c5:d8cf::
15: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
33: ifb4lan3: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether f2:94:38:f9:f5:7b brd ff:ff:ff:ff:ff:ff
36: ifb4lan4: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 46:20:d8:60:f7:7c brd ff:ff:ff:ff:ff:ff
38: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none 
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::24bb:7cbc:3c37:cb5/64 scope link stable-privacy proto kernel_ll 
       valid_lft forever preferred_lft forever

And `curl -k --interface lan1 https://vps-ip:65500/` works for all interfaces.

Ysurac, 2024-07-23:
It's missing direct server ip route in routing table.
Can you send me the result of ```uci show openmptcprouter``` ?
Also, you can, via SSH on the router, do ```uci set openmptcprouter.settings.debug='true' && uci commit openmptcprouter```, reboot, wait 2 minutes and send me again the result of Status->System log ?

Thank you for you cooperation and sorry for these problems.

YB1200, 2024-07-23:
Result of `uci show openmptcprouter`
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.sipalg='1'
openmptcprouter.settings.restrict_to_lan='1'
openmptcprouter.settings.vpn='openvpn'
openmptcprouter.settings.master='balancing'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='3'
openmptcprouter.settings.status_getip_timeout='1'
openmptcprouter.settings.enable_nodelay='1'
openmptcprouter.settings.scaling_governor='performance'
openmptcprouter.settings.menu='OpenMPTCProuter'
openmptcprouter.settings.disableintfrename='1'
openmptcprouter.settings.tracebox='0'
openmptcprouter.settings.version='0.61-snapshot-6.1'
openmptcprouter.settings.allmptcpovervpn='0'
openmptcprouter.settings.mptcpovervpn='wireguard'
openmptcprouter.settings.dns64='0'
openmptcprouter.settings.proxy='shadowsocks-rust'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.encryption='aes-256-gcm'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.apilc='1721746127'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.set_firewall='1'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.current='1'
openmptcprouter.vps.password='108128D771131D4450E5103060D867B20645B4EFA8C463C2E47E0A520AD21799'
openmptcprouter.vps.admin_error='1'
openmptcprouter.vps.token_error='1'
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.wgkey='qxHH+jTZdbcw0zO0OClrN9+m7QxpPdxQHFZEI+x0Jhg='
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.kernel='6.6.36-x64v2-xanmod1'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.1031-test'
openmptcprouter.vps.lastbackup='1719998775.0262434'
openmptcprouter.vps.lastchange='1721746116'
openmptcprouter.vps.ip='193.169.104.0'
openmptcprouter.vps.token='token-vps' 
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'openvpn_bonding' 'mlvpn'
openmptcprouter.vps.available_proxy='shadowsocks' 'shadowsocks-go' 'v2ray' 'v2ray-vmess' 'v2ray-socks' 'v2ray-trojan' 'xray' 'xray-vless-reality' 'xray-vmess' 'xray-socks' 'xray-trojan' 'xray-shadowsocks'
openmptcprouter.vps.allbackup='openmptcprouter-1709733563-backup.tar.gz|1709733563.2597651' 'openmptcprouter-1709736067-backup.tar.gz|1709736067.9583619' 'openmptcprouter-1709894431-backup.tar.gz|1709894431.2882643' 'openmptcprouter-1719998775-backup.tar.gz|1719998775.0262434' 'openmptcprouter-backup.tar.gz|1719998775.0262434'
openmptcprouter.omr=router
openmptcprouter.omr.ss_sss0='down'
openmptcprouter.omr.detected_public_ipv4='193.169.104.0'
openmptcprouter.loopback=interface
openmptcprouter.loopback.multipath='off'
openmptcprouter.loopback.metric='7'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='9999'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.lan.testspeed='0'
openmptcprouter.wan1=interface
openmptcprouter.wan1.multipath='on'
openmptcprouter.wan1.metric='9'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.testspeed='0'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.latency='50'
openmptcprouter.wan1.latency_previous='54'
openmptcprouter.wan1.manufacturer='huawei'
[... 469 lines cut ...]
Jul 23 14:59:53 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 
Jul 23 14:59:53 OpenMPTCProuter user.notice post-tracking-003-up: Set server vps (193.169.104.0) default route 193.169.104.0 nexthop via 192.168.1.1 dev lan3 weight 1 nexthop via 192.168.3.1 dev lan3 weight 1 nexthop via 192.168.2.1 dev lan3 weight 100
Jul 23 14:59:53 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 
Jul 23 14:59:54 OpenMPTCProuter user.notice post-tracking-003-up: Set server vps (193.169.104.0) default route 193.169.104.0 nexthop via 192.168.1.1 dev lan2 weight 1 nexthop via 192.168.3.1 dev lan2 weight 1 nexthop via 192.168.2.1 dev lan2 weight 100
Jul 23 14:59:54 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 
Jul 23 14:59:54 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)
Jul 23 15:00:00 OpenMPTCProuter user.notice post-tracking-003-up: Set server vps (193.169.104.0) default route 193.169.104.0 nexthop via 192.168.1.1 dev lan1 weight 1 nexthop via 192.168.3.1 dev lan1 weight 1 nexthop via 192.168.2.1 dev lan1 weight 100
Jul 23 15:00:00 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 
Jul 23 15:00:00 OpenMPTCProuter user.notice post-tracking-003-up: Set server vps (193.169.104.0) default route 193.169.104.0 nexthop via 192.168.1.1 dev lan3 weight 1 nexthop via 192.168.3.1 dev lan3 weight 1 nexthop via 192.168.2.1 dev lan3 weight 100
Jul 23 15:00:00 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 
Jul 23 15:00:01 OpenMPTCProuter user.notice post-tracking-003-up: Set server vps (193.169.104.0) default route 193.169.104.0 nexthop via 192.168.1.1 dev lan2 weight 1 nexthop via 192.168.3.1 dev lan2 weight 1 nexthop via 192.168.2.1 dev lan2 weight 100
Jul 23 15:00:01 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 
[... cut ...]
Jul 23 15:00:09 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)
Jul 23 15:00:12 OpenMPTCProuter user.notice post-tracking-003-up: Set server vps (193.169.104.0) default route 193.169.104.0 nexthop via 192.168.1.1 dev lan1 weight 1 nexthop via 192.168.3.1 dev lan1 weight 1 nexthop via 192.168.2.1 dev lan1 weight 100
Jul 23 15:00:12 OpenMPTCProuter user.notice post-tracking-003-up: New server route is 

You're welcome! I also enjoy working on tasks like debugging to learn more. Thank you!

Ysurac, 2024-07-23:
Thanks, a new image is compiling and should fix VPS server routes issue.

YB1200, 2024-07-24:
Thanks, it works now. I will test DSCP and get back to you if there is a problem.

In the meantime, I have a question: When we have this configuration, does it mean that OMR will send most of the data on lan1 and the rest will be sent on lan2 and lan3 because lan1 has the highest weight? From my understanding, OMR sends traffic on the link with the minimum latency until it becomes saturated.
```
default metric 1
nexthop via 192.168.1.1 dev lan1 weight 100
nexthop via 192.168.3.1 dev lan2 weight 1
nexthop via 192.168.2.1 dev lan3 weight 1
```

Ysurac, 2024-07-24:
Only when VPN is down or on initial connection, to do some load balancing.
Else the weight doesn't have any importance for aggregation and only latency is used in default Multipath scheduler.

YB1200, 2024-07-24:
Understood, thanks.
I tested DSCP, and it doesn't seem to be working. I generated an iperf test using the UDP protocol from a laptop connected to OMR to my server. I classified this traffic with EF-Voice and launched a packet capture with tcpdump. However, I observed that all the traffic was classified with CS0 instead of EF.

closed 2024-10-27 by github-actions[bot]