Ysurac/openmptcprouter #3451: Entware Plugin/Add-on for Entware

mrkaffeine92, 2024-07-16:
**Is your enhancement proposal related to a problem? Please describe.**
A lot of the devices are currently unsupported, but I think it would be more feasible if a plugin/add-on was created where instead of creating/writing images for specific devices, it is written to work with the current foundation within Asus-Merlin. 

**Describe the solution you'd like**
I'd like a package for Entware (see https://gist.github.com/1951FDG/3cada1211df8a59a95a8a71db6310299) where the package can be installed and will run the script and we can just follow the VPS install etc.
*https://bin.entware.net/armv7sf-k3.2/Packages.html

**Describe alternatives you've considered**
Installing all the packages independently, but the set up has left me with rise in blood pressure.

**Additional context**
I think it's a good idea to expand the mobility of what Ysurac has done and get his name and reputation out there.

Ysurac, 2024-07-23:
OpenMPTCProuter use a custom kernel (even if less custom since 6.x) and lot of customisation, it's difficult to put all in a package. It's really a full Linux Distribution.

mrkaffeine92, 2025-02-09:
> OpenMPTCProuter use a custom kernel (even if less custom since 6.x) and lot of customization, it's difficult to put all in a package. It's really a full Linux Distribution.

Yep, and I can tell you are already trying to make it as minimal with resources as possible to crank out all the speed and computations.

closed 2025-02-09 by mrkaffeine92