Ysurac/openmptcprouter #3453: Hetzner AMD VPS 2GB has memory leak with OpenMPTCProuter (server stops responding to SSH after 24 hours)

BigFanOpenmptcprouter, 2024-07-18:
## Expected Behavior
VPS server should not freeze (no response to SSH)

## Current Behavior
* Clean install of OpenMPTCProuter script (v0.6 current stable) on Debian 12 using Hetzner web interface. The VPS freezes after about 24 hours of usage, the only solution is reset via Hetzner web interface and then it's good for another 24 hours.

* Server does not respond to SSH connections. Hetzner AMD VPS 2 GB RAM, larger swap does not fix it. Reinstall of the OS does not fix it. Upgrade to larger RAM does not fix it (we tried 4 GB RAM).

## Possible Solution
We already cancelled the VPS so we can't really debug it further, consider yourself warned, if anybody is interested to debug this feel free to do so.

Workaround no1: Get Hetzner dedicated server (anything works - we got the budget one from "auction"). Be warned the routing might be a bit different for each of their datacenters).
Workaround no2: Get VPS from OVH (only tested the 4GB RAM). Works fine.

## Steps to Reproduce the Problem

  1. Install Debian 12 on Hetzner VPS (we had the AMD 2 GB RAM)
  2. Install OpenMPTCProuter VPS script version 0.6 (or whatever was available as "default" on 2024-06-11)
  3. Use the OpenMPTCProuter router via the local machine for about 24 hours.
  4. Memory leak occurs (see screenshot), hard reset workoaround good for 24 hours.

## Context (Environment)
Fresh install of Debian 12 + OpenMPTCProuter VPS script, nothing else. Hetzner VPS AMD 2 GB.

## Specifications

  - OpenMPTC
[image]
Prouter version: 0.6 (or whatever was available as "default" on 2024-06-11)
  - OpenMPTCProuter VPS version: 0.6 (or whatever was available as "default" on 2024-06-11)
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64
  - Country: EU/no restrictions/network independent.

Ysurac, 2024-07-23:
I have the same issue (at least it looks likes the same) currently on another VPS provider with latest snapshot image, I'm not sure the exact reason of this problem but I'm actively working on it.

BakaMai, 2024-07-27:
> I have the same issue (at least it looks likes the same) currently on another VPS provider with latest snapshot image, I'm not sure the exact reason of this problem but I'm actively working on it.

I found ss-redir of shadowsocks-rust or xray shadowsocks-2022 is leaking memory .
Just do some speedtest continuously ...about 10 times , 2G memory is full ...stop a while , it release memory .
if one tcp connection never disconnect , memory usage of ss-redir will be bigger and bigger . until it crash the OS .

 Change to Xray Vless , it never happened .

closed 2024-10-30 by github-actions[bot]