Ysurac/openmptcprouter #3459: Manually edit config file

BakaMai, 2024-07-23:
## Expected Behavior
Proxy Encryption Chiper aes-128-gcm

## Current Behavior
Only aes-256-gcm option

Hello ~
I would like to change the config manually in Client side .
Tried edit the Xray client directly in 
/var/etc/xray/xray.main.json
But the config change back to aes-256-gcm after some sec ... So i realized it's not the correct place to change the config ...

I have seen a "show all setting" tab under OpenMPTCPRouter menu . But , it's a preview , not changeable .
Can i change config file like ubnt router in one config file ?
Not only the Encryption Chiper i would like to change , but also other settings about xray i would like to change .

Thank you .

Ysurac, 2024-07-24:
Only XRay Shadowsocks protocol encryption is done via System->OpenMPTCProuter, Wizard, advanced settings tab.

BakaMai, 2024-07-24:
> otocol encryption is done via System->OpenMPTCProuter, Wizard, advanced settings

Thank you . I found there only aes-256-gcm , but no aes-128-gcm . And chacha20 is not working in some protocol ( setting still in aes-256-gcm , no change) . Hope this setting will be improved in the furture

closed 2024-07-24 by BakaMai

Ysurac, 2024-07-24:
chacha20 is not supported everywhere, so AES is used when not possible to use it.
I disabled AES-128-GCM support some times ago, I think it's still possible to set it manually via uci command or in /etc/config and then launch ```/etc/init.dopenmptcprouter-vps restart``` to sync the VPS.