Ysurac/openmptcprouter #3460: OMR-Bypass stopped working after upgrade to newest version

serlowenbrau, 2024-07-23:
## Expected Behavior
omr-bypass should continue to route traffic through one wan or the other

## Current Behavior
omr-bypass does not seem to have much effect at all.  I have tried by domain, by ASN, by source IP address, by service.
Streaming services and websites are showing "disable your vpn" to continue.  I rebuilt my router from scratch after the upgrade to see if that would fix it, but that did not work.

This bit is occurring in the system log.  
Jul 23 17:25:38 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:25:42 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:26:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Jul 23 17:26:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jul 23 17:26:00 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:26:00 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:26:00 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:26:00 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth3_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jul 23 17:26:01 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 23 17:26:01 OpenMPTCProuter user.notice omr-bypass: Restart dnsmasq...
Jul 23 17:26:01 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Jul 23 17:26:01 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running

## Possible Solution
I saw another issue where the source IP address rules were fixed. This is the main way I did it on the previous version... set the smart tv to a static address, then add a rule to bypass all traffic from that address.  Do I need to download a version newer than 060-61?

## Steps to Reproduce the Problem

  1.  add omr-bypass rule
  2.  test service
  3.  gets error

## Context (Environment)
I am running 2 wans through an old x86 computer that boots to openmptcprouter on a thumb drive.  

After upgrading to the latest version, the omr-bypass rules do not seem to work.

## Specifications

  - OpenMPTCProuter version: 060-6.1
  - OpenMPTCProuter VPS version: 0.1030.6.1.0-10-amd64
  - OpenMPTCProuter VPS provider: kamatera, cloudfanatic
  - OpenMPTCProuter platform: x86_64
  - Country: USA

Ysurac, 2024-07-24:
You can try snapshot release: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

serlowenbrau, 2024-07-25:
Any idea about the "file not found" messages in the omr-bypass portions of
the system log?  Bad thumb drive?

Ysurac, 2024-07-29:
Nothing to worry about, it's during omr-bypass restart due to dnsmasq configuration.

pbonnard, 2024-07-31:
My two cents: was having issue also with OMR bypass not working since 0.60 ... So applied latest snapshot 0.61 with full parameters resets, updated VPS script and now working ONLY for MAC address based bypass - not for service based (tested with mycanal and free)
Using shadowsocks for proxy as I noticed shadowsocks-rust or V2RAY are causing admi script connection issue and breaks OMR bypass as well...

Ysurac, 2024-07-31:
@pbonnard Can you open your own issues ? At least about proxy issues. All tests are first done with Shadowsocks-Rust, so this seems strange.
For Services and domains OMR-ByPass, you must be sure that OMR IP is used as DNS.

pbonnard, 2024-07-31:
> @pbonnard Can you open your own issues ? At least about proxy issues. All tests are first done with Shadowsocks-Rust, so this seems strange. For Services and domains OMR-ByPass, you must be sure that OMR IP is used as DNS.

Ok thanks. I'm using Adguard Home as DNS so that's why...

serlowenbrau, 2024-07-31:
I'll try the .61 snapshot with mac address option this weekend.  Thank you both.

serlowenbrau, 2024-08-07:
I haven't tried all the options, but source IP and mac address bypass are working in the new snapshot.

closed 2024-11-10 by github-actions[bot]