Ysurac/openmptcprouter #3463: TP Link Deco X50 - No Server ping response after 1 second #3462

posmykmedia, 2024-07-25:
## Expected Behavior
I want to set up an internet connection via fiber optic and 5G (Telekom) and combine it.

## Current Behavior
An Internet connection via the 5G router ( TP Link Deco X50) is not possible. The following error message appears:  No Server ping response after 1 second 

However, Internet access via the 5G router works if you go directly to the router. Then you have an Internet connection.

## Specifications

  - OpenMPTCProuter version:  v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version:latest
  - OpenMPTCProuter VPS provider: Ionos
  - OpenMPTCProuter platform: Debian 12, Ionos VPS
  - 
I use a Firewall Micro PC with 8 GB DDR4 RAM, 256 GB storage and 4 2.5gbs Ethernet ports (https://www.amazon.de/gp/product/B0BMVFGQSF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

The 5G router is a TP Link Deco X50, 5G router with 2.5 Gigabit port (https://www.amazon.de/gp/product/B0BWS5WBQ7/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1).

[image]
[image]

Ysurac, 2024-07-25:
Can you try, via SSH on the router a ```ping -B -I eth2 <vpsip>``` ?

posmykmedia, 2024-07-25:
Hi, thanks for the message. Can I run this via Mac Terminal? Or which console?

Ysurac, 2024-07-25:
You can use ssh or also via web interface in Services->Terminal

posmykmedia, 2024-07-26:
Thanks. I got this info:

root@OpenMPTCProuter:~# ping -B -I eth2 xxx.xxx.xx.xxx
PING xxx.xxx.xx.xxx (xxx.xxx.xx.xxx) from 192.168.50.2 eth2: 56(84) bytes of data.
From 192.168.50.1 icmp_seq=1 Destination Net Unreachable
From 192.168.50.1 icmp_seq=2 Destination Net Unreachable

Other ethernet ports are reachable. When I access the 5G router directly via WiFi and ping the IP it works. So it seems to be a DHCP problem? Can you help me?

posmykmedia, 2024-07-26:
In protocol there is:

Jul 26 22:16:15 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 192.168.50.2 error (1.2.4.8,80.67.169.40,114.114.114.114)
Jul 26 22:16:15 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to xxx.xxx.xx.xxx dev eth2

Maybe because of that?

Ysurac, 2024-07-27:
You changed the IP ? It was 192.168.100.100 with 192.168.100.10 as gateway in the screenshot and now seems to be 192.168.50.2 ?
What IP are you able to ping ?

posmykmedia, 2024-07-29:
Oh, I solved the problem. The problem was that the 5G router wanted to establish an IPv6 connection and the openmptcprouter could not establish it. Now all outgoing connections work after switching to IPv4.

I now have the problem that I only have around 100 Mbits download and 100 Mbits upload. But I already have around 1 GB downloads (standalone) via my fiber optic connection alone.

The Ionos VPS only has a throughput of 1GBs, but I would still not be able to use just 10% of the line. Where would be the best place to debug this? I use the latest cables (Cat7).

Which proxy should I best set up?

What next steps could I take to debug the problem?

Ysurac, 2024-07-29:
Use Shadowsocks-Rust as Proxy.
I would need a screenshot of status page.

wlew-via, 2024-08-02:
> Oh, I solved the problem. The problem was that the 5G router wanted to establish an IPv6 connection and the openmptcprouter could not establish it. Now all outgoing connections work after switching to IPv4.
> 
I have a similar problem with the same error. How did you fix this?

posmykmedia, 2024-08-03:
To set up my Internet connection, an IP V4 IP address is required. If your
Internet provider can only establish an IP V6 connection, you have to try
to change it to IP V4. With the Deco (Router), I had to change Telekom's
access data to establish an IP V4 connection. That helped me. Please check.

closed 2024-11-06 by github-actions[bot]