Ysurac/openmptcprouter #3464: MPTCP kernel on router and VPS doesn't match after upgrade

murateri75, 2024-07-26:
## Expected Behavior

My local router is working fine but cannot connect to the VPS server
"MPTCP kernel on router and VPS doesn't match"
I did upgrade VPS server with command line found here
https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

apt-get update && apt-get upgrade
then 
wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.1" sh

Documentation 
## Current Behavior

After reboot 

uname -a 
Linux aja-rtr-01 5.10.0-30-amd64 #1 SMP Debian 5.10.218-1 (2024-06-01) x86_64 GNU/Linux

ls /boot
root@aja-rtr-01:~# ls -l /boot
total 135984
-rw-r--r-- 1 root root   236469 Jul 27  2023 config-5.10.0-23-amd64
-rw-r--r-- 1 root root   236326 Jun  1 09:24 config-5.10.0-30-amd64
-rw-r--r-- 1 root root   236220 Sep 23  2021 config-5.10.0-8-amd64
-rw-r--r-- 1 root root   209132 Jul 24  2022 config-5.4.207-mptcp
drwx------ 2 root root     4096 Jan  1  1970 efi
drwxr-xr-x 5 root root     4096 Jul 14 08:19 grub
-rw-r--r-- 1 root root 29012721 Aug 10  2023 initrd.img-5.10.0-23-amd64
-rw-r--r-- 1 root root 30212057 Jul 14 08:19 initrd.img-5.10.0-30-amd64
-rw-r--r-- 1 root root 28157884 Aug 10  2023 initrd.img-5.10.0-8-amd64
-rw-r--r-- 1 root root 22240160 Aug 10  2023 initrd.img-5.4.207-mptcp
-rw-r--r-- 1 root root       83 Jul 27  2023 System.map-5.10.0-23-amd64
-rw-r--r-- 1 root root       83 Jun  1 09:24 System.map-5.10.0-30-amd64
-rw-r--r-- 1 root root       83 Sep 23  2021 System.map-5.10.0-8-amd64
-rw------- 1 root root  2985575 Jul 24  2022 System.map-5.4.207-mptcp
-rw-r--r-- 1 root root  7036544 Jul 27  2023 vmlinuz-5.10.0-23-amd64
-rw-r--r-- 1 root root  7053824 Jun  1 09:24 vmlinuz-5.10.0-30-amd64
-rw-r--r-- 1 root root  6820832 Sep 23  2021 vmlinuz-5.10.0-8-amd64
-rw------- 1 root root  4766080 Jul 24  2022 vmlinuz-5.4.207-mptcp

Documentation says Debian will be upgraded to Debian 12 seems not the case

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022
  - OpenMPTCProuter VPS version: should be 6.1 but after upgrade kernal is 5.10
  - OpenMPTCProuter VPS provider: personal VM at hosting provider (my wife's company
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.5

Ysurac, 2024-07-26:
What is the result of ```cat /etc/os-release``` ?
The script worked correctly to the end ?

Ysurac, 2024-07-26:
So it's updated to Debian 12 but kernel is not installed, maybe because wasn't installed via "apt" ?
You can try a ```apt-get install linux-image-amd64```

murateri75, 2024-07-26:
Done with reboot
And now I am on right Kernel
***@***.***:~# uname -a
Linux aja-rtr-01 6.1.0-22-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.94-1
(2024-06-21) x86_64 GNU/Linux

[image: image.png]
and
[image: image.png]

Everything is working now

Thanks for very quick answer
Eric

closed 2024-07-29 by Ysurac