Ysurac/openmptcprouter #3465: New route constantly detected; restarting tunnel

mrkaffeine92, 2024-07-27:
Hello, every hour System Log shows that a new route is detected (10.x.x.2) to 3 to 4 - causing the tunnel to disconnect - VPN DOWN, then reconnecting. 

Can anyone help me diagnose issue? I've trued using OpenVPN TUN and Agg. Bonding, MLVPN, and the Dead Simple VPN.

Is it an issue with the VPS or my modem connections? Logs from modem do not show any info that'd cause a route change.

Ysurac, 2024-07-28:
Wrong category, should be in issues with template...
You can test latest snapshot VPS script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

mrkaffeine92, 2024-07-28:
> Wrong category, should be in issues with template... You can test latest snapshot VPS script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

Why would a snapshot be more supportive to the routing issue than RC's? I was using the latest release.

Ysurac, 2024-07-28:
Because OpenVPN config is not the same on VPS.

mrkaffeine92, 2024-07-28:
> Because OpenVPN config is not the same on VPS.

Kindly asking for more clarification. 
1) Is the same VPS configuration not able to interface correctly with: reformats of the same local server release version? In other words, every time a new installation of image (regardless if same or newer release) - requires a reimage of the VPS?
2) Have tried a few different VPS providers - same thing pretty much. Can you please share any configs or settings to enable that would help diagnose and identify what is the root cause/issue?
3) Should any particular settings be enabled/disabled on the upstream router that handles the NAT for modem? e.g., DMZ, Port Forwarding, static routes?

Thank you in advance for your assistance,

Ysurac, 2024-07-28:
In Snapshot install script, there a change in OpenVPN config, duplicate-cn is now not enabled, so IP shouldn't change.
In any case, the disconnection is not due to the VPN, you have same problem with all VPNs.
There is only latest snapshot available, so no choice. Only choice is kernel, keep same kernel version when you only want to test a fix.

mrkaffeine92, 2024-07-28:
> In Snapshot install script, there a change in OpenVPN config, duplicate-cn is now not enabled, so IP shouldn't change. In any case, the disconnection is not due to the VPN, you have same problem [...]

Have tried it on both latest RC as well as Snapshot; various VPS's with clean installs. Are you available on Instant Message (IM) on a relay or messenger and willing to help troubleshoot?

closed 2024-11-02 by github-actions[bot]