Ysurac/openmptcprouter #3469: OpenMPTCP systems fine but no bonding and traffic going to only one internet router

murateri75, 2024-07-29:
## Expected Behavior

2 DSL connection bonding (approx 5+5 Mb) 
## Current Behavior

All traffic going to only one DSL connection
Only half of theroical capability
Its like if one internet router is not recognised as working
## Specifications

  - OpenMPTCProuter version: : v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 6.1.0-22-amd64 0.1030
  - OpenMPTCProuter VPS provider: Private VM Debian 12 with Small Hosting Provider
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.4

Ysurac, 2024-07-29:
Can you put a screenshot of System->OpenMPTCProuter, Status page ?
Can you also run, via SSH on the router, ```omr-test-speed``` and ```omr-test-speed <interface>``` where interface is a real interface (you can find them with ```ip a``` or in status page) ?

murateri75, 2024-07-29:
Hello Yannick
Here you are
Status page joined
[image: image.png]

***@***.***:~# omr-test-speed
Select best test server...
host: scaleway.testdebit.info - ping: 169
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping:
host: lyon.testdebit.info - ping:
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping: 222
host: appliwave.testdebit.info - ping: 906
host: speedtest.frankfurt.linode.com - ping: 314
host: speedtest.tokyo2.linode.com - ping: 340
host: speedtest.singapore.linode.com - ping: 259
host: speedtest.newark.linode.com - ping: 177
host: speedtest.atlanta.linode.com - ping: 199
host: speedtest.dallas.linode.com - ping: 220
host: speedtest.fremont.linode.com - ping: 174
host: speed.hetzner.de - ping: 66
host: ipv4.bouygues.testdebit.info - ping: 114
host: par.download.datapacket.com - ping: 113
host: nyc.download.datapacket.com - ping: 181
host: ams.download.datapacket.com - ping: 118
host: fra.download.datapacket.com - ping: 127
host: lon.download.datapacket.com - ping: 130
host: mad.download.datapacket.com - ping: 140
host: prg.download.datapacket.com - ping: 133
host: sto.download.datapacket.com - ping: 144
host: vie.download.datapacket.com - ping: 136
host: war.download.datapacket.com - ping: 146
host: atl.download.datapacket.com - ping: 193
host: chi.download.datapacket.com - ping: 224
host: lax.download.datapacket.com - ping: 252
host: mia.download.datapacket.com - ping: 220
host: nyc.download.datapacket.com - ping: 188
host: speedtest.milkywan.fr - ping: 120
Best server is https://speed.hetzner.de/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time
 Current
                                 Dload  Upload   Total   Spent    Left
 Speed
100   190  100   190    0     0    309      0 --:--:-- --:--:-- --:--:--
309

***@***.***:~# omr-test-speed eth1
Select best test server...
host: scaleway.testdebit.info - ping: 116
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 123
host: lyon.testdebit.info - ping:
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping: 263
host: appliwave.testdebit.info - ping: 279
host: speedtest.frankfurt.linode.com - ping: 244
host: speedtest.tokyo2.linode.com - ping: 416
host: speedtest.singapore.linode.com - ping: 477
host: speedtest.newark.linode.com - ping: 232
host: speedtest.atlanta.linode.com - ping: 233
host: speedtest.dallas.linode.com - ping: 248
host: speedtest.fremont.linode.com - ping: 295
host: speed.hetzner.de - ping: 137
host: ipv4.bouygues.testdebit.info - ping: 114
host: par.download.datapacket.com - ping: 117
host: nyc.download.datapacket.com - ping: 187
host: ams.download.datapacket.com - ping: 127
host: fra.download.datapacket.com - ping: 145
host: lon.download.datapacket.com - ping: 169
host: mad.download.datapacket.com - ping: 168
host: prg.download.datapacket.com - ping: 216
host: sto.download.datapacket.com - ping: 215
host: vie.download.datapacket.com - ping: 220
host: war.download.datapacket.com - ping: 385
host: atl.download.datapacket.com - ping: 233
host: chi.download.datapacket.com - ping: 215
host: lax.download.datapacket.com - ping: 263
host: mia.download.datapacket.com - ping: 235
host: nyc.download.datapacket.com - ping: 189
host: speedtest.milkywan.fr - ping: 118
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time
 Current
                                 Dload  Upload   Total   Spent    Left
 Speed
  0 9536M    0  737k    0     0  11239      0  10d 07h  0:01:07  10d 07h
 7261^C

***@***.***:~# omr-test-speed eth2
Select best test server...
host: scaleway.testdebit.info - ping: 14
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 24
host: lyon.testdebit.info - ping:
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping: 15
host: appliwave.testdebit.info - ping: 14
host: speedtest.frankfurt.linode.com - ping: 24
host: speedtest.tokyo2.linode.com - ping: 267
host: speedtest.singapore.linode.com - ping: 329
host: speedtest.newark.linode.com - ping: 97
host: speedtest.atlanta.linode.com - ping: 105
host: speedtest.dallas.linode.com - ping: 123
host: speedtest.fremont.linode.com - ping: 160
host: speed.hetzner.de - ping: 26
host: ipv4.bouygues.testdebit.info - ping: 14
host: par.download.datapacket.com - ping: 13
host: nyc.download.datapacket.com - ping: 91
host: ams.download.datapacket.com - ping: 21
host: fra.download.datapacket.com - ping: 25
host: lon.download.datapacket.com - ping: 20
host: mad.download.datapacket.com - ping: 36
host: prg.download.datapacket.com - ping: 37
host: sto.download.datapacket.com - ping: 46
host: vie.download.datapacket.com - ping: 29
host: war.download.datapacket.com - ping: 44
host: atl.download.datapacket.com - ping: 105
host: chi.download.datapacket.com - ping: 105
host: lax.download.datapacket.com - ping: 143
host: mia.download.datapacket.com - ping: 113
host: nyc.download.datapacket.com - ping: 90
host: speedtest.milkywan.fr - ping: 21
Best server is http://par.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time
 Current
                                 Dload  Upload   Total   Spent    Left
 Speed
  0 9536M    0 25.7M    0     0   718k      0  3:46:23  0:00:36  3:45:47
 726k^C

Today its a bit low probably consequences of Internet sabotage in France
last night (2:15 AM CET)
In general with speetdtest from Google I have approx 5Mb/s on each DSL
connection

Eric

murateri75, 2024-08-01:
Hello Yannick
In Network -> MPTCP
I changed WAN2 as master and WAN1 as active and apply then all good
(previously was WAN1 master and WAN2 active)
Strange
Will retest with the opposite later this week
Thanks for your help
Eric

closed 2024-11-04 by github-actions[bot]