Ysurac/openmptcprouter #3470: Topic compilation

fdyf520, 2024-07-29:
How to switch the default theme to another one, such as argon, at compile time. The compiled file looks a lot different from the original openwrt, can not find the corresponding path to modify, hope to get a reply

Ysurac, 2024-07-30:
You can add a file to use the uci command in a new package or add in an uci-defaults sub directory of an existing one.
You can get uci command when you change theme and only click on "Save" and click on "Unsaved change" at the upper right on default theme.

fdyf520, 2024-07-30:
It may be that problems caused by translation are not clearly described
I am now deciding on the theme that needs to be displayed by default when I want to compile, that is, when you brush into the image, you do not need to manually switch to luci.
Now the theme that defaults to openmptcprouer is the default launch theme, and I want to replace it with argon as the default theme. Fix this problem directly before recompiling, rather than after booting the system. I can't find a file path that can replace the default theme. There is no configuration file in the makefile

Ysurac, 2024-07-30:
As I said, you need to create a new package with uci-defaults files or add a file in an existing package. You can look at openmptcprouter package for example.

closed 2024-11-02 by github-actions[bot]