Ysurac/openmptcprouter #3471: Trying to compile an image of r5s, stuck in this position

18750319950, 2024-07-30:
Running Sphinx v1.8.5
building [mo]: all of 0 po files
building [man]: all source files
updating environment: 34 added, 0 changed, 0 removed
reading sources... [100%] tsig-keygen                                                                                                                                                                                                    
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
writing... arpaname.1 { } ddns-confgen.8 { } delv.1 { } 
Warning, treated as error:
../../bin/delv/delv.rst:109:unknown option: +root
make[7]: *** [Makefile:1005: manpages.stamp] Error 2
make[7]: Leaving directory '/home/gary/openmptcprouter/r5s/6.1/source/build_dir/target-aarch64_generic_musl/bind-9.18.19/doc/man'
make[6]: *** [Makefile:454: all-recursive] Error 1
make[6]: Leaving directory '/home/gary/openmptcprouter/r5s/6.1/source/build_dir/target-aarch64_generic_musl/bind-9.18.19/doc'
make[5]: *** [Makefile:627: all-recursive] Error 1
make[5]: Leaving directory '/home/gary/openmptcprouter/r5s/6.1/source/build_dir/target-aarch64_generic_musl/bind-9.18.19'
make[4]: *** [Makefile:533: all] Error 2
make[4]: Leaving directory '/home/gary/openmptcprouter/r5s/6.1/source/build_dir/target-aarch64_generic_musl/bind-9.18.19'
make[3]: *** [Makefile:296: /home/gary/openmptcprouter/r5s/6.1/source/build_dir/target-aarch64_generic_musl/bind-9.18.19/.built] Error 2
make[3]: Leaving directory '/home/gary/openmptcprouter/feeds/6.1/packages/net/bind'
time: package/feeds/packages/bind/compile#6.33#4.12#10.90
    ERROR: package/feeds/packages/bind failed to build.
make[2]: *** [package/Makefile:128: package/feeds/packages/bind/compile] Error 1
make[2]: Leaving directory '/home/gary/openmptcprouter/r5s/6.1/source'
make[1]: *** [package/Makefile:122: /home/gary/openmptcprouter/r5s/6.1/source/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gary/openmptcprouter/r5s/6.1/source'
make: *** [/home/gary/openmptcprouter/r5s/6.1/source/include/toplevel.mk:232：world] 错误 2

## Specifications

  - OpenMPTCProuter version: 

  - OpenMPTCProuter platform: <r5s>

Ysurac, 2024-07-30:
Seems to be a problem with compilation of bind. Never have this issue, difficult to help here.

Gustavo251197, 2024-07-31:
This problem is caused by the sphinx version you are running. Probably because of the Ubuntu/Debian version used to compile. This happened to me in Ubuntu 20.04, which is limited to Sphinx v1.8.5.
Ubuntu 22.04 uses Sphinx 4.3.2, using this newer version allowed me to compile bind and its documentation.
If you wanna keep using your current sphinx version you'll need to patch bind to disable the warnings as errors in sphinx:  https://salsa.debian.org/dns-team/bind9/-/blob/debian/9.18/debian/patches/0001-Disable-treat-warnings-as-errors-in-sphinx-build.patch

closed 2024-08-26 by 18750319950