Ysurac/openmptcprouter #3472: Can,t get public IP address from Shadowsocks

sidelong, 2024-07-30:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

sidelong, 2024-07-30:
After updating bot VPS (Digital Ocean) and Raspberry Pi (4B, 64 bit) to latest firmware all seems fine, except that this message shows up in VPS details on Dasboard. Is this a problem - everything seems to be working well? What could be causing it?

Ysurac, 2024-07-30:
Please complete the template.

sidelong, 2024-08-01:
One additional remark. Having read through some of the discussions on upgrading, having started by using the sysupgrading route and experiencing errors, I upgraded via the 'System/Backup/Flash Firmware' route. this was successful as far as I can assess, except for the error message mentioned above.

Ysurac, 2024-08-01:
What is the proxy you use ? Shadowsocks-libev or Shadowsocks-Rust ? Shadowsocks-Rust is the new default.
You have still issue if you validate again the wizard ?

sidelong, 2024-08-02:
Bonjour,

 

Thank you for responding. I changed the default proxy to Shadowsocks-Rust and revalidated the wizard. Now all is fine, the error message no longer appears.

 

Best regards,

Mike Phillips

closed 2024-08-02 by Ysurac