Ysurac/openmptcprouter #3473: OMR Bypass for Free service (oqee.tv) missing proxad.net domain

pbonnard, 2024-07-31:
## Expected Behavior
Define free service in OMR Bypass and connect to https://oqee.tv/home or Oqee app

## Current Behavior
Cannot watch Oqee.tv for web browser or app

## Possible Solution
add proxad.net in the domain list for OMR Bypass

## Steps to Reproduce the Problem
Define free service in OMR Bypass and connect to https://oqee.tv/home or Oqee app

## Context (Environment)
Free ADSL and Free 4G multipath
openmptcprouter v0.61-snapshot-6.1 r0+24843-acf40c022e

## Specifications

  - OpenMPTCProuter version: v0.61-snapshot-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: FirstHeberg
  - OpenMPTCProuter platform: x86_64
  - Country: France

Ysurac, 2024-08-01:
Do you use OpenMPTCProuter IP as DNS on you client ?
Free service bypass free.fr, freebox.fr, oqee.tv and oqee.net domains. You tried to also bypass proxad.net and this is working ?

Ysurac, 2024-08-01:
ok, I add proxad.net to domain to bypass. Thanks

closed 2024-11-04 by github-actions[bot]