Ysurac/openmptcprouter #3476: What in the OMR converts the DNS into the server's IP?

vempire-ghost, 2024-08-03:
When I use a DNS instead of an IP as the address of my VPS, which OMR process converts this DNS into an IP and configures it in the OMR to connect? Would it be the OMR-server tracker?

Ysurac, 2024-08-05:
OMR-Tracker Server

vempire-ghost, 2024-08-05:
> OMR-Tracker Server

Does it belong to the OMR-tracker? I didn't find this process independently. Would restarting it speed up the function of updating the server's IP?

Ysurac, 2024-08-05:
It's launched by OMR-Tracker init script, but it's an independent program. You can change settings in Services->OMR-Tracker, "Server" tab.

vempire-ghost, 2024-08-05:
Would this be the best configuration to ensure that the new IP is found as quickly as possible?
[image]

I searched in `init.d` and only found `omr-tracker`. Is the `omr-tracker server` located somewhere else?

Ysurac, 2024-08-05:
As I said, it's the same init script that run all omr-tracker-* programs.
I think values are too low here, but test and you will see.
Using a DDNS is never a good idea for OMR, this may not work correctly.

vempire-ghost, 2024-08-05:
It works perfectly for my use case, but sometimes OMR takes a bit longer to identify the new IP. In this screenshot, will it check the server every 1 second? Does it resolve the DNS and ping the IP?

In the tests I conducted with the values on this screen, I didn’t notice any significant change. Sometimes OMR finds the IP at the same time as my computer does, and other times it takes a few minutes longer.

Ysurac, 2024-08-05:
Yes, it resolve IP on each test. Possible that the resolution take sometimes a longer time.
You can check system log when you have the issue to check if there is something strange.

vempire-ghost, 2024-08-05:
> Yes, it resolve IP on each test. Possible that the resolution take sometimes a longer time. You can check system log when you have the issue to check if there is something strange.

Usually, when this happens, the log shows that it was able to ping the server but didn’t get a response from the API.
After a few minutes, the `omr-tracker server` shows that the server is back online.

Is it possible to configure `omr-tracker server` to consider the VPS online based solely on the ping response? I tried changing this configuration in `omr-tracker proxy`, but it doesn’t exist on the server.

Ysurac, 2024-08-05:
In latest v0.61 snapshot yes, you can choose the way to check if server is up or not.

vempire-ghost, 2024-08-05:
> In latest v0.61 snapshot yes, you can choose the way to check if server is up or not.

This screenshot is from the latest snapshot of version 0.61. Where can I configure this?

Ysurac, 2024-08-06:
It's fully implemented. What do you have exactly in log ?

vempire-ghost, 2024-08-06:
> It's fully implemented. What do you have exactly in log ?

Aug  6 23:07:42 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on vps.ddns.net, server API answer on vps.ddns.net)

Ysurac, 2024-08-07:
It's not related to OMR-Tracker server, It's the openmptcprouter-vps script.

closed 2024-11-10 by github-actions[bot]