Ysurac/openmptcprouter #3478: Scheduler Selection and Wireless Configuration Issue in OpenMPTCPRouter v0.59

Raquel1L, 2024-08-05:
## Current Behavior
In March, I successfully used the image _openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-factory.img_ on my Raspberry Pi. In June, I updated to the current version, _openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-bcm27xx-bcm2711-rpi-4-ext4-factory 2.img,_ and confirmed today that it is functioning correctly.

However, version _v0.59.1_ includes a feature critical to my needs: the ability to select multiple schedulers. In contrast, version _v0.60_ offers only a single, default scheduler.
As a result, I reinstalled _v0.59.1_ from a copy stored in my downloads folder since March. Upon doing so, I encountered the following error on the wireless network configuration page:

_**TypeError: Right-hand side of 'instanceof' is not an object**_

This error prevents any further actions on this tab, blocking the addition and configuration of a wireless network.
To resolve this issue, I installed the image openmptcprouter-v0.60-5.4-r0+16862-170d9e447d-bcm27xx-bcm2711-rpi-4-ext4-factory.img, but the same error persisted, and I remained unable to configure a wireless network.
This issue does not occur with the current version openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-bcm27xx-bcm2711-rpi-4-ext4-factory 2.img. However, I require the ability to test with multiple schedulers.

How can I use OpenMPTCPRouter with multiple schedulers while avoiding the error mentioned?

## Specifications

* OpenMPTCProuter version: v0.60
* OpenMPTCProuter VPS version: 0.1028
* OpenMPTCProuter VPS provider: GCP
* OpenMPTCProuter platform: RPI4

Ysurac, 2024-08-05:
Do you have a more complete error message ? Can you make a screenshot ? What you do to get the error ?
You can try v0.61 snapshot image, with 6.6 kernel, schedulers are back.

Raquel1L, 2024-08-06:
Here is the screenshot of the error I get:

[image]

closed 2024-11-10 by github-actions[bot]