Ysurac/openmptcprouter #3481: DMZ or Help with NAT for public IPs

JshGrn, 2024-08-06:
## Expected Behavior
I want all traffic that comes to my VPS to be routed to a specific IP address bypassing NAT on the router. For example anything that comes into my VPS 123.123.123.123 I want routed to 192.168.80.100. I do not want NAT at all here if possible.

## Current Behavior
I add a port forward for 192.168.80.100 and the port does not show open until I add a NAT rule. When I add a NAT rule the server I am using sees all public traffic as 192.168.80.1. I tried SNAT and MASQUEADE, I also tried changing Loopback to external IP with no success.

I actually don't really need the router features at all, I literally just need the VPS to forward all traffic to a IP within my network. My current setup is Mikrotik handling 3 WANs, I have a VLAN interface which the OpenMPTCPRouter serves its LAN. I have 2 other VLAN interfaces which are WAN1 and WAN2. Everything else is handled within the network

Is it possible to do this? If not, how can I make it so that the public IP is visible to my service rather than the router gateway IP?

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-23-amd64
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-08-07:
If you check on VPS, routes to the local network should be available. You can add a rule to use them in ```/etc/shorewall/rules``` on the VPS.

JshGrn, 2024-08-07:
Is adding them on the router not doing this?

JshGrn, 2024-08-07:
Looking on the VPS, when I run `route` I see:

```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         vpsgatewayip 0.0.0.0         UG    0      0        0 enp1s0
10.255.246.0    0.0.0.0         255.255.255.0   U     0      0        0 client-wg0
10.255.247.0    0.0.0.0         255.255.255.0   U     0      0        0 wg0
10.255.248.0    0.0.0.0         255.255.255.0   U     0      0        0 omr-bonding
10.255.248.2    0.0.0.0         255.255.255.255 UH    0      0        0 omr-bonding
10.255.250.0    0.0.0.0         255.255.255.0   U     0      0        0 tun1
10.255.251.2    0.0.0.0         255.255.255.255 UH    0      0        0 dsvpn0
10.255.252.0    0.0.0.0         255.255.255.0   U     0      0        0 tun0
10.255.253.0    0.0.0.0         255.255.255.0   U     0      0        0 mlvpn0
10.255.254.0    0.0.0.0         255.255.255.252 U     0      0        0 gt-udp-tun0
10.255.255.0    0.0.0.0         255.255.255.252 U     0      0        0 gt-tun0
45.77.228.0     0.0.0.0         255.255.254.0   U     0      0        0 enp1s0
169.254.169.254 vpsgatewayip     255.255.255.255 UGH   0      0        0 enp1s0
192.168.80.0    10.255.252.2    255.255.255.0   UG    0      0        0 tun0
```

So the route to 192.168.80.0 should be using the gateway, but I cannot ping the service at 192.168.80.162 nor can I curl it.

Ysurac, 2024-08-07:
You should be able to ping it from VPS, as ping is open in router firewall, you can't curl as it's close by router firewall rules.
You can accept all traffic from/to VPN in network->Firewall

JshGrn, 2024-08-07:
I cannot ping it from the VPS. All traffic from/to VPN is enabled in Network -> Firewall 

```
root@vultr:~# ping 192.168.80.1
PING 192.168.80.1 (192.168.80.1) 56(84) bytes of data.
^C
--- 192.168.80.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

root@vultr:~# ping 192.168.80.162
PING 192.168.80.162 (192.168.80.162) 56(84) bytes of data.
^C
--- 192.168.80.162 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2024ms
```

JshGrn, 2024-08-07:
Not really sure what I need to look at next to debug why I can't ping from the VPS one of the devices connected to the router?

Ysurac, 2024-08-07:
What is the VPN used ? From the IP, I would say it's OpenVPN TCP.
What is the LAN IP of OpenMPTCProuter ?
What do you get when you try a ```tcpdump -i tun0 icmp``` via SSH on OpenMPTCProuter while doing a ping to OpenMPTCProuter IP ?

JshGrn, 2024-08-07:
> What is the VPN used ? From the IP, I would say it's OpenVPN TCP. What is the LAN IP of OpenMPTCProuter ? What do you get when you try a `tcpdump -i tun0 icmp` via SSH on OpenMPTCProuter while doing a ping to OpenMPTCProuter IP ?

VPN = Standard config, OpenVPN
LAN IP of OMR = 192.168.80.1

Ping on VPS:
```
root@vultr:~# ping 192.168.80.1
PING 192.168.80.1 (192.168.80.1) 56(84) bytes of data.
^C
--- 192.168.80.1 ping statistics ---
7 packets transmitted, 0 received, 100% packet loss, time 6145ms
```

TCPDump on Router:
```
root@OpenMPTCProuter:~# tcpdump -i tun0 icmp
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on tun0, link-type RAW (Raw IP), snapshot length 262144 bytes
18:56:21.952033 IP 10.255.252.2 > 10.255.252.1: ICMP echo request, id 28893, seq 1, length 64
18:56:21.967609 IP 10.255.252.1 > 10.255.252.2: ICMP echo reply, id 28893, seq 1, length 64
18:56:27.046759 IP 10.255.252.2 > 10.255.252.1: ICMP echo request, id 12355, seq 1, length 64
18:56:27.063779 IP 10.255.252.1 > 10.255.252.2: ICMP echo reply, id 12355, seq 1, length 64
18:56:32.144615 IP 10.255.252.2 > 10.255.252.1: ICMP echo request, id 51727, seq 1, length 64
18:56:32.161344 IP 10.255.252.1 > 10.255.252.2: ICMP echo reply, id 51727, seq 1, length 64
18:56:37.244890 IP 10.255.252.2 > 10.255.252.1: ICMP echo request, id 9785, seq 1, length 64
18:56:37.261097 IP 10.255.252.1 > 10.255.252.2: ICMP echo reply, id 9785, seq 1, length 64
18:56:42.342077 IP 10.255.252.2 > 10.255.252.1: ICMP echo request, id 62030, seq 1, length 64
18:56:42.357961 IP 10.255.252.1 > 10.255.252.2: ICMP echo reply, id 62030, seq 1, length 64
```

So, why isn't it responding.... hmm..

JshGrn, 2024-08-07:
What is the reason OpenVPN is default? I could change the VPN type but I don't understand the tradeoffs between them.

Ysurac, 2024-08-09:
As proxy, some VPNs works better in some usage.
I modified VPS script, OpenVPN should allow route now.

To configure proxy or VPN, it's in System->OpenMPTCProuter, "wizard" tab and "advanced settings" checkbox.

JshGrn, 2024-08-09:
What is the process to update VPS?

Ysurac, 2024-08-09:
https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

JshGrn, 2024-08-09:
That page is not clear, and looks like thats for the router?

Ysurac, 2024-08-09:
Please...

JshGrn, 2024-08-09:
Is that page snapshots for the VPS? There is a lot of translational issues on the Wiki and instructions, including on the router, so it is difficult to understand specifically sometimes.

JshGrn, 2024-08-09:
I updated the VPS, and routing table is still the same and unable to ping the router nor the client connected to router from the VPS.

Ysurac, 2024-08-09:
I fixed the script, should be better

JshGrn, 2024-08-09:
I re-ran the script, no change

Ysurac, 2024-08-09:
You reboot after ?
What do you have in /etc/openvpn/ccd on the server ?

JshGrn, 2024-08-09:
```
root@vultr:/etc/openvpn/ccd# cat ipp_udp.txt 
openmptcprouter,10.255.252.2,
```

```
root@vultr:/etc/openvpn/ccd# cat ipp_tcp.txt 
openmptcprouter,10.255.252.2,
```

```
root@vultr:/etc/openvpn/ccd# cat openmptcprouter 
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
[... 99 lines cut ...]
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
iroute 192.168.80.0/24 255.255.255.0
```

Ysurac, 2024-08-10:
Can you do a rm /etc/openvpn/ccd/openmptcprouter ?

JshGrn, 2024-08-14:
The file recreated after a reboot without the /24:

```
iroute 192.168.80.0 255.255.255.0
```

closed 2024-11-17 by github-actions[bot]