Ysurac/openmptcprouter #3482: Cannot aggregate UDP traffic

NiColtro, 2024-08-07:
## Expected Behavior
I am trying to aggregate UDP traffic, but with all kinds of methods it doesn't works.
Specifically, I'm trying to pass an IPSec tunnel inside OMR.

## Current Behavior
With Shadowsocks (Rust and "old") with TCP traffic I can easily reach about 400Mbps.
Using V2Ray / XRay proxies (labelled as "UDP aggregation-ready") both UDP and TCP traffic is slowed down to 4Mbps.
Enabling Shadowsocks obfuscation with both http or tls methods TCP performs good but UDP traffic is still about 7Mbps.
Am I doing something wrong? Many thanks in advance.

## Specifications

  - OpenMPTCProuter version: : v0.60-6.1
  - OpenMPTCProuter VPS version: 6.1.0-23
  - OpenMPTCProuter VPS provider: Aruba - VMWare
  - OpenMPTCProuter platform: x86_64

NiColtro, 2024-08-07:
Thanks. I've tried several *Ray proxies... but even with the flag enabled, the UDP speed is slower (2Mbit ish) than every single WAN. TCP speed also drops down from 600Mbit with Shadowsocks to 500Mbit (V2Ray VLESS specifically) but on the other side if also aggregates UDP will be more than enough.
I've read that there's no "Best VRay / XRay" but you have to try them all, based on the use case. Is that correct?

NiColtro, 2024-08-08:
Hello.
Still couldn't find the right settings.
Shadowsocks remains the best in TCP performances, and UDP won't go over 5Mbit with any kind of *Ray Proxy (with the flag you told me yesterday activated).
Do I need to provide you more informations about the issue?
Thanks again.

NiColtro, 2024-08-08:
Hello.

I finally found out that UDP aggregation was working.
In the first place it was tested using an IPSec tunnel, reaching that performance mark (5Mbit)... but then switching to iPerf3 test the download / upload speed are slighly below the TCP ones (150Mbit TCP / 110Mbit UDP).

Remains the issue that IPSec is not going through aggregated... and is a main usage case for us.
Have you ever had the opportunity to test this use case? Have you ever encountered any problems with this?

Thanks again.

Ysurac, 2024-08-08:
You can check IPsec MTU.
If UDP is aggregated, then everything that is not TCP use the same way.

NiColtro, 2024-08-08:
Ok thanks, I am gonna try that.

NiColtro, 2024-08-10:
Hi.
I conducted another test without IPSec, using only UDP traffic (saturating the link capacity with a Speedtest via Cloudflare WARP).
Using V2Ray with only TCP traffic in OMR, I can easily achieve download speeds of 500 Mbps and upload speeds of 100 Mbps.
However, with UDP, the behavior is really strange: I can aggregate “upload” speeds effectively, reaching 30 Mbps (which is slow, but at least it aggregates and distributes the load between WANs). In contrast, when downloading, I consistently achieve only 4 Mbps using master WAN.

It seems that the incoming traffic (from the VPS to OMR) is not aware of the individual WANs’ performance, leading to this issue. Does this make sense?

Thanks.

closed 2024-11-13 by github-actions[bot]