Ysurac/openmptcprouter #3483: OMR-Bypass Netflix

Majaxx, 2024-08-07:
## Expected Behavior
Not being recognized as a VPN user on Netflix

## Current Behavior
Unable to launch certain series on Netflix.
OMR bypass test URL: 
fast.com
nflxvideo.net
nflximg.com
netflix.com
nflxext.com
nflximg.net - no response
nflxso.net - no reponse

Test via tracetcp ok on wan1

Error message netflix ; Il semblerait que vous utilisiez un vpn 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-snapshot-6.6 r0+26695-820823198d
  - OpenMPTCProuter VPS version: 0.10.30 6.6.36-x64-v3-xanmod1
  - OpenMPTCProuter VPS provider: Pulseheberg
  - OpenMPTCProuter platform: Optiplex 3060 - Proxmox 8.2

JshGrn, 2024-08-07:
This will be because of the WAN IP of your VPS

Majaxx, 2024-08-07:
OMR ByPass would be ineffective for Netflix?

JshGrn, 2024-08-07:
No, your IP of your VPS provider is flagged with Netflix for being a VPN provider. You'd need to use a lowkey VPS provider to get away with it.

Ysurac, 2024-08-07:
OMR-ByPass allow to exit directly via a WAN. I don't really understand what you mean with tracetcp, a traceroute ?
For OMR-ByPass to work correctly, you need to have OpenMPTCProuter IP as DNS on your client computer. This may also take some time to refresh local client DNS cache.

JshGrn, 2024-08-07:
He is hitting a limit that Netflix have in place to stop people using non-residential connections to stream.

Ysurac, 2024-08-07:
True, it's why OMR-ByPass is created, this bypass VPS and exit directly via a direct connection (so with no aggregation) for selected service/domain/....

Majaxx, 2024-08-07:
TraceTCP : https://github.com/0xcafed00d/tracetcp

DNS : 192.168.100.1 ( OpenMTCPRouter)

tracert : Connect netflix.com via vps 
```
tracert netflix.com
Détermination de l’itinéraire vers netflix.com [52.214.181.141]
avec un maximum de 30 sauts :
  1     2 ms     2 ms     2 ms  OpenMPTCProuter.lan [192.168.100.1]
  2    39 ms    48 ms    44 ms  10.255.252.1
```

tracetcp :  pass to wan1
```
 .\tracetcp.exe netflix.com

Tracing route to 18.200.8.190 [ec2-18-200-8-190.eu-west-1.compute.amazonaws.com] on port 80
Over a maximum of 30 hops.
1       8 ms    6 ms    5 ms    192.168.100.1   [OpenMPTCProuter.lan]
2       4 ms    7 ms    4 ms    192.168.2.254
```

Ysurac, 2024-08-07:
From where is done the first tracert and the second one ?

Majaxx, 2024-08-07:
The two tests since my computer ( ipv4 only )

Ysurac, 2024-08-07:
ok, so it's seems to be working on TCP. You may try to reboot else wait for Netflix to check again that your account is not using a VPN.

Dragonox77, 2024-08-08:
> From where is done the first tracert and the second one ?

First test is Tracert and second one is TraceTCP, they are not the same test. Tracert seems to pass through OMR and not exiting through his WAN directly.

Ysurac, 2024-08-08:
@Dragonox77 yes, as I said TCP is working, tracert use ICMP.