Ysurac/openmptcprouter #3484: Link saturation on aggregated bandwidth

MB178, 2024-08-08:
Hello,
I have a use case and would like to understand the behavior of OMR in this scenario:
If I have two links connected to OMR via eth0 and eth1, and I perform a speed test using omr-test-speed to measure the aggregated bandwidth, then initiate a second speed test on eth0 with omr-test-speed eth0 to saturate that link, how will OMR respond? Specifically, will OMR shift all traffic to the non-saturated link because it will have the lowest latency, causing the aggregated bandwidth to decrease? Alternatively, will the congestion algorithm in OMR attempt to mitigate the issue and manage the traffic distribution to reduce the impact of saturation?

Ysurac, 2024-08-11:
OMR will switch the the non satured link.
There is multiple TCP Congestion Control available in OMR, BBR3 will help a little. SQM may also be needed in this case to not saturate totally a link.

MB178, 2024-08-12:
Will the aggregated bandwidth decrease?

closed 2024-11-16 by github-actions[bot]