Ysurac/openmptcprouter #3485: No redundant option in multipath TCP scheduler

ccmks, 2024-08-08:
## Expected Behavior
There is redundant option on kernel 6.1

## Current Behavior
There is none redundant on multipath TCP scheduler

Is there anyway to get it back? or planning to get it back on future release?

Thank you

davidohne, 2024-10-29:
I'm on the latest snapshot for z8102ax. Can't see the option it as well?

[image]

Ysurac, 2024-10-29:
Burst => same as default
First => always picks the first subflow to send data
Red => send all packets redundantly on all available subflows
rr => always pick the next available subflow to send data (round-robin)

closed 2025-02-01 by github-actions[bot]