Ysurac/openmptcprouter #3488: iPhone Hotspot not working

AppleCarmine, 2024-08-10:
## Expected Behavior
I configured OpenMPTCP on Raspberry and with a USB cable connected to my iPhone creating the interface on openMPTCP using DHCP it works correctly. I then decided to use a VM on Proxmox, I configured everything, but the connection with the iPhone does not work

## Current Behavior
When I configure the interface and go into the OpenMPTCP state I see the main interface of my Vodafone connection working, but that of my iphone hotspot that does not work
No IP defined
No Gateway defined

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS provider: Digitalocean
  - OpenMPTCProuter platform: x86_64

[image]
[image]

AppleCarmine, 2024-08-10:
logs: https://pastebin.com/62JfW0Pj

Ysurac, 2024-08-11:
DHCP client is running on eth1 but seems to get no answer from the iPhone.
You can see DHCP traffic, using, via SSH on the router, ```tcpdump -i eth1 -pvn port 67 and port 68```

AppleCarmine, 2024-08-11:
> DHCP client is running on eth1 but seems to get no answer from the iPhone. You can see DHCP traffic, using, via SSH on the router, `tcpdump -i eth1 -pvn port 67 and port 68`

[image]

AppleCarmine, 2024-08-11:
I also tried assigning a static IP address with the following settings:
IPv4 Address: 172.20.10.3
IPv4 Subnet Mask: 255.255.255.240
IPv4 Gateway: 172.20.10.1
IPv4 Broadcast: 172.20.10.1
DNS: 172.20.10.1

But it still doesn’t work. The issue isn’t with passthrough on Proxmox, as I can connect to the internet using the iPhone’s hotspot on other VMs.

AppleCarmine, 2024-08-12:
Someone can help me?

Ysurac, 2024-08-12:
I don't have an iPhone, maybe this can help: https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering

AppleCarmine, 2024-08-12:
> I don't have an iPhone, maybe this can help: https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering

Nothing to do, I tried everything in the guide and I continue to have the same problem. The problem is not the iPhone because with Raspberry PI3 it works fine.

closed 2024-11-16 by github-actions[bot]