Ysurac/openmptcprouter #3494: Connectivity Issue with Round Robin and Red in OpenMTPCProuter 6.1

Raquel1L, 2024-08-13:
I am currently using version 6.1 of OpenMTPCProuter on my router (RPi4) and have encountered an issue: the Round Robin and Red schedulers are not functioning as expected. When I select either option, I lose internet connectivity. In contrast, only the First, Default, and Burst schedulers appear to work properly.

Is anyone else experiencing this issue, or could it be due to an error on my part? Is this a known problem?

Ysurac, 2024-08-14:
These schedulers are not available on kernel 6.1

Raquel1L, 2024-08-14:
So what could I do to use the schedulers?

Ysurac, 2024-08-14:
So this is snapshot 0.61 with kernel 6.6.
The schedulers are not yet tested.

Ysurac, 2024-08-14:
I just tested both rr and red MPTCP scheduler and I have no connectivity issues.
Do you have something in Status->kernel log or in Status->System log ?

ccmks, 2024-08-16:
What is red on the MPTCP scheduler? Is it redundant?

Ysurac, 2024-08-24:
Yes, and rr for round-robin

closed 2024-11-28 by github-actions[bot]